lepton-eda/lepton-eda #646: lepton-schematic seems to have a major memory leak

majenkotech, 2020-06-01:
Just loading a simple schematic with just a couple of components, and moving around the work area causes the memory usage to grow and grow and grow (just watch it in `htop`...) until it uses all your available memory and the OOM killer terminates it.

Here's a brief report from `valgrind` after just a few seconds of use:

```
==163378== HEAP SUMMARY:
==163378==     in use at exit: 18,714,328 bytes in 192,721 blocks
==163378==   total heap usage: 1,502,868 allocs, 1,310,147 frees, 302,173,508 bytes allocated
==163378== 
==163378== LEAK SUMMARY:
==163378==    definitely lost: 1,055,420 bytes in 6,739 blocks
==163378==    indirectly lost: 11,082,860 bytes in 99,680 blocks
==163378==      possibly lost: 66,466 bytes in 540 blocks
==163378==    still reachable: 5,032,350 bytes in 75,577 blocks
==163378==                       of which reachable via heuristic:
==163378==                         length64           : 13,072 bytes in 190 blocks
==163378==                         newarray           : 2,256 bytes in 61 blocks
==163378==         suppressed: 0 bytes in 0 blocks
```

graahnul-grom, 2020-06-02:
@majenkotech Thanks for the report. Could you please show the
`lepton-schematic --version` and `guile --version` output?
What operating system are you using?
Do you see anything suspicious in the `lepton-schematic` log
when this happens?

majenkotech, 2020-06-02:
Lepton is git HEAD:

    Lepton EDA 1.9.10.20200319 (git: 1e4c6d7)

Guile is 2.0.13:

    guile (GNU Guile) 2.0.13
    Packaged by Debian (2.0.13-deb+1-5.4)

OS is Ubuntu 20.04 64-bit

Where's the lepton-schematic log? You mean the log window? No, there's nothing in there.

vzh, 2020-06-02:
Thanks once more! I see, it's almost master HEAD. Unfortunately, I see the issue, and it is in my new brand rendering code :-( To be more precise, in the text bounds calculation code with all its renderer contexts and surfaces creation, destroying, updating etc.  It definitely needs much more attention.

majenkotech, 2020-06-02:
Good luck with that then :)

graahnul-grom, 2020-06-02:
I can reproduce it by zooming in/out many times by a large amount
and monitoring memory usage in `htop`.

vzh, 2020-06-02:
Please check if #647 fixes it.

majenkotech, 2020-06-02:
I think you have found the major memory leak there.

There is still something not being freed somewhere, but that may be something that is allocated once and never changed, since it doesn't seem to be growing. So it can be, for now, ignored.

```
==127258== HEAP SUMMARY:
==127258==     in use at exit: 6,390,285 bytes in 87,925 blocks
==127258==   total heap usage: 1,748,273 allocs, 1,660,348 frees, 328,914,175 bytes allocated
==127258== 
==127258== LEAK SUMMARY:
==127258==    definitely lost: 40,542 bytes in 624 blocks
==127258==    indirectly lost: 156,577 bytes in 6,572 blocks
==127258==      possibly lost: 9,530 bytes in 100 blocks
==127258==    still reachable: 5,625,284 bytes in 76,488 blocks
==127258==                       of which reachable via heuristic:
==127258==                         length64           : 13,152 bytes in 192 blocks
==127258==                         newarray           : 2,256 bytes in 61 blocks
==127258==         suppressed: 0 bytes in 0 blocks
```

vzh, 2020-06-02:
Please don't rush closing it. Github will close it automatically when the branch will be applied to master.

PR #647 (2020-06-02): Fix serious memory leak in text rendering code
Closes #646.
closed 2020-06-02 by vzh