lepton-eda/lepton-eda #650: Suggestions for shortcuts

zo0om, 2020-06-07:
* Ctrl-S or its OS-specific equivalent is universally used to save, but doesn't work (at least on Ubuntu)
* Ctrl-- and Ctrl-+/Ctrl-= are common keys to zoom out and in

graahnul-grom, 2020-06-07:
You can override default shortcuts by placing the following
to `$HOME/.config/lepton-eda/gschemrc` file:

```scheme
( global-set-key "<Control>s" '&file-save )
( global-set-key "<Control>plus" '&view-zoom-in )
( global-set-key "<Control>minus" '&view-zoom-out )
( global-set-key "<Control>equal" '&view-zoom-extents )
```

Please see the [menu file](https://github.com/lepton-eda/lepton-eda/blob/master/schematic/scheme/conf/schematic/menu.scm) for other commands.

vzh, 2020-06-08:
I'm for such a change. Nothing breaks if we add these shortcuts.  I would even add just `plus` and `minus` (without any modifier keys) as defaults. And probably `=` as an alias for `V E` (zoom extents). WDYT, folks?

graahnul-grom, 2020-06-08:
Yes, it would be convenient.

```scheme
( global-set-key "plus"         '&view-zoom-in )
( global-set-key "KP_Add"       '&view-zoom-in )
( global-set-key "minus"        '&view-zoom-out )
( global-set-key "KP_Subtract"  '&view-zoom-out )
( global-set-key "equal"        '&view-zoom-extents )
```

`Ctrl+S`: in the same vein, we could add `Ctrl+N` for `file-new`
and `Ctrl+O` for `file-open`;

```scheme
( global-set-key "<Control>s" '&file-save )
( global-set-key "<Control>n" '&file-new  )
( global-set-key "<Control>o" '&file-open )
```

vzh, 2020-06-08:
@graahnul-grom, could you make a PR with those changes, please?

graahnul-grom, 2020-06-08:
#652
I've also added `Ctrl+P` for `file-print`.
@zo0om, @vzh Could we consider this issue as closed then?

vzh, 2020-06-08:
Ehmm, @zo0om asked for 'ctrl-+' and 'ctrl--'. @graahnul-grom, could you please add them, too?

graahnul-grom, 2020-06-08:
And `Ctrl+=`?

vzh, 2020-06-08:
I'm not sure about it. @zo0om, WDYT about making `Ctrl`-`equal` work as `V E` (zoom extents)?

zo0om, 2020-06-08:
`Ctrl-equal` should do the same as `Ctrl-+`, zoom in (`Z`), so having to press shift is optional. I think that's the convention - it certainly does it in evince (PDF viewer).

zo0om, 2020-06-08:
While your enthusiasm is engaged, you might want to add `Shift-Ctrl-Z` for redo (as well as `Ctrl-Y`). I find it useful for moving around the undo buffer quickly.

graahnul-grom, 2020-06-08:
`+` - zoom in, `-` - zoom out. It's logical and convenient. Without
any modifiers.
Although we are all accustomed to `Ctrl+S` and `Ctrl+O`,
there are no standard "conventions" for zoom, `redo` and other keys.
Different apps do use different combinations.
`evince` is not an example of perfect GUI and usability. IMHO,
quite the contrary. `+` and `=` zooms in, however, while `-` zooms
out, `_` does not. It ignores the `Ctrl` modifier, BTW.
And I think `=` for `view-zoom-extents` is OK.

graahnul-grom, 2020-06-08:
Another possibility is to use `+` and `=` for zoom in, `-` for zoom out
and `0` for zoom restore. WDYT?

graahnul-grom, 2020-06-08:
...and keypad `*` for view-zoom-extents.

graahnul-grom, 2020-06-08:
I've tried it and I like it. :-) Three adjacent keys both on keypad and
main keyboard, no need to press any modifiers:

```scheme
( global-set-key "equal" '&view-zoom-in )
( global-set-key "minus" '&view-zoom-out )
( global-set-key "0"     '&view-zoom-extents )

( global-set-key "KP_Add"      '&view-zoom-in )
( global-set-key "KP_Subtract" '&view-zoom-out )
( global-set-key "KP_Multiply" '&view-zoom-extents )
```

graahnul-grom, 2020-06-08:
FWIW, [evince keyboard shortcuts](https://help.gnome.org/users/evince/stable/shortcuts.html.en).

graahnul-grom, 2020-06-08:
I'm for `Ctrl+Shift+Z` => `edit-redo`. It doesn't conflict with any
existing actions.

zo0om, 2020-06-08:
On a lot of keyboards including UK and US, + is accessed with shift, which is why the unshifted character = is mapped to the same function. Therefore you can zoom in and out either by strictly adhering to the + and - convention or not bothering with the shift key for +, which is inconvenient.

Firefox uses this exact mapping.

Chromium also maps _, so the state of the shift key is ignored completely. I don't have any objection to this.

While we're on the subject of zooming, Ctrl-0 also zooms you to the default level!

vzh, 2020-06-09:
I'm OK with everything suggested.  Yes, it is more convenient if `+` and `=` do the same job.  `0` and probably `Ctrl+0` may be a better for `zoom extents`. @zo0om, zooming to extents is not some default zooming level, it's just such a zoom when all visible objects are included inside the visible area and other, not populated space, excluded.  I wouldn't mind to use `Ctrl+Shift+Z` for redo as well.

graahnul-grom, 2020-06-09:
Matthew, could you please put the following expressions to
`~/.config/lepton-eda/gschemrc` and test the shortcuts?

```scheme
; File:
;
( global-set-key "<Control>N" '&file-new  )
( global-set-key "<Control>O" '&file-open )
( global-set-key "<Control>S" '&file-save )
( global-set-key "<Control>P" '&file-print )

; Edit->Redo:
;
( global-set-key "<Control><Shift>Z" '&edit-redo )

; Zoom:
;
( global-set-key "equal" '&view-zoom-in )
( global-set-key "minus" '&view-zoom-out )
( global-set-key "0"     '&view-zoom-extents )

( global-set-key "<Control>equal" '&view-zoom-in )
( global-set-key "<Control>minus" '&view-zoom-out )
( global-set-key "<Control>0"     '&view-zoom-extents )

; Zoom with keypad:
;
( global-set-key "KP_Add"      '&view-zoom-in )
( global-set-key "KP_Subtract" '&view-zoom-out )
( global-set-key "KP_Multiply" '&view-zoom-extents )
```

[gschemrc.txt](https://github.com/lepton-eda/lepton-eda/files/4750799/gschemrc.txt)

zo0om, 2020-06-09:
All working in 1.9.10 on Ubuntu 20.04!

graahnul-grom, 2020-06-10:
Thanks! Any other suggestions? What about merging #652?