lepton-eda/lepton-eda #651: Cannot select a locked object

zo0om, 2020-06-07:
Obviously you don't want to be able to edit it, but it's confusing for us newbies, so maybe it could produce a dialogue telling you it's locked, and possibly offering to unlock it? I suppose that should only happen if you click on an individual object rather than multi-selecting by dragging an area.

graahnul-grom, 2020-06-21:
@zo0om Matthew, I hope the attached Guile script will help you to
work with locked objects, please test it.
It binds two keyboard shortcuts: `Ctrl+Shift+L` to select all locked
objects on the active page, and `Shift+L` to select each one in
succession - then, you can press `E Shift+L` to unlock the object
or `E E` to edit its attributes.
Keys are customizable, edit the shortcuts definitions in the
script (the first two lines of code).

[image]

With the schematic above (attached) when you press `Ctrl+Shift+L`,
all locked components should become selected: R1, C1
and the title block. Pressing `Shift+L` several time should select
R1, then C1, then titleblock, then R1 again, etc.
You should also see messages in the log window:

[image]

Shortcuts:

[image]

The script (`select-locked.scm`) can be loaded from the `gschemrc`
configuration file (e.g. `~/.config/lepton-eda/gschemrc`):

```scheme
( primitive-load "/path/to/select-locked.scm" )
```

[select-locked.tar.gz](https://github.com/lepton-eda/lepton-eda/files/4809614/select-locked.tar.gz)

graahnul-grom, 2020-06-22:
I have improved the [script](https://graahnul-grom.github.io/scm/select-locked.scm.html), `Ctrl+L` now selects a locked component
under the mouse cursor.

[select-locked-improved.tar.gz](https://github.com/lepton-eda/lepton-eda/files/4814236/select-locked-improved.tar.gz)

graahnul-grom, 2020-06-23:
Simplified and fixed [script](https://graahnul-grom.github.io/scm/select-locked.scm.html) [[download](https://graahnul-grom.github.io/scm/select-locked.scm)].
Correctly handle symbols with no graphical elements and
placeholders for missing symbols.
At the moment, placeholders for locked components can be
selected by cycling through them (`Shift+L`), but not by
pointing with the mouse (`Ctrl+L`).
Further improvement would be to determine the placeholder type
used (read it from config: `schematic.gui::small-placeholders`)
and calculate each placeholder's bounds, including them to the
list of locked objects tested against the mouse pointer position.

You do not need to modify script's code to change the
key bindings anymore. Set them the usual way by adding
the `global-set-key` expressions to your `gschemrc` file
(changing shortcuts as needed):

```scheme
( global-set-key "<Control><Shift>L" 'select-locked-all )
( global-set-key "<Shift>L"          'select-locked-next )
( global-set-key "<Control>L"        'select-locked-under-mouse )
```

@zo0om Could you please test it?
In case this script works for you (I hope), could you consider
it as a solution of the problem described in this issue?

[select-locked-improved-2.tar.gz](https://github.com/lepton-eda/lepton-eda/files/4820342/select-locked-improved-2.tar.gz)

graahnul-grom, 2020-07-04:
Done. The [script](https://graahnul-grom.github.io/scm/select-locked.scm.html) [[download](https://graahnul-grom.github.io/scm/select-locked.scm)] determines the type of placeholders used and calculates their bounds.
Now placeholders for missing locked components under the mouse
cursor can be selected with `Ctrl+L`.

[select-locked-improved-3.tar.gz](https://github.com/lepton-eda/lepton-eda/files/4873325/select-locked-improved-3.tar.gz)

graahnul-grom, 2023-03-03:
I've adapted my `select-locked` script [1] for `lepton-schematic`.
`Edit->Select Locked` menu item has been added; it cycles
selection through locked components on the current page,
printing log messages.
No other functions (there were 'select all locked' and
'select locked under mouse' - is it worth to retain them?).
Script itself has been converted to module `(schematic slck)`,
exporting `select-locked-next()` function.
The *action* is named `&edit-select-locked`.
Haven't assigned a key to it yet.

@vzh Could you please take a look at these changes [2]?

[1] github.com/graahnul-grom/graahnul-grom.github.io.git/scm/select-locked.scm
[2] https://github.com/graahnul-grom/lepton-eda/tree/tb_364_select_locked

PR #1022 (2023-05-16): Selection of locked components in lepton-schematic
Add new `&edit-select-locked` action; `Edit->Select Locked`
menu item and `E K` shortcut for it.
It selects locked components on current page one by one.
If there is none, shows a message box about that.
Information about currently selected component is shown
in the log and, briefly, in the status bar.
Implemented in new `(schematic sellock)` module.
Based on my [select-locked](https://graahnul-grom.github.io/scm/select-locked.scm.html) script.

Closes #651.

[image]

[image]

[image]

[image]
closed 2023-05-16 by graahnul-grom