lepton-eda/lepton-eda #653: Document compability

spth, 2020-06-11:
I am currently using gschem 1.8.2, the latest gschem version in Debian. Apparently the geda-gschem / geda-gaf maintainer moved to lepton, so (like probably many pre-fork gschem users) wonder:

* Will I be able to work with old gEDA files using lepton (I guess yes)?
* Will I be able to work with past-fork gEDA files using lepton (I have no idea)?
* Will I be able to work with lepton files using post-fork gEDA (I have no idea)?

It would be good to have a section on compability in the README file.

graahnul-grom, 2020-06-11:
Lepton EDA and gEDA uses the same file format, nothing has changed
in this regard, so
1) yes
2) yes
3) yes

Thank you for the good suggestion, we will add notes on compatibility
to the README file.

vzh, 2020-06-11:
One small addition to what @graahnul-grom has already said.  So far, we supported gEDA file format and are planning to do so in future since there are lots of symbols/schematics created using it.  If there will be some additions on the `geda-gaf` side (e.g., wrt attributes, or new file format version, and such), it depends on users of Lepton, on you, to point us to what you think is not supported but should be.  The same is true wrt other features in `geda-gaf` Lepton may not have.  We are not able to fully keep track of that project, so again, we depend on our users here and need your feedback.

PR #691 (2020-10-30): Manual improvements
- Split up the manual into several sections
- Add geda file format document (under GFDL).
- Add a couple of pictures.
- Add edits to README about compatibility with geda-gaf.

Affects #284 
Closes #653
closed 2020-10-30 by vzh