lepton-eda/lepton-eda #655: Overbar doesn't scale vertically

zo0om, 2020-06-12:
lepton-schematic 1.9.10 (git: 56bb6e5) Ubuntu.

When you put an overbar on text by using `\_`, it doesn't scale in the vertical axis relative to the text (i.e vertically if the text is horizontal or horizontally if the text is vertical) when you zoom the page.
It therefore progressively obliterates the text below as you zoom out (or gets weak and weedy if you zoom in an awful long way):
[image]
[image]

Another observation (not a bug) is that you only need the opening `\_` for the overbar to appear; not the closing one too, as documented.

graahnul-grom, 2020-06-13:
I believe this has been fixed in #629.
Could you please check it with the latest Lepton EDA release (1.9.11)?
lepton-eda-1.9.11 package is [available](https://repology.org/project/lepton-eda/versions) in Ubuntu 20.10.

zo0om, 2020-06-13:
Thanks! I should remember I don't have the latest version. I only have Ubuntu 20.04 so built from source and can confirm that the problem is fixed.

closed 2020-06-13 by zo0om

graahnul-grom, 2020-06-14:
> Another observation (not a bug) is that you only need the opening \_ for the overbar to appear; not the closing one too, as documented.

Confirmed, thank you.
[eda_pango_parse_overbars()](https://github.com/lepton-eda/lepton-eda/blob/f04cd559ea3dd1f54ae8e996e6fc48924dea0ba6/libleptonrenderer/edapangorenderer.c#L370) in `libleptonrenderer`.
I'll look into this.

graahnul-grom, 2020-06-18:
It's intentional:

```c
/* If we've previously found an overbar delimiter, and either we
* find a null byte or another overbar delimiter, create an
* overbar attribute for the intervening run of characters. */
if (overbar_start && (overbar_end || (*in_ptr == '\0')))
{
  // ...
}
```

`xxx-\_overbar1\_-xxx-\_overbar2\_-xxx-\_the-rest-of-the-string`

[image]

Something wrong with the documentation, then.
@zo0om Matthew, where did you read about that?

graahnul-grom, 2020-06-18:
You do not have to build `1.9.11` from source, it's possible to download
the package from https://packages.ubuntu.com/groovy/lepton-eda
for Ubuntu 20.10 ("groovy"\*) and install it with `dpkg -i` on
20.04 LTS ("focal"\*) (just do not forget to uninstall the previous
version and delete the Guile cache: `~/.cache/guile`).

\* *A lyrical aside*
Why, why do we have to memorize not only Ubuntu (and others) versions,
but also idiotic childish "nicknames" of the releases?! Do they
want us to train our memory this way? Thanks, we've got absolutely
nothing else to do. Some 100500 Linux distros out there, easy!
Several years ago, similarly gifted kids tried to do something like
that to FreeBSD. Fortunately, the adults were passing by at that time.
They caught those naughty little boys, slightly slapped them to their
rear ends, while giving them a usual lecture: respect your parents,
brush your teeth, be diligent at school, etc. There was a little
whining, but five minutes later we saw them laughing and playing
again with their beloved toy cars and wooden horses.
Everything's back to normal.
Берегите детей! Дети - наше всё. И ваше. :-)

zo0om, 2020-06-18:
Ah ok - might be worth putting a note somewhere that the 20.10 version will work on 20.04 as that's current and also long-term supported so many will stick with it.

As for the overbar doc, in my system it's at `file:///usr/share/doc/lepton-eda/wiki/geda-gschem_symbol_creation.html`. I understand now that the character sequence is a toggle operation rather than a wrapping operation.

graahnul-grom, 2020-06-18:
Ubuntu page added:
https://github.com/lepton-eda/lepton-eda/wiki/Ubuntu