lepton-eda/lepton-eda #661: Overscored signal names are unreadable in PDF

rundel-tech, 2020-06-29:
I am using lepton-cli export to generate a PDF of my design. The command is:

lepton-cli export --format=pdf --paper=iso_a4 -o $out/Schematics.pdf *.sch

The resulting schematics look fine except that every signal with an overscore is replaced by a rectangle. A similar issue affects lepton-schematic, but there the height of the overscore is affected by the zoom level. This also affects labels in symbols.

`lepton-cli --version` reports the version as "Lepton EDA 1.9.10.20200319 (git: 56bb6e5)".

PDF schematics:

[image]

lepton-schematic:

[image]

vzh, 2020-06-29:
This has been fixed along with other issues in #622 and the last release, 1.9.11, includes the fixes and is recommended for use in distributions.

rundel-tech, 2020-06-30:
That's good news. Thanks for the quick response.

rundel-tech, 2020-06-30:
For the benefit of anyone else encountering this issue, the version in the Ubuntu repositories is currently 1.9.10-2. https://packages.debian.org/unstable/lepton-eda has 1.9.11-1 (lepton-eda_1.9.11-1_amd64.deb). Installing this fixed the issue.

vzh, 2020-06-30:
Please also take a look at [this page](https://github.com/lepton-eda/lepton-eda/wiki/Ubuntu) :-)

graahnul-grom, 2020-07-02:
@rundel-tech @vzh Could we close this issue?

vzh, 2020-07-03:
I'm afraid if we close it just now, somebody will report it once again.
OTOH, the more issues we close, the better we work ;-)

closed 2020-07-03 by vzh