lepton-eda/lepton-eda #677: Incorrect position of lower aligned text on screen

adrian-purser, 2020-08-13:
The position of some text is incorrect in version 1.9.11 where it was correct in 1.9.10. The text shown in the image below has lower-right alignment. 

[image]

vzh, 2020-08-14:
Please see #622, especially [this](https://github.com/lepton-eda/lepton-eda/issues/622#issuecomment-619219190), and #629.  We've changed the way how fonts are rendered in the process of fixing pango-1.44 issues.  Now, fonts are aligned using their *baseline* instead of their *visible contents*, so you'll probably have to realign some text objects in your schematics.  Choosing another font that has other logical dimensions might be another option.

vzh, 2020-10-10:
Hi Adrian,
Please check if the branch in #688 fixes your issue.

closed 2020-10-20 by vzh

adrian-purser, 2020-10-20:
Hi, Thank you for looking into this. Sorry I missed the first message, I've only just seen it today when I got the latest email this morning. I can't seem to get the latest versions of Lepton to run on any of my machines so as soon I can figure it out I will give it a try.

Thanks again,
Ade

vzh, 2020-10-20:
OK, Adrian.
Please let us know if you'll have any issues with the change.

PR #688 (2020-10-20): Align lower aligned text in lepton-schematic by its baseline.
Instead of just using logical font sizes, take into account font
descent value for lower aligned text.
Closes #677