lepton-eda/lepton-eda #681: Lepton should migrate from GTK2 to GTK3

vzh, 2020-09-28:
It is one of Lepton goals. Moreover, @bdale reported on gitter:

```
Debian has started to tag packages that have a hard dependency on GTK version 2 with a bug.
```

vzh, 2020-09-28:
[Migrating from GTK+ 2.x to GTK+ 3](https://developer.gnome.org/gtk3/stable/gtk-migrating-2-to-3.html)

graahnul-grom, 2020-11-18:
I am strongly against *migration* to GTK3.
Lepton should *support* GTK3.
`./configure --with-gtk3`, something like that.
And I propose to change this issue's title accordingly.
Anyway, it is important to postpone all work wrt GTK3
to release 1.9.13, and not the upcoming one.
These are not simple matters. Don't be hasty, please.

graahnul-grom, 2020-11-22:
> It is one of Lepton goals. Moreover, @bdale...

I think noone could object that we do our best in helping
our users and always take into account their opinions.
Just take a look at our issue tracker, the geda-user
mailing list and gitter.im history. Many nice features were
implemented thanks to feedback from our users.
But it's not Debian team's business to define our project goals.

> Debian has started to tag packages that have a hard dependency on GTK version 2 with a bug.

Now to our dear comrade Bdale's "bug" (why in quotes - see below).
Everybody can visit the Debian's [package tracker page](https://tracker.debian.org/pkg/lepton-eda) for lepton-eda.
From there it's easy to get to the following [document](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=967565) describing what's going on.
Let me cite the most relevant part of it (emphasis mine):

"GTK 2 is used by some important productivity applications like GIMP, and
has also historically been a popular UI toolkit for proprietary software
that we can't change, so perhaps **removing GTK 2 from Debian will never be feasible**.
However, it has reached the point where a dependency on it is a bug -
**not a release-critical bug**, and **not a bug that can necessarily be fixed quickly**,
but a piece of technical debt that maintainers should be aware of."

"Be aware of" - so much for the bug.

It is perfectly OK to stay with GTK2 (for now).
Maybe it's desirable to have optional GTK3 support,
but it's not at all a top priority task.

There are many *much* more important things on our todo list.
Take, for example, the dire situation with interdependency of
`liblepton` and `libleptonrenderer` (#690), one of the reasons
why we've lost ~~Gorgeous George~~ Windows (cygwin) compatibility.

bdale, 2020-11-22:
I thought it was worth sharing the Debian bug, but feel no sense of urgency for gtk3.

Bdale

graahnul-grom, 2020-11-22:
@bdale Thank you for informing us about situation with the Debian package!
Apparently, Vladimir perceived that like a release critical bug,
and started working hard day and night (breaking everything) to add gtk3 support. :-)
That message is for him, I hope it will help him to save strength. :-)

vzh, 2020-11-22:
@graahnul-grom, thank you for your review!  I wasn't going to release Lepton in such a rough state.  Just spent about two months to port Lepton to GTK3 without any feedback after all :-)  The next release will be with GTK2 (I'm working on it now, having too little time these days).
OTOH, supporting both GTK2 and GTK3 is not feasible IMO for such a great number of active devs we have now.  So later on, I'll try to merge both libs support in the Lepton code.  However, if it starts to be too hard, I'll better switch to GTK3.  I believe it's easier to fix bugs in the new code than support anything at once.
And please, don't diminish the role of Debian in the current FOSS world.  Many other Linux distributions just pull and release its packages, and there're lots of them.  (Once again, to avoid misunderstandings, I'm not about releasing GTK3 version without proper testing.)

graahnul-grom, 2020-11-24:
In my opinion, we should continue using GTK2 as long as
practically possible. I don't think we'll be forced to
switch in near future (if ever). And it's likely that
by the time we have to migrate from GTK2, it will be
GTK4 that we'll need to upgrade to.

> Just spent about two months to port Lepton to GTK3

Perhaps, you didn't examine the bug description
beforehand, did you? :-)

> supporting both GTK2 and GTK3 is not feasible

Let's support GTK2.
Though we may continue working on GTK3 (having in mind
future migration), we could state that we unable to
provide technical support for GTK3 version yet.

It's unfortunate, that Lepton (actually, gEDA) is so
strongly depends on particular GUI toolkit.
All in all, I am very sceptical about GTK3, considering
what kind of "people" took over the GTK project [1].
Not to mention its utterly disgusting "look and feel".
There's a hope that maybe GTK4 will be better,
probably we should wait for it.

> And please, don't diminish the role of Debian

I didn't do that. Debian is not that bad. Debian's good! :)

[1] a very revealing [article](https://igurublog.wordpress.com/2012/11/05/gnome-et-al-rotting-in-threes/),
and no, after 8 years nothing has changed.

vzh, 2020-11-24:
I was aware of the bug contents.  Please note, I haven't merged the branch.  And, if my port's look-n-feel currently looks ugly for you, it's just issue with it.  Many users requested GTK3 in gEDA/Lepton before.  For me, some GTK programs look pretty nice these days.  And I believe, it will be easier to port from GTK3 to GTK4 rather than from GTK2, anyways.

Fair enough, I'll fix the description of the issue.  Let's see if I can do something useful with my branch later.

graahnul-grom, 2020-11-24:
Thank you for fixing the title, so I do not lose sleep over it. :-)
look-n-feel is not an issue with your code, I'm talking about GTK3
in general.

vzh, 2020-11-24:
As for me, working with any GTK version is a hard job.  Haven't found very much difference when worked on the branch.  The problem was the fact of transition itself, i.e. that I could not see the result until replaced all deprecated GTK2 stuff.  After that, I'm not surprised with the result :-) (and even glad it mostly works).  The core problem of GTK is its core, Glib, IMO, learning curve of which is so steep.  And the big size of the library.  (I understand though, anyone can say the same about any large and complex  enough library or language, especially having non-familiar sides, say, about Lisp/Scheme :-))

PR #685 (2020-12-03): GTK3 support for lepton-gui
Together with #682 closes #681
closed 2020-12-03 by vzh