lepton-eda/lepton-eda #686: Spaces in tEDAx output strings from lepton-netlist should be escaped

barjac, 2020-10-09:
Hi,
I am just about to package lepton-eda for the Mageia distro, for which I am the maintainer of pcb-rnd.
I have started to use lepton-eda myself and so far it is behaving well. By that I mean not crashing and losing work as has been happening for years with gschem!

However I have hit the above issue, where parametric footprints (in particular) which contain spaces between the parameters are passed un-escaped from the exporter resulting in missing footprints in pcb-rnd.

Since the main object of adding lepton-eda to our repository is for use in partnership with pcb-rnd as an alternative to gschem/pcb it would be great if this could be resolved.

Thanks,
Barry

vzh, 2020-10-10:
Well, a little schematic example with your workflow example (commands you launch to get the netlist) would not harm :-)

barjac, 2020-10-10:
tEDAx created from testboard.sch using:

lepton-netlist -g tEDAx -o testboard.tdx testboard.sch

file:///home/baz/Geda/Testboard/testboard.sch

When the generated tEDAx file is imported to pcb-rnd then we see the error:
import_sch2: reimport with tEDAx -> 0x7f5ec7dc8d00
tedax: not importing refdes=C1: no footprint specified

It is valid for spaces to be in text strings in the footprint, or other fields, but not in strings sent to a tEDAx file. These should be escaped. This is in the tEDAx spec:
http://repo.hu/projects/tedax/syntax.html

I understood that this was a known issue so did not elaborate previously, sorry.

Thanks,
Barry

vzh, 2020-10-10:
IIUC, in the tEDAx format, spaces and tabs should be escaped in any attribute value, component and net name, right?

barjac, 2020-10-12:
Well yes, but to fix this properly so future problems don't arise, the tEDAx spec needs to be implemented fully, not just enough to patch up this bug.
You need to escape any string you write, there's no exception. The easiest way would be to pass any field (word) you are going to write to an escaper function.

vzh, 2020-10-14:
IMO, patching up any bug is a step forward, so I am and will always (I hope) encourage anyone who reports or fixes issues.  There are too many issues we're trying to address, so there ever will be need of help from the users :-)
I've tried to resolve the issue in #689, and added a solution from another question reported on the geda-user list some days ago.  Looking forward for your review...

barjac, 2020-10-15:
Yes, this looks fine with the PR patch applied. Tested import to pcb-rnd with tEDAx exported where several fields had spaces and all is now good.
I did notice that fields with accidental trailing spaces are exported with these still in place. They are included in the tEDAx file suitably escaped and do not impact pcb-rnd, so do not really affect this bug report. 
However I thought it worth mentioning, as when importing to other programs these may cause issues.
Cheers,
Barrry

PR #689 (2020-10-15): Fix two issues in the tEDAx backend
- Fix escaping of whitespaces as reported in #686 (Closes #686)
- Fix output of `nettag` attributes (reported on the *geda-user* mailing list)
closed 2020-10-15 by vzh