lepton-eda/lepton-eda #687: Colour scheme changes saved to file linked in gschemrc are ignored

barjac, 2020-10-09:
I have edited the colour scheme and saved to a file, /home/baz/Geda/colours
I have created ~/.config/lepton-eda/gschemrc
containing only:
primitive-load "/home/baz/Geda/colors"
The colour code I changed, (just for 'net' at present) is correct in the 'colours' file.
What am I doing wrong?
This gschemrc is being read OK as any syntax errors in it stop the program launch.
I am using the dark theme, is this relevant?
Dark blue net on black b/g is impossible to use for me.

Thanks,
Barry

vzh, 2020-10-10:
Hi Barry,
Could you please provide us with the contents of your files `gschemrc` and `colours` or just attach them here?  What value have you changed and how?

barjac, 2020-10-10:
Hi,
This is the colors file content:
`[baz@jackodesktop Geda]$ cat colors
(display-color-map '(
  (background           "#000000")
  (pin                  "#ffffff")
  (net-endpoint         "#ff0000")
  (graphic              "#00ff00")
  (net                  "#97ff05")
  (attribute            "#ffff00")
  (logic-bubble         "#00ffff")
  (dots-grid            "#bebebe")
  (detached-attribute   "#ff0000")
  (text                 "#00ff00")
  (bus                  "#00ff00")
  (select               "#ffa500")
  (bounding-box         "#ffa500")
  (zoom-box             "#00ffff")
  (stroke               "#e5e5e5")
  (lock                 "#bebebe")
  (output-background    #f)
  (freestyle1           "#308a8e")
  (freestyle2           "#d02f2f")
  (freestyle3           "#b93ac1")
  (freestyle4           "#cccccc")
  (junction             "#ffff00")
  (mesh-grid-major      "#1e1e1e")
  (mesh-grid-minor      "#171717")
))

(display-outline-color-map '(
  (background           #f)
  (pin                  "#ababab")
  (net-endpoint         "#00ab00")
  (graphic              "#00ab00")
  (net                  "#97ff05")
  (attribute            "#abab00")
  (logic-bubble         "#00abab")
  (dots-grid            #f)
  (detached-attribute   "#ab0000")
  (text                 "#00ab00")
  (bus                  "#00ab00")
  (select               "#ffa500")
  (bounding-box         "#ffa500")
  (zoom-box             "#00ffff")
  (stroke               "#e5e5e5")
  (lock                 "#a9a9a9")
  (output-background    #f)
  (freestyle1           "#308a8e")
  (freestyle2           "#d02f2f")
  (freestyle3           "#b93ac1")
  (freestyle4           "#cccccc")
  (junction             "#abab00")
  (mesh-grid-major      #f)
  (mesh-grid-minor      #f)
))

[baz@jackodesktop Geda]$ 
`
...and this is the content of ~/.config/lepton-eda/gschemrc (only the one line, with line feed after):
primitive-load "/home/baz/Geda/colors"

I used the colour picker from the gui 'view' menu to change only the 'net' colour, and saved it to 'colors'.

Hope that helps,
Barry

vzh, 2020-10-10:
Barry,
IIUC, your call for `primitive-load` is without parentheses?
If so, Guile (and other Scheme interpreters) will interpret the code as two expressions, one of which returns a function named `primitive-load`, and the next one is just a string and returns just the string. So the interpreter won't even warn you since anything looks and interprets well.
What you need then to make a function call of those two expressions is to enclose them in parens, since a call for any Scheme function looks like this: `(function-name arg1 arg2 ...)`.
Therefore just use: `(primitive-load "/home/baz/Geda/colors")`

HTH

barjac, 2020-10-10:
Thanks! I will do that, but it is not clear from the message in the color-chooser that the parenthesis is part of the text to be used. 
I am familiar with function call syntax like somefunction(arg1 arg2), and if that had been the case then it would have been obvious to me.
Maybe you could add "including the parenthesis/brackets" to that text. Otherwise it looks like an alternative way of quoting.

Works perfectly!
Cheers,
Barry

closed 2020-10-10 by barjac

vzh, 2020-10-10:
Well, **Lisp**s (and one of their dialect named Scheme) have this syntax for decades (since about 1960s).  You just need to have used to it :-)  It's straightforward, since functions and data have the same syntax.  You have just to remember to use (func-name args) instead of C's func-name(args).
If you feel like it should be reflected in the color chooser dialog, please open another issue, hopefully we'll discuss what text would be the most intuitive for the users then.