lepton-eda/lepton-eda #690: Git master build fails with -Wl,--no-undefined

wally-mageia, 2020-10-15:
Git master build fails with -Wl,--no-undefined added to LDFLAGS.

```
make[2]: Entering directory '/builddir/build/BUILD/lepton-eda-1.9.11/libleptonrenderer'
/bin/sh ../libtool --silent  --tag=CC   --mode=link x86_64-mageia-linux-gnu-gcc -Wall -Werror=uninitialized -std=c99   -I/usr/include/guile/2.0 -pthread  -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include  -I/usr/include/gtk-2.0 -I/usr/lib64/gtk-2.0/include -I/usr/include/pango-1.0 -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/harfbuzz -I/usr/include/freetype2 -I/usr/include/libpng16 -I/usr/include/libmount -I/usr/include/blkid -I/usr/include/fribidi -I/usr/include/libxml2 -I/usr/include/cairo -I/usr/include/pixman-1 -I/usr/include/gdk-pixbuf-2.0 -pthread  -I/usr/include/cairo -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/pixman-1 -I/usr/include/freetype2 -I/usr/include/libpng16 -I/usr/include/libxml2  -I/usr/include/pango-1.0 -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/harfbuzz -I/usr/include/freetype2 -I/usr/include/libpng16 -pthread -I/usr/include/libmount -I/usr/include/blkid -I/usr/include/fribidi -I/usr/include/libxml2 -I/usr/include/cairo -I/usr/include/pixman-1  -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -pthread -I/usr/include/libmount -I/usr/include/blkid  -O2 -g -pipe -Wformat -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fstack-protector --param=ssp-buffer-size=4 -fasynchronous-unwind-tables -version-info 3:0:0   -lguile-2.0 -lgc   -lgdk-x11-2.0 -lpangocairo-1.0 -lpango-1.0 -lharfbuzz -lcairo -lgdk_pixbuf-2.0 -lgobject-2.0 -lglib-2.0  -lglib-2.0  -lcairo  -lpango-1.0 -lgobject-2.0 -lglib-2.0 -lharfbuzz  -lgdk_pixbuf-2.0 -lgobject-2.0 -lglib-2.0  -Wl,--as-needed -Wl,--no-undefined -Wl,-z,relro -Wl,-O1 -Wl,--build-id -Wl,--enable-new-dtags -o libleptonrenderer.la -rpath /usr/lib64 edacairo.lo edapangorenderer.lo edarenderer.lo  -lm 
/usr/bin/ld: .libs/edarenderer.o: in function `eda_renderer_prepare_text':
/builddir/build/BUILD/lepton-eda-1.9.11/libleptonrenderer/edarenderer.c:864: undefined reference to `geda_text_object_get_size_in_points'
/usr/bin/ld: .libs/edarenderer.o: in function `eda_renderer_calc_text_position':
/builddir/build/BUILD/lepton-eda-1.9.11/libleptonrenderer/edarenderer.c:953: undefined reference to `geda_text_object_get_angle'
/usr/bin/ld: /builddir/build/BUILD/lepton-eda-1.9.11/libleptonrenderer/edarenderer.c:958: undefined reference to `geda_text_object_get_alignment'
/usr/bin/ld: .libs/edarenderer.o: in function `eda_renderer_prepare_text':
/builddir/build/BUILD/lepton-eda-1.9.11/libleptonrenderer/edarenderer.c:903: undefined reference to `geda_text_object_get_angle'
/usr/bin/ld: .libs/edarenderer.o: in function `eda_renderer_draw_hatch':
/builddir/build/BUILD/lepton-eda-1.9.11/libleptonrenderer/edarenderer.c:584: undefined reference to `m_hatch_path'
/usr/bin/ld: /builddir/build/BUILD/lepton-eda-1.9.11/libleptonrenderer/edarenderer.c:606: undefined reference to `geda_fill_type_draw_first_hatch'
/usr/bin/ld: /builddir/build/BUILD/lepton-eda-1.9.11/libleptonrenderer/edarenderer.c:609: undefined reference to `geda_fill_type_draw_second_hatch'
/usr/bin/ld: /builddir/build/BUILD/lepton-eda-1.9.11/libleptonrenderer/edarenderer.c:576: undefined reference to `m_hatch_box'
/usr/bin/ld: /builddir/build/BUILD/lepton-eda-1.9.11/libleptonrenderer/edarenderer.c:580: undefined reference to `m_hatch_circle'
/usr/bin/ld: .libs/edarenderer.o: in function `eda_renderer_draw_circle':
/builddir/build/BUILD/lepton-eda-1.9.11/libleptonrenderer/edarenderer.c:745: undefined reference to `geda_circle_object_get_radius'
/usr/bin/ld: /builddir/build/BUILD/lepton-eda-1.9.11/libleptonrenderer/edarenderer.c:744: undefined reference to `geda_circle_object_get_center_y'
/usr/bin/ld: /builddir/build/BUILD/lepton-eda-1.9.11/libleptonrenderer/edarenderer.c:743: undefined reference to `geda_circle_object_get_center_x'
/usr/bin/ld: .libs/edarenderer.o: in function `eda_renderer_draw_arc':
/builddir/build/BUILD/lepton-eda-1.9.11/libleptonrenderer/edarenderer.c:722: undefined reference to `geda_arc_object_get_sweep_angle'
/usr/bin/ld: /builddir/build/BUILD/lepton-eda-1.9.11/libleptonrenderer/edarenderer.c:721: undefined reference to `geda_arc_object_get_start_angle'
/usr/bin/ld: /builddir/build/BUILD/lepton-eda-1.9.11/libleptonrenderer/edarenderer.c:720: undefined reference to `geda_arc_object_get_radius'
/usr/bin/ld: /builddir/build/BUILD/lepton-eda-1.9.11/libleptonrenderer/edarenderer.c:719: undefined reference to `geda_arc_object_get_center_y'
/usr/bin/ld: /builddir/build/BUILD/lepton-eda-1.9.11/libleptonrenderer/edarenderer.c:718: undefined reference to `geda_arc_object_get_center_x'
/usr/bin/ld: .libs/edarenderer.o: in function `eda_renderer_draw_pin':
/builddir/build/BUILD/lepton-eda-1.9.11/libleptonrenderer/edarenderer.c:681: undefined reference to `geda_pin_object_get_width'
/usr/bin/ld: .libs/edarenderer.o: in function `eda_renderer_is_drawable':
/builddir/build/BUILD/lepton-eda-1.9.11/libleptonrenderer/edarenderer.c:556: undefined reference to `geda_object_get_drawing_color'
/usr/bin/ld: .libs/edarenderer.o: in function `eda_renderer_default_draw':
/builddir/build/BUILD/lepton-eda-1.9.11/libleptonrenderer/edarenderer.c:522: undefined reference to `geda_object_get_drawing_color'
/usr/bin/ld: .libs/edarenderer.o: in function `eda_renderer_is_drawable':
/builddir/build/BUILD/lepton-eda-1.9.11/libleptonrenderer/edarenderer.c:556: undefined reference to `geda_object_get_drawing_color'
/usr/bin/ld: .libs/edarenderer.o: in function `eda_renderer_default_draw_grips':
/builddir/build/BUILD/lepton-eda-1.9.11/libleptonrenderer/edarenderer.c:1101: undefined reference to `geda_circle_object_get_center_y'
/usr/bin/ld: /builddir/build/BUILD/lepton-eda-1.9.11/libleptonrenderer/edarenderer.c:1101: undefined reference to `geda_circle_object_get_radius'
/usr/bin/ld: /builddir/build/BUILD/lepton-eda-1.9.11/libleptonrenderer/edarenderer.c:1100: undefined reference to `geda_circle_object_get_center_x'
/usr/bin/ld: /builddir/build/BUILD/lepton-eda-1.9.11/libleptonrenderer/edarenderer.c:1100: undefined reference to `geda_circle_object_get_radius'
/usr/bin/ld: .libs/edarenderer.o: in function `eda_renderer_draw_arc_grips':
/builddir/build/BUILD/lepton-eda-1.9.11/libleptonrenderer/edarenderer.c:1183: undefined reference to `geda_arc_object_get_center_x'
/usr/bin/ld: /builddir/build/BUILD/lepton-eda-1.9.11/libleptonrenderer/edarenderer.c:1184: undefined reference to `geda_arc_object_get_center_y'
/usr/bin/ld: /builddir/build/BUILD/lepton-eda-1.9.11/libleptonrenderer/edarenderer.c:1186: undefined reference to `geda_arc_object_get_radius'
/usr/bin/ld: /builddir/build/BUILD/lepton-eda-1.9.11/libleptonrenderer/edarenderer.c:1187: undefined reference to `geda_arc_object_get_start_angle'
/usr/bin/ld: /builddir/build/BUILD/lepton-eda-1.9.11/libleptonrenderer/edarenderer.c:1188: undefined reference to `geda_arc_object_get_sweep_angle'
/usr/bin/ld: .libs/edarenderer.o: in function `eda_renderer_get_text_user_bounds':
/builddir/build/BUILD/lepton-eda-1.9.11/libleptonrenderer/edarenderer.c:1520: undefined reference to `eda_config_get_context_for_path'
/usr/bin/ld: /builddir/build/BUILD/lepton-eda-1.9.11/libleptonrenderer/edarenderer.c:1521: undefined reference to `eda_config_get_string'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:560: libleptonrenderer.la] Error 1
make[2]: Leaving directory '/builddir/build/BUILD/lepton-eda-1.9.11/libleptonrenderer'
make[1]: *** [Makefile:571: all-recursive] Error 1
make: *** [Makefile:503: all] Error 2
```

wally-mageia, 2020-10-15:
It looks like `libleptonrenderer` needs symbols from `liblepton` and vice versa.

`liblepton` can't be built without building `libleptonrenderer.la` first, but it's impossible because of undefined symbols available only in `liblepton` .

wally-mageia, 2020-10-15:
I can continue `liblepton` build if I remove the build dependency to `libleptonrenderer.la` and add `$(GIO_LIBS)` to `liblepton_la_LIBADD`, but then the build fails with:
 ```
  CCLD     liblepton.la
/usr/bin/ld: .libs/liblepton_la-geda_text_object.o: in function `geda_text_object_calculate_bounds':
/builddir/build/BUILD/lepton-eda-1.9.11/liblepton/src/geda_text_object.c:102: undefined reference to `eda_renderer_get_text_user_bounds'
collect2: error: ld returned 1 exit status
```

vzh, 2020-10-15:
Yeah, the two libraries, `liblepton` and `libleptonrenderer`, are interdependent.
What one should depend on another one, WDYT?

wally-mageia, 2020-10-16:
> Yeah, the two libraries, `liblepton` and `libleptonrenderer`, are interdependent.
> What one should depend on another one, WDYT?

Unfortunately I don't have that kind of knowledge about these two libraries itself.

I'm helping to include `lepton-eda` pkg to Mageia distribution and we build our packages using `-Wl,--no-undefined` linker option by default. It can be disabled in build time, but IMO interdependent libraries should be either split out into a third library that both can be linked against or merged into a single library. I don't know the exact technical reasons in this case, though.

graahnul-grom, 2020-11-23:
> What one should depend on another one, WDYT?

No one.
I think it is become evident that both libraries
(their code) are tightly coupled, one cannot be
used without the other.
But we try to artificially separate what
is, in fact, a single entity, and, sure enough, it
leads to all sorts of problems. `libleptonrenderer`
should be merged into `liblepton`, that would be a
natural design decision. As a bonus, it will simplify
our build system and who knows, maybe we'll be able
to repair a broken `Cygwin` build then.

It's not that hard, just did it (the ugly way,
without any git commits :-)).

graahnul-grom, 2020-11-28:
> `libleptonrenderer` should be merged into `liblepton`

Yes! Not only it fixes this issue,
it makes Lepton build under `cygwin` again!
And without [acrobatics](https://github.com/lepton-eda/lepton-eda/wiki/Cygwin), just simple commands:
```
# ./configure --prefix=/usr/local
# make LDFLAGS="-no-undefined -lgio-2.0 -lgobject-2.0"
# make LDFLAGS="-no-undefined" install
```

NLS works. Though there is no `gtkextra` package in cygwin,
it can easily be compiled from sources, so, one can build
`lepton-attrib`, too (set `PKG_CONFIG_PATH=/usr/local/lib/pkgconfig`
if `gtkextra` is installed in `/usr/local`).

You can build it, but you can't run it, thanks to bloody FFI
and errors PR #700 should fix.
It is solved by substituting libraries' names.

Find them:
```
# git grep dynamic-link
liblepton/scheme/lepton/ffi.scm:  (dynamic-link (or (getenv "LIBLEPTON") "liblepton")))
liblepton/scheme/lepton/log.scm:(define libglib (dynamic-link "libglib-2.0"))
libleptonattrib/lepton-attrib.scm:(define libgtk (dynamic-link "@LIBGTK@"))
libleptonattrib/lepton-attrib.scm:(define libleptonattrib (dynamic-link "libleptonattrib"))
libleptongui/scheme/schematic/ffi.scm:  (dynamic-link (or (getenv "LIBLEPTONGUI") "libleptongui")))
libleptongui/scheme/schematic/ffi/gtk.scm:(define libgtk (dynamic-link "libgtk-x11-2.0"))
```

Change them:
```
liblepton       => cyglepton-6
libleptongui    => cygleptongui-1
libleptonattrib => cygleptonattrib-1
libglib-2.0     => cygglib-2.0-0
libgtk-x11-2.0  => cyggtk-x11-2.0-0
```

ура:
[image]

graahnul-grom, 2020-11-29:
@wally-mageia Please check if changes in #701
fixes the build with `-Wl,--no-undefined`.

wally-mageia, 2020-11-29:
> @wally-mageia Please check if changes in #701
> fixes the build with `-Wl,--no-undefined`.

Yes, #701 fixes the issue. Build succeeds with `-Wl,--no-undefined`.

graahnul-grom, 2020-11-29:
@wally-mageia Thank you.

PR #701 (2020-12-02): Merge libleptonrenderer into liblepton
Source code of either library cannot be used without
the other's library code. Both are strongly interdependent.
Artificial "separation" as it is done now, makes no sense
whatsoever. Sure enough, such a "brilliant" design leads
to nothing, but problems.

Since release 1.9.10 it's impossible to build Lepton under
`Cygwin`. Eliminating unnatural separation of code allows
us to [repair](https://github.com/lepton-eda/lepton-eda/issues/690#issuecomment-735249953) a broken `Cygwin` build.

Packagers report that the above nonsense breaks build
with certain flags [... cut]
closed 2020-12-02 by graahnul-grom