lepton-eda/lepton-eda #696: Cannot execute actions via lepton-schematic command line (-c)

graahnul-grom, 2020-11-22:
For example (the same result with other actions):

```
# lepton-schematic -c '(&file-quit)'
ERROR: Found invalid lepton-schematic window smob #f
Please see log file for more information.
```

In the log window:
```
Backtrace:
          10 (primitive-load "/home/dmn/lepton/bin.tb/bin/lepton-sch…")
In ice-9/boot-9.scm:
    829:9  9 (catch #t #<procedure 80e957760 at ice-9/eval.scm:330:…> …)
In unknown file:
           8 (eval (begin (eval-string "(&file-quit)")) #<directory …>)
           7 (eval-string "(&file-quit)" #<undefined>)
In ice-9/boot-9.scm:
   2312:4  6 (save-module-excursion #<procedure 80e966380 at ice-9/e…>)
In ice-9/eval-string.scm:
     38:6  5 (read-and-eval #<input: string 80eb78070> #:lang _)
In ice-9/eval.scm:
    619:8  4 (_ #(#(#<directory (schematic action) 80b8d05a0> # …) #))
In unknown file:
           3 (%file-quit)
In ice-9/boot-9.scm:
   751:25  2 (dispatch-exception 0 misc-error (#f "Found invalid …" …))
In ice-9/eval.scm:
   626:19  1 (_ #(#(#<directory (lepton eval) 802f048c0> #<vari…>) …))
In unknown file:
           0 (make-stack #t)

Found invalid lepton-schematic window smob #f
```

It works in 1.9.11 release, but is broken in the master branch.

vzh, 2020-11-22:
Yeah, in the new code, window is not initialized if you use the `-c` option.  IIRC, lepton-schematic always exited after executing commands specified with `-c`.
Do you have any real life example where you would need using of actions without opening the main lepton-schematic window?

graahnul-grom, 2020-11-22:
> lepton-schematic always exited after executing commands specified with -c

That's not correct. Please try `-c` with 1.9.11 release.

graahnul-grom, 2020-11-22:
`-c` is a convenient way to make lepton-schematic execute
some guile script on startup.

vzh, 2020-11-22:
Ah, OK, I see.  Sorry for the noise.

PR #712 (2020-12-10): Fix lepton-schematic option processing
Closes #696 
Closes #704
closed 2020-12-10 by vzh