lepton-eda/lepton-eda #715: Lepton manual files on case-insensitive file systems

graahnul-grom, 2020-12-10:
There are two generated `html` files:
- `index.html` (lower-case "i"): table of contents (start page)
- `Index.html` (upper-case "i"): index page

On filesystems that do not distinguish lower- and
upper-case file names, `make` overwrites one or
the other file. Thus, on Windows/Cygwin, for example,
the table of contents is lost.
In the Scheme manual, there are "concept index",
"variable index" and "function index", no problems.
The "Index" page should be renamed somehow.

graahnul-grom, 2020-12-10:
It's possible, but what name should we choose
instead of just "Index"?

```diff
diff --git a/docs/manual/geda-file-format-spec.texi b/docs/manual/geda-file-format-spec.texi
index 308ea3229..fb4848a0e 100644
--- a/docs/manual/geda-file-format-spec.texi
+++ b/docs/manual/geda-file-format-spec.texi
@@ -1,4 +1,4 @@
-@node gEDA file format, Index, Development, Top
+@node gEDA file format, Eklmn Index, Development, Top
 @chapter gEDA/gaf File Format Document
 @cindex gEDA file format
 
diff --git a/docs/manual/index.texi b/docs/manual/index.texi
index e244e6554..08f0db494 100644
--- a/docs/manual/index.texi
+++ b/docs/manual/index.texi
@@ -1,3 +1,3 @@
-@node Index, , gEDA file format, Top
-@unnumbered Index
+@node Eklmn Index, , gEDA file format, Top
+@unnumbered Eklmn Index
 @printindex cp
diff --git a/docs/manual/lepton-manual.texi b/docs/manual/lepton-manual.texi
index c195af363..f424ea8e2 100644
--- a/docs/manual/lepton-manual.texi
+++ b/docs/manual/lepton-manual.texi
@@ -52,7 +52,7 @@ Documentation License.''
 * Communication::               Places for discussions and chatting.
 * Development::                 Useful information for developers.
 * gEDA file format::            File format currently in use.
-* Index::
+* Eklmn Index::
 
 @detailmenu
  --- The Detailed Node Listing ---
```

graahnul-grom, 2020-12-10:
With the example patch above, the index topic is displayed as
"Eklmn Index", and the generated file name is "Eklmn-Index.html".

vzh, 2020-12-10:
Since all those index entries are generated from `@cindex`, it should be **Concept index** I believe.

graahnul-grom, 2020-12-10:
Could we say "Topic Index"?

vzh, 2020-12-10:
Please do, if you want.

PR #716 (2020-12-10): Rename the index file in the Lepton EDA Reference Manual
Generated HTML files for index and the table
of contents have similar names: only case of
the first letter is different ("Index.html"
and "index.html"). On file systems that do not
distinguish lower- and upper-case file names,
"make" overwrites one or the other file.
Fix it by using a new name: "Topic Index".

Closes #715.
closed 2020-12-10 by vzh