lepton-eda/lepton-eda #718: lepton-netlist produces strange output

386th, 2020-12-20:
Good day everyone!

lepton-netlist produces strange output for me and does nothing useful. Backtrace follows. This behavior can be solved by reinstalling cad/lepton-eda port, then lepton-netlist works just fine UNTIL lepton-schematic launched and eventually being closed. Then it again starts spamming with this text until reinstalled.

FreeBSD box 12.1-RELEASE-p5 FreeBSD 12.1-RELEASE-p5 BOX  amd64

lepton-eda-1.9.11 (installed from ports)

---

Backtrace:
In ice-9/threads.scm:
    390:8 19 (_ _)
In ice-9/boot-9.scm:
  2994:20 18 (_)
   2312:4 17 (save-module-excursion #<procedure 8020e4600 at ice-9/b…>)
  3014:26 16 (_)
In unknown file:
          15 (primitive-load-path "netlist/config" #<procedure 80177…>)
In netlist/config.scm:
     19:0 14 (_)
In ice-9/boot-9.scm:
   2874:4 13 (define-module* _ #:filename _ #:pure _ #:version _ # _ …)
  2887:24 12 (_)
   222:29 11 (map1 (((ice-9 format)) ((ice-9 match)) ((lepton #)) # …))
   222:29 10 (map1 (((ice-9 match)) ((lepton config)) ((lepton #)) #))
   222:17  9 (map1 (((lepton config)) ((lepton log)) ((netlist #))))
  2800:17  8 (resolve-interface (lepton config) #:select _ #:hide _ # …)
In ice-9/threads.scm:
    390:8  7 (_ _)
In ice-9/boot-9.scm:
  2726:13  6 (_)
In ice-9/threads.scm:
    390:8  5 (_ _)
In ice-9/boot-9.scm:
  2994:20  4 (_)
   2312:4  3 (save-module-excursion #<procedure 8020e45a0 at ice-9/b…>)
  3014:26  2 (_)
In unknown file:
           1 (primitive-load-path "lepton/config" #<procedure 80177e…>)
In lepton/config.scm:
     86:0  0 (_)

lepton/config.scm:86:0: In procedure module-lookup: Unbound variable: %config-legacy-mode?

vzh, 2020-12-21:
Can it be that there are several `lepton/config.scm` installed on your system, and different programs use different files?  On Debian, I would check it using `locate lepton/config.scm`, though I'm not sure if that utility exists on FreeBSD.  The line with `%config-legacy-mode` is the last line in that file.  Please check if it exists.

Please also check your `gafrc` and `gschemrc` files if the `config-legacy-mode?` is used somewhere in them.  Lepton does no longer use it.  (Probably it's the time to actually remove it.)

graahnul-grom, 2020-12-21:
Hi 386th!
From the outset, I strongly descourge you from using FreeBSD 12.x.
It's a disaster. So lucky for the number: 12.
If you follow the FreeBSD mailing lists (and you should),
you may notice the tremendous amount of complaints,
I haven't seen that much since 8.0 was out (that was a time
I abandoned the Linux idiocity).
As for your problem, the first thing I would do: delete the Guile cache folder: `rm -rrvf ~/.cache/guile/`.
In fact, I elude that kind of problems by setting the `$GUILE_AUTO_COMPILE` env var to `0`.
It makes things go a bit little slower,
but at least, кошмарные DBCS-ники 
не будут сниться тебе по ночам.
Take care.
And happy New Year, comrade.

graahnul-grom, 2020-12-21:
@386th Being my FreeBSD fellow, could you please
try to install the new version of Lepton EDA
https://github.com/graahnul-grom/freebsd-lepton-eda/tree/release1913

386th, 2020-12-21:
Thank you guys!

Commenting out "(define-public config-legacy-mode? %config-legacy-mode?)" in lepton/config.scm solves this problem. I will test this more thoroughly over the weekend. All my custom settings are stored in **lepton-user.conf**, **gafrc** only contains **reset-component-library** and **component-library-search**, **gschemrc** does not exist at all.

One day I would like to see lepton-eda free of that scheme and guile. According to my experience, pure C or C++ software works much more predictable and stable.

@graahnul-grom I've been using FreeBSD since 6.2, and despite possible bugs it fits all my needs since then. Короче, Linux не осилил.

Happy new year!

graahnul-grom, 2020-12-24:
@386th Unfortunately, I can't reproduce this on my system
(FreeBSD 11.4-STABLE amd64 r367719). The next step
should be to download and install exactly the same
FreeBSD version in VirtualBox (12.1-RELEASE-p5). I hope
I'll have enough time to do it on this weekend.

The latest lepton-eda release 1.9.13 has been committed
to the ports tree (just yesterday, binary packages should
be available within several days). Could you please try
to upgrade and check if it solves your problem?

graahnul-grom, 2020-12-24:
@vzh `config-legacy-mode?()` is the companion predicate
procedure of `config-set-legacy-mode!()`, which is used
by the `lepton-upcfg` configuration upgrade utility.
I'm not sure if we should remove it yet.

vzh, 2020-12-24:
@graahnul-grom, I'm sorry, I thought it was some pre-fork stuff related to disabling legacy gnetlist.

graahnul-grom, 2020-12-27:
@386th, I have installed `FreeBSD 12.1-RELEASE amd64` and
updated it with `freebsd-update`: the current patch level
is `12`, so now it is `12.1-RELEASE-p12` (not `5`).
Whatever I do with `lepton-netlist` and `lepton-schematic`
from the `lepton-eda-1.9.11` package, I cannot reproduce
the error.
The strange thing that in the backtrace output you provided
above, the location reported is `lepton/config.scm:86:0`.
But `config-legacy-mode?()` is defined on line 87!
Could it be that the package was corrupted somehow? Didn't
you try to verify it with `pkg check --checksums lepton-eda`?
It would be helpful to check if this error could be reproduced
in clean environment (e.g. under a new user account).

386th, 2020-12-27:
@graahnul-grom, '_pkg check --checksums lepton-eda_' shows that lepton/config.scm file is damaged, and this is normal, since line 87 is commented out by me for last 7 days. After uncommenting back this line, the check passes without errors, so package is not corrupted.

Lines 86 and 87 from my lepton/config.scm:
config.scm line 86: (define-public config-set-legacy-mode! %config-set-legacy-mode!)
config.scm line 87: (define-public config-legacy-mode? %config-legacy-mode?)

After I uncommented line 87, lepton-netlist started giving error again.

Then I cleared guile cache and then lepton-netlist runs without any errors (but first startup took about a minute or even more).

I then run lepton-netlist directly and sch2pcb for several times, everything worked well. Until I run lepton-schematic. Immediately after lepton-schematic has started, lepton-netlist stops working (shows that error from #1)...

I don't know what it is or how to reproduce this error on another computer. Lepton-eda package installed on another computer works fine. On this computer I just commented out line 87 in lepton/config.scm and keep working, there are no errors anymore.

Some time ago, maybe a year or more, I compiled lepton-eda from source manually, since version in ports collection was outdated at a time, perhaps it somehow influenced.

I'll try latest version of lepton-eda some days later, then I'll write here what has changed.

vzh, 2020-12-28:
@386th, can it be that you have some scheme `.scm` (or compiled scheme `.go`) files from your manual installation somewhere in `/usr/local`, or in whatever directory you used for installation? Please check this.

graahnul-grom, 2020-12-28:
Yes, it's possible that there are some remnants of prior
installations. As it was suggested by @vzh, I would run
`/usr/libexec/locate.updatedb` and use `locate(1)` to
find suspicious `.scm` and `.go` files. Another places to
look at are `PATH` and `LD_LIBRARY_PATH` environment
variables, `libmap.conf` configuration files, paths printed
by `/etc/rc.d/ldconfig restart` and output of `ldconfig -r`.

Since Guile release 2.2 compilation times became intolerable
(though they claim that execution now should be faster).
Since then I do not use Guile cache at all, except the one
I create with `make precompile` for the installation
which I use for electronics (schematic capture, simulation).
In all other cases (development, testing) I turn caching off
by setting `GUILE_AUTO_COMPILE=0` environment variable.

I think it's worth trying to delete `~/.cache/guile/`, then
set `GUILE_AUTO_COMPILE=0` in environment and run tools,
trying to trigger the error.

vzh, 2020-12-28:
Another hint is to use built-in Guile functionality to search for modules used.  The following command:
> lepton-schematic -c '(display %load-path) (newline) (display %load-compiled-path) (newline) (primitive-exit 0)'
will output paths where modules are searched for `lepton-schematic`.

The same command can be used for `lepton-netlist`, just replace `lepton-schematic` with `lepton-netlist`.

graahnul-grom, 2020-12-29:
It is unlikely, but just to be sure: please check the options the
`lang/guile2` port was compiled with (IIUC, you are using ports,
not a binary packages):

```
$ make -C /usr/ports/lang/guile2/ showconfig
===> The following configuration options are available for guile2-2.2.7_1:
     NLS=on: Native Language Support
     THREADS=on: Threading support
===> Use 'make config' to modify these settings
```

It should be built with `THREADS=on`.

vzh, 2021-03-26:
Any news here?  I'm going to close the issue.

graahnul-grom, 2021-03-26:
@386th, do you still have problems with `lepton-netlist`?
And let me ask you a couple of questions:
1. Did you try the [latest release](https://github.com/lepton-eda/lepton-eda/releases/tag/1.9.13-20201211) (1.9.13) or build from git `master`?
(lepton-eda 1.9.13 is in the FreeBSD ports collection, binary
packages are also available).
2. BTW, FreeBSD 12.1 is [EoL](https://www.freebsd.org/security/unsupported/) as of January 31, 2021,
did you upgrade the system?
Thanks.

closed 2021-05-17 by vzh