lepton-eda/lepton-eda #719: lepton-sch2pcb does not seem to understand pcb-rnd parametric footprints

coldelectrons, 2020-12-20:
Looking in /usr/share/pcb-rnd/pcblib/parametric, I want to use 'connector' for a basic 40-pin header.

The documentation says to use:
```
connector(40, 2)
```
as the footprint.

lepton-sch2pcb complains and does not include the element:
```
> lepton-sch2pcb -d /usr/share/pcb-rnd/pcblib 1.sch

Warning: argument passing may have been confused by
         a comma in a component value:
         Check (connector(40 40 -2
         Maybe just use a space instead of a comma?
(connector(40: can't find PCB element for footprint 40 (value=-2)
So device (connector(40 will not be in the layout.
```

vzh, 2020-12-21:
@coldelectrons,
AFAIK, `gsch2pcb` was designed (long ago) to support `pcb`, not `pcb-rnd`, devs of which invented parametric footprints after the fork.  So there's nothing strange here.  `pcb-rnd` is supported by means of the `tEDAx` backend.

I may be wrong.  There are people using Lepton with pcb-rnd who may shed some light on this.

@bdale, did you ever use parametric footprints with Lepton?

barjac, 2020-12-21:
Just import your foo.sch created in lepton-eda directly into pcb-rnd using the import -> configure import schematics -> format: gnetlist. Navigate to your foo.sch and hit import button.
N.B. It is easier to use the Window -> Library option to find the 'Parametric' folder and create your footprint in the tool provided. There is a button to edit the parameters while it displays the resulting footprint. You can then simply copy the footprint string and paste it into lepton-eda.

coldelectrons, 2020-12-21:
Thank you. I'm getting back into certain hobbies after almost a decade, and I find I have preconceived notions about what current software can do.

closed 2020-12-21 by coldelectrons