lepton-eda/lepton-eda #732: Organize the repository

graahnul-grom, 2021-01-30:
Now that the dust has settled, we have time to look at the
repository layout. It's far from perfect. "Inconsistency"
is the keyword here.

Let me propose the following changes:

- Move `lepton-cli` source code into a hierarchy similar to
that of the other components: `cli/src/`, `cli/docs/`, etc.

- Do not mix symbols with other stuff. Split up the contents
of the symbols directory, e.g.:
  symbols/
  |-- docs/
  |-- scheme/
  |-- sym/

BTW, do we really need the `nc.doc` satiric document?

- Remove the `docs/toplevel` directory with the only file
which contains the only link to the wiki main page.

- Create `docs/news/` and move `NEWS-1.9.2.txt` there.

- Move `libleptonattrib/design` to `docs/`.

- Clean up the mess in `utils/`. All that utils/scripts
separation is obsolete. We do not have "utils" or "scripts",
we have the *components* of the suite, which are installed
into the system `bin/` and `lib/` directories.
Here they are, in no particular order:

1) liblepton
2) libleptongui
3) libleptonattrib
4) netlist
5) schematic
6) cli
7) attrib
8) tragesym
9) sch2pcb
10) archive
11) symcheck
12) symfix
13) embed
14) schdiff
15) pcb_backannotate
16) renum
17) refdes_renum
18) upcfg

Eighteen. Not 100, not 50. I believe each of the above
should live in its own directory, even if it's under
`utils/` (and we have to decide what we consider a "utility").

- Consider moving `lepton-renum` to `contrib/`. It does exactly
the same as `lepton-refdes_renum`, IIUC.

- Rename all `geda-*` files under `m4/` to `lepton-*`.

@vzh Perhaps, I missed something, but why parts of the `netlist`
and `symcheck` Scheme code are now in the `liblepton/scheme` subdir?

vzh, 2021-02-02:
IMO, *utility* is a useful tool for solving of a particular issue, not main, usually CLI.

vzh, 2021-02-02:
All Scheme modules can be used by many tools in the suite, e.g. `lepton-schematic` uses `lepton`, `netlist`, and `symcheck` modules.  That is why they are in the same directory.  It facilitates unit-testing as well.  I would move that directory to the root of the repository (as Guile itself has the `module/` directory). However, we also have so named `core` modules.  I would prefer to get rid of them first.

vzh, 2021-02-02:
I second moving utils into own directories in `utils/` with all related docs, tests, examples, etc.

vzh, 2021-02-02:
Too many problems in one issue.  I would prefer to see them in different issues.  Otherwise we never fix them :-)

graahnul-grom, 2021-02-02:
> IMO, utility is a useful tool for solving of a particular issue, not main, usually CLI.

Agree. All things, but lepton-schematic, are utilities, then? :-)

> Too many problems in one issue.

Absolutely. What should we move to separate issues, WDYT?
And what could we already start working on?

vzh, 2021-02-02:
> > IMO, utility is a useful tool for solving of a particular issue, not main, usually CLI.
> 
> Agree. All things, but lepton-schematic, are utilities, then? :-)

Well, yes, I believe :-)
 
> > Too many problems in one issue.
> 
> Absolutely. What should we move to separate issues, WDYT?

Almost any paragraph in your first message here starting with dot :-)

> And what could we already start working on?

graahnul-grom, 2021-02-11:
[utils/README](https://github.com/lepton-eda/lepton-eda/blob/master/utils/README) mostly contains information about utilities in `contrib/`.
Other programs are documented in man pages, their authors are
also mentioned there.
Could we reduce this file to contain info only on utils in `contrib/`
and move it there?

graahnul-grom, 2021-02-11:
There are other quite old/obsolete files, e.g.:
- `liblepton/HACKING`
- `libleptonattrib/NOTES`
- `libleptonattrib/ToDos`
- `libleptongui/TODO`
- `netlist/BUGS`

I would consider at least moving them to corresponding
`docs/` directories.

graahnul-grom, 2021-02-11:
And I have serious doubts in usefulness of `ChangeLog`
files we generate during build (taking up space and time):

1) We maintain the `NEWS.md` file
2) All history is available in `git`
3) All history is available on `github.com` web site

vzh, 2021-02-12:
> [utils/README](https://github.com/lepton-eda/lepton-eda/blob/master/utils/README) mostly contains information about utilities in `contrib/`.
> Other programs are documented in man pages, their authors [...]

Yep, though a small README file with info on how to get better info on the utilities would not harm :-)  As well as sections in the manual for each utility.

vzh, 2021-02-12:
> There are other quite old/obsolete files, e.g.:
> 
>     * `liblepton/HACKING`
> 
>     * `libleptonattrib/NOTES`
> 
>     * `libleptonattrib/ToDos`
> 
>     * `libleptongui/TODO`
> 
>     * `netlist/BUGS`
> 
> 
> I would consider at least moving them to corresponding
> `docs/` directories.

Please do. (It would be nice to revise them some day.)

vzh, 2021-02-12:
> And I have serious doubts in usefulness of `ChangeLog`
> files we generate during build (taking up space and time):
> 
>     1. We maintain the `NEWS.md` file
> 
>     2. All history is available in `git`
> 
>     3. All history is available on `github.com` web site

If we drop ChangeLog creation  as Guile devs did in 2008 we have to add something like [this](http://git.savannah.gnu.org/cgit/guile.git/tree/ChangeLog).

Another way is to create one common ChangeLog in the root directory like [here](https://github.com/spk121/guile-gi/blob/master/Makefile.am).  Some people install tools from tarballs and expect to see the info there.

I would not mind if any of the options is implemented.

graahnul-grom, 2021-02-17:
@vzh I think the single `ChangeLog` file in the root
directory with explanation that we no longer use
log files and referring users to `git` like in Guile
is the way to go.

> Some people install tools from tarballs and expect to see the info there.

Since #473 we install `NEWS.md`. IMHO, it's sufficient
for the end users. Developers/advanced users shouldn't have
any problems with git/github, I hope.

vzh, 2021-02-17:
As I've already said, I'm OK with both approaches.

graahnul-grom, 2021-02-19:
We put the `lepton-attrib` script to the same dir
as `libleptonattrib`, it doesn't look good.
Should be placed in its own `attrib` dir (like
`lepton-schematic`) or moved to `utils/`.
Or something else. WDYT?

vzh, 2021-02-19:
I would move anything executable to `utils/` :-)

graahnul-grom, 2021-02-20:
Please, let's not move `lepton-schematic` to `utils/`,
at least for now! It is *not* a utility and will be looking odd there.

graahnul-grom, 2021-02-20:
What do you think about `symbols/docs/nc.doc`?
Frankly, it looks like a joke.
Could we move/remove or at least do not build and install it?

vzh, 2021-02-20:
Probably move its contents to manual?  Just to make it not so boring :-)

graahnul-grom, 2021-02-20:
:-) It would be nice!

vzh, 2021-02-20:
OK, I'll try to do it then.

vzh, 2021-03-07:
@graahnul-grom, is anything else we should do to close this issue?

graahnul-grom, 2021-03-07:
IIRC, we thought about moving common Scheme files (now
piled up in `liblepton`) to a separate top-level directory.

And a couple of (minor) points:
- To stay consistent, we should rename the `gettext` domain
for `libleptongui` in `libleptongui/po/Makevars` and
elsewhere: it's still called `lepton-schematic`.
- It won't hurt to revise `.gitignore` files: many of them
could be removed and/or bring up to date.

vzh, 2021-03-07:
As for toplevel directory for Scheme files, it was just an idea.  I would prefer to not do it now.  Let's do it when we get rid of C `core` modules.  I'm working on this now and having some progress.  Not sure, though, when it will be complete.

Two other (minor) issues can be solved just now, if someone would want to work on them, or, as an option, we could delay them by making up a couple of new issues.  It's just not so very convenient to orient yourself in the mass of comments and done stuff here.

WDYT?

graahnul-grom, 2021-03-07:
You are right about the `core` modules, let's get rid of them first.
IMHO, gettext/gitignore stuff is not worth opening new issues,
and certainly won't keep us from closing this one.
Our repository now looks *much* better, thank you!

I'm closing #732 "Organize the repository".

closed 2021-03-07 by graahnul-grom