lepton-eda/lepton-eda #821: Switcap backend doesn't work completely, like it does in gEDA/gaf

AweVision7, 2021-07-04:
Lepton's switcap backend doesn't work completely, like it does in gEDA/gaf by appending the test.ana file to the end of the netlist for it to be accepted as the proper input to the switcap program, sw.exe.

As stated in the .pdf document "gEDA/gaf Switcap Symbols and Netlister" by Dan McMahill, April 13th, 2003, found at http://web.mit.edu/~geda/arch/sun4x_59/versions/20060124/share/doc/geda-doc/switcap/switcap.pdf

"Figure 3 also shows an instantiation of an analysis block which directs the netlister to include the contents of the file test.ana in the output netlist"

The problem is that the command, lepton-netlist -g switcap -o example.scn ckt.sch clocks.sch analysis.sch
 generates the netlist file, example.scn, but only appends the following comment line,
 
/* reading analysis from "test.ana" */

to this example.scn netlist file, but FAILS to APPEND the contents of test.ana immediatly after this comment line, which is required to run the switcap simulation properly.

Please fix this backend problem with lepton.

Thank you,

Steven Astravas

graahnul-grom, 2021-07-04:
@AweVision7 Steven, thank you for the report.
Could you please attach a simple schematic to test
the backend along with expected and actual output?

graahnul-grom, 2021-07-04:
I get a backtrace while running the following command
in `utils/netlist/examples/switcap`:
`lepton-netlist -g switcap -o - ckt.sch clocks.sch analysis.sch`

```
Backtrace:
           4 (primitive-load "/home/dmn/lepton/bin.master/bin/lepton-netlist")
           3 (%with-toplevel #<geda-toplevel 0x800e39b90> #<procedure 803ef2d80 at ice-9/eval.scm:330:13 ()>)
In ice-9/eval.scm:
    619:8  2 (_ #(#(#(#<directory (guile-user) 801b3d140>) #f) ("GND" "OUT" "unnamed_net1" "unnamed_net2" "unnamed_net3" "unnamed_net4" "unnamed_net5" "unnamed_net6") ("ANA1" "C1" "C2" "C3" "CLK1" "E1" "OPTIONS" "S1" "S2" "S3" "S4" "S5" "S6" "S7" "S8" "TIMING" "TITLE" "V1")))
    619:8  1 (_ #(#(#(#<directory (guile-user) 801b3d140>) ("ANA1" "C1" "C2" "C3" "CLK1" "E1" "OPTIONS" "S1" "S2" "S3" "S4" "S5" "S6" "S7" "S8" "TIMING" "TITLE" "V1")) "ANA1"))
   336:13  0 (_ #<input: test.ana 12>)

ice-9/eval.scm:336:13: Wrong number of arguments to #<procedure switcap:cat-file (a b)>
```

graahnul-grom, 2021-07-04:
@AweVision7 Please try this patch, it works for me.
Apply it to the latest git sources with
`git apply gnet-switcap.diff.txt`

```diff
diff --git a/utils/netlist/scheme/backend/gnet-switcap.scm b/utils/netlist/scheme/backend/gnet-switcap.scm
index 7cf15b108..5ba6740da 100644
--- a/utils/netlist/scheme/backend/gnet-switcap.scm
+++ b/utils/netlist/scheme/backend/gnet-switcap.scm
@@ -376,7 +376,7 @@
     (let ((fname (switcap:write-attrib package "file")))
       (display "\" */\n")
       (if (file-exists? fname)
-        (switcap:cat-file (open-input-file fname))
+        (switcap:cat-file (open-input-file fname) (current-output-port))
         (netlist-error 1 "ERROR: Analysis file ~S not found.\n" fname)))))
 
 (define switcap:cat-file
```

[gnet-switcap.diff.txt](https://github.com/lepton-eda/lepton-eda/files/6760423/gnet-switcap.diff.txt)

vzh, 2021-07-05:
@AweVision7, as @graahnul-grom already asked, could you please provide us with a simple test schematic to include it in the test suite?

vzh, 2021-07-06:
Ah, I see, the example is also in the `examples/` directory. Sorry for the noise.

closed 2021-07-07 by graahnul-grom

vzh, 2021-07-20:
@AweVision7, have you ever used the SWITCAP program itself, I mean `sw` or `sw2`?  I've not found a way to download it anywhere.

vzh, 2021-07-20:
JFTR, just found: http://innovation.columbia.edu/technologies/Proxy30_optimal-simulator-for-mixed/licensing

vzh, 2021-07-20:
Hmm, they require my phone number for signing up. Why?  Leaving out.

PR #822 (2021-07-07): Fix an error in the switcap netlist backend
Closes #821.