lepton-eda/lepton-eda #867: lepton-schematic-1.9.17 fails to run (no code for module (lepton srfi-37)) in Mageia cauldron

barjac, 2022-01-17:
Testing a new build for Mageia I hit this issue.
The file /usr/share/lepton-eda/scheme/lepton/srfi-37.scm is installed from the package.
I will attach the full package build log for your perusal.
Thanks,
Barry

Running from command line:
[baz@localhost ~]$ lepton-schematic
;;; note: source file /usr/bin/lepton-schematic
;;;       newer than compiled /home/baz/.cache/guile/ccache/3.0-LE-8-4.5/usr/bin/lepton-schematic.go
;;; note: auto-compilation is enabled, set GUILE_AUTO_COMPILE=0
;;;       or pass the --no-auto-compile argument to disable.
;;; compiling /usr/bin/lepton-schematic
;;; WARNING: compilation of /usr/bin/lepton-schematic failed:
;;; no code for module (lepton srfi-37)
Backtrace:
          10 (primitive-load "/usr/bin/lepton-schematic")
In ice-9/eval.scm:
   721:20  9 (primitive-eval (use-modules (ice-9 match) (srfi #) # …))
In ice-9/psyntax.scm:
  1230:36  8 (expand-top-sequence ((use-modules (ice-9 match) (…) …)) …)
  1222:19  7 (parse _ (("placeholder" placeholder)) ((top) #(# # …)) …)
   259:10  6 (parse _ (("placeholder" placeholder)) (()) _ c&e (eval) …)
In ice-9/boot-9.scm:
  3927:20  5 (process-use-modules _)
   222:29  4 (map1 (((ice-9 match)) ((srfi srfi-1)) ((lepton #)) # #))
   222:29  3 (map1 (((srfi srfi-1)) ((lepton srfi-37)) ((system …)) …))
   222:17  2 (map1 (((lepton srfi-37)) ((system foreign)) ((# ffi))))
  3928:31  1 (_ ((lepton srfi-37)))
   3329:6  0 (resolve-interface (lepton srfi-37) #:select _ #:hide _ …)

ice-9/boot-9.scm:3329:6: In procedure resolve-interface:
no code for module (lepton srfi-37

[build.x86_64.0.20220117213615.log](https://github.com/lepton-eda/lepton-eda/files/7884962/build.x86_64.0.20220117213615.log)
)

vzh, 2022-01-18:
May it be the usual issue with previously compiled Scheme code? Have you tried to wholly remove `~/.cache/guile`? Have you used `make precompile` previously with lepton. Some `.go` files may still exist then in your installation directory.

graahnul-grom, 2022-01-18:
Do you have the same problem when building with just
`./configure && make && make install` on your system
(i.e. not using the rpmbuild procedure)?

barjac, 2022-01-18:
It is related to ldconfig and not finding the proper libs including .so.major. We had a hack for this which no longer works but it was pointed out that you had included code to try to solve the problem here: https://github.com/lepton-eda/lepton-eda/commit/2f9c7ae2357481f4928fa9f56657978356be6b66#diff-2466c34d3613b509adae8681c3b463e4d8874839028f7c43709da385ed8faaea
However exporting USE_LDCONFIG does get past the problem we were seeing above and now we have:

[baz@localhost ~]$ lepton-schematic
;;; note: auto-compilation is enabled, set GUILE_AUTO_COMPILE=0
;;;       or pass the --no-auto-compile argument to disable.
;;; compiling /usr/bin/lepton-schematic
;;; compiling /usr/share/lepton-eda/scheme/lepton/srfi-37.scm
;;; compiling /usr/share/lepton-eda/scheme/lepton/core/gettext.scm
;;; compiling /usr/share/lepton-eda/scheme/lepton/m4.scm
;;; compiled /home/baz/.cache/guile/ccache/3.0-LE-8-4.5/usr/share/lepton-eda/scheme/lepton/m4.scm.go
;;; compiled /home/baz/.cache/guile/ccache/3.0-LE-8-4.5/usr/share/lepton-eda/scheme/lepton/core/gettext.scm.go
;;; compiled /home/baz/.cache/guile/ccache/3.0-LE-8-4.5/usr/share/lepton-eda/scheme/lepton/srfi-37.scm.go
;;; compiling /usr/share/lepton-eda/scheme/lepton/ffi.scm
;;; compiling /usr/share/lepton-eda/scheme/lepton/ffi/lib.scm
;;; compiled /home/baz/.cache/guile/ccache/3.0-LE-8-4.5/usr/share/lepton-eda/scheme/lepton/ffi/lib.scm.go
;;; WARNING: compilation of /usr/share/lepton-eda/scheme/lepton/ffi.scm failed:
;;; In procedure dlopen: file "libglib-2.0.so", message "libglib-2.0.so: cannot open shared object file: No such file or directory"
;;; WARNING: compilation of /usr/bin/lepton-schematic failed:
;;; Unbound variable: libglib
Backtrace:
           3 (primitive-load "/usr/bin/lepton-schematic")
In ice-9/eval.scm:
   187:27  2 (_ _)
   223:20  1 (proc #<directory (guile-user) 7ff0e6daac80>)
In unknown file:
           0 (%resolve-variable (7 . liblepton_init) #<directory (gu…>)

ERROR: In procedure %resolve-variable:
Unbound variable: liblepton_init
[baz@localhost ~]$

> Do you have the same problem when building with just `./configure && make && make install` on your system (i.e. not using the rpmbuild procedure)?

If you really feel it will help I will test in a VM snapshot.

barjac, 2022-01-18:
Manual build in Mageia cauldron fully updated produces the same error as the package build.
I had to force gtk3 with  --with-gtk3 and I exported USE_LDCONFIG before running ./configure.

[baz@localhost ~]$ lepton-schematic
;;; note: auto-compilation is enabled, set GUILE_AUTO_COMPILE=0
;;;       or pass the --no-auto-compile argument to disable.
;;; compiling /usr/local/bin/lepton-schematic
;;; compiling /usr/local/share/lepton-eda/scheme/lepton/srfi-37.scm
;;; compiling /usr/local/share/lepton-eda/scheme/lepton/core/gettext.scm
;;; compiling /usr/local/share/lepton-eda/scheme/lepton/m4.scm
;;; compiled /home/baz/.cache/guile/ccache/3.0-LE-8-4.5/usr/local/share/lepton-eda/scheme/lepton/m4.scm.go
;;; compiled /home/baz/.cache/guile/ccache/3.0-LE-8-4.5/usr/local/share/lepton-eda/scheme/lepton/core/gettext.scm.go
;;; compiled /home/baz/.cache/guile/ccache/3.0-LE-8-4.5/usr/local/share/lepton-eda/scheme/lepton/srfi-37.scm.go
;;; compiling /usr/local/share/lepton-eda/scheme/lepton/ffi.scm
;;; compiling /usr/local/share/lepton-eda/scheme/lepton/ffi/lib.scm
;;; compiled /home/baz/.cache/guile/ccache/3.0-LE-8-4.5/usr/local/share/lepton-eda/scheme/lepton/ffi/lib.scm.go
;;; WARNING: compilation of /usr/local/share/lepton-eda/scheme/lepton/ffi.scm failed:
;;; In procedure dlopen: file "liblepton.so", message "liblepton.so: cannot open shared object file: No such file or directory"
;;; WARNING: compilation of /usr/local/bin/lepton-schematic failed:
;;; Unbound variable: liblepton
Backtrace:
           3 (primitive-load "/usr/local/bin/lepton-schematic")
In ice-9/eval.scm:
   187:27  2 (_ _)
   223:20  1 (proc #<directory (guile-user) 7fda481d3c80>)
In unknown file:
           0 (%resolve-variable (7 . liblepton_init) #<directory (gu…>)

ERROR: In procedure %resolve-variable:
Unbound variable: liblepton_init
[baz@localhost ~]$

vzh, 2022-01-18:
Use of `USE_LDCONFIG` is described in NEWS, you may want to do: `./configure USE_LDCONFIG=/sbin/ldconfig`.  At the `make` stage, it should create `m4.scm` with library definitions, that will be installed e.g. in `/usr/local/share/lepton-eda/scheme/lepton/m4.scm`.  Please check if that file contains non-empty strings in definitions of `%ldconfig-something` variables and that those files really exist.

barjac, 2022-01-19:
Reading the notes again with that patch I realized that it was expecting more than just USE_LDCONFIG so I corrected that with: export USE_LDCONFIG=/usr/sbin/ldconfig before running ./configure
It made no difference to the manual build (I preceded it with make clean). Runtime errors were the same.

However rebuilding the package with the corrected USE_LDCONFIG=... and installing it (after doing a make uninstall of the local package) it now runs!
I will push this to our build system and go to bed. I will test it in the morning and let you know.
I have a couple of schematics to draw and was hitting an issue with copy/pasting which was what prompted me to try the latest version, so I will be doing some testing once it's up and running :)
Thanks for your help.

vzh, 2022-01-19:
My pleasure.  Glad to hear it worked for you.  As for `make clean`, it won't work if the package was configured before with wrong `USE_LDCONFIG` setting. It's better to do `autoreconf -ifv && ./configure ...` before `make` in such a case.

barjac, 2022-01-19:
Hi,
Thanks for the tips - I never build on my machine except for packages so it's a long time since I used make; make install ;)

Something odd is happening.
Today I have tested the package built overnight which ran OK in my VM from my local test package.
On first run I was back with the original error in the top post - ;;; no code for module (lepton srfi-37)
I logged out and logged in to a clean user and it ran OK - I logged back into my regular cauldron test user and it also ran OK.
So it appears to have needed a log out/in after install. Any thoughts?

vzh, 2022-01-19:
I have only one guess. After installation, your environment/paths were not updated to use new Lepton dynamic libraries, thus the error. Usually, `sudo ldconfig` cures this in place. After log out and in those things were set up from scratch by your shell/wm, which fixed it. (Disclaimer: I may be wrong ;-) )

closed 2022-01-19 by vzh