lepton-eda/lepton-eda #871: make and make install not working

RGT1995, 2022-02-21:
When running make command it shows-
make: *** No rule to make target 'version.h', needed by 'all'.  Stop.
btw
I'm using windows ubuntu app on windows 10

vzh, 2022-02-21:
Are you building from a tarball or git repo?  If the former, the source code should have `version.h` in the root directory in which you extract the tarball contents.  If the latter, try doing first `autoreconf -ifv && ./configure`.  Please let us know if this helps or does not help.

RGT1995, 2022-02-22:
the make command ran, but make install command showed error
mkdir: cannot create directory ‘/usr/local/share/icons’: Permission denied
mkdir: cannot create directory ‘/usr/local/share/icons’: Permission denied
Failed to create directory /usr/local/share/icons/hicolor/16x16/mimetypes
make[4]: *** [Makefile:692: install-data-local] Error 1
make[4]: Leaving directory '/mnt/d/lepton-eda/lepton-eda/liblepton/data'
make[3]: *** [Makefile:572: install-am] Error 2
make[3]: Leaving directory '/mnt/d/lepton-eda/lepton-eda/liblepton/data'
make[2]: *** [Makefile:544: install-recursive] Error 1
make[2]: Leaving directory '/mnt/d/lepton-eda/lepton-eda/liblepton'
make[1]: *** [Makefile:815: install-recursive] Error 1
make[1]: Leaving directory '/mnt/d/lepton-eda/lepton-eda'
make: *** [Makefile:1269: install] Error 2

vzh, 2022-02-22:
Usually on Linux, you need superuser permissions to install anything into a directory under `/usr`. So instead of plain `make install` you have to use something like `sudo make install` if `sudo` is properly configured on your system.  Another option is to log in as the `root` user and do `make install`.  BTW, please first ensure the directory `/usr/local/share` exists on your system (I'm not sure what ubuntu app provides on windows).

RGT1995, 2022-02-22:
I think the sudo make install worked, but when i try to run lepton-schematic, this happens
(guile-2.2:8863): Gtk-WARNING **: 13:47:06.686: cannot open display:

vzh, 2022-02-22:
It may be due to running the program under the user that does not access to the current X server display.  E.g., if I log in as `vzh` and run `sudo lepton-schematic` in terminal under `root`, I get the same error `cannot open display: :0`.  Of course, it can be somehow solved using X utilities such as `xhost` and `xauth` though in my case I'd better run the program without `sudo` :-) Please check if the user you're running the program under has appropriate permissions to access the X server display reported.  Googling for the issue may help in this case, I believe.

RGT1995, 2022-02-22:
First of all, I really appreciate your help and thank you for bearing with a newb.
I think I solved the problem `(guile-2.2:8863): Gtk-WARNING **: 13:47:06.686: cannot open display:`
But now when I enter `lepton-schematic`  no GUI opens up

vzh, 2022-02-22:
@RGT1995, do you have any other GTK app installed on your system? Maybe running it would provide you with some new hints in solving the issue.

RGT1995, 2022-02-23:
@graahnul-grom  I tried that but it did not work.

RGT1995, 2022-02-23:
@vzh  I tried to run gnome-calculator it showed the same error

vzh, 2022-02-23:
@RGT1995, it's obviously not an issue with Lepton then.  I doubt I would be of much help here.  Just never used such a configuration.  You have to learn how to connect to the X server on your machine.  First, I would try `ps ax|grep X` or `ps ax|grep xorg` to find out the X process which runs on your machine and learn what args it runs with.  Then maybe use `top` or some similar program to find the process owner, it may be `root` or your local user account.  Probably, that information could provide you with some farther hints.  And... maybe some forums dedicated to `ubuntu-app` exist?  You could try to ask about the issue there or search for issues mentioned there before.  Anyways,  if you resolve your issue, please let us know.  I'd be glad to add such info to our wiki.

graahnul-grom, 2022-02-23:
I've found a relevant documentation at
https://wiki.ubuntu.com/WSL, especially this chapter:
https://wiki.ubuntu.com/WSL#Running_Graphical_Applications

It seems that you have to install the X server separately
and run it before starting any GUI apps.

Another possibility to run Lepton EDA under Windows
is to use [Cygwin](https://cygwin.com).
We have a wiki page describing the installation process:
https://github.com/lepton-eda/lepton-eda/wiki/Cygwin

RGT1995, 2022-03-14:
@vzh  @graahnul-grom 
Lepton-schematic is running on my windows.
Thanks for the support.

closed 2022-03-14 by RGT1995

vzh, 2022-03-14:
Hooray!
Could you please share with us the way you solved your last issue with connection to X?

RGT1995, 2022-03-30:
I started with dowloading WSL and ubuntu.
Then using `sudo apt install` most of the dependencies could be installed.
After installing lepton, I downloaded Xming and launched Xlaunch with no access control.
In the ubuntu terminal I ran
`export DISPLAY=$(awk '/nameserver / {print $2; exit}' /etc/resolv.conf 2>/dev/null):0`

vzh, 2022-03-31:
@RGT1995, @graahnul-grom , thank you! I've added the [page on installation under Windows](https://github.com/lepton-eda/lepton-eda/wiki/Windows) referring to this issue and some links above.