lepton-eda/lepton-eda #873: lepton-schematic ubuntu impish fails to run

DavidAustin, 2022-02-24:
Trying to run lepton-schematic from the ubuntu impish package gives:

```
$ lepton-schematic 
;;; note: auto-compilation is enabled, set GUILE_AUTO_COMPILE=0
;;;       or pass the --no-auto-compile argument to disable.
;;; compiling /usr/bin/lepton-schematic
;;; WARNING: compilation of /usr/bin/lepton-schematic failed:
;;; In procedure dynamic-link: file: "libglib-2.0", message: "file not found"
Backtrace:
           3 (primitive-load "/usr/bin/lepton-schematic")
In ice-9/eval.scm:
   187:27  2 (_ _)
   223:20  1 (proc #<directory (guile-user) 7f975fb8cc80>)
In unknown file:
           0 (%resolve-variable (7 . liblepton_init) #<directory (gu…>)

ERROR: In procedure %resolve-variable:
Unbound variable: liblepton_init
```

I have libglib2.0-0:amd64 installed which provides /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.6800.4

graahnul-grom, 2022-02-25:
IIUC, it's Lepton EDA 1.9.16 installed from the official
Ubuntu ("Impish Indri", 21.10) repository, right?

Try to delete Guile cache directory: `~/.cache/guile/`

DavidAustin, 2022-02-25:
Thanks for that suggestion.  Unfortunately, it didn't help I get a bit more compiling but same file error.

vzh, 2022-02-25:
This message:
> ;;; In procedure dynamic-link: file: "libglib-2.0", message: "file not found"

suggests that the canonical library for `libglib` has not been found.  It's a known issue for 1.9.16 on Debian ([#999699](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=999699)) and its derivatives. The workaround for this version is to install corresponding `*-dev` packages for missing libs.  In this case those are `libglib2.0-dev` and `libgtk2.0-dev` (`libgtk-3-dev` for GTK3 version, though it should not be the case on Ubuntu).
Another workaround is to download and install a newer `lepton-eda` package directly from the Debian repository.  The issue should be fixed there in 1.9.16-2 or 1.9.16-3.

DavidAustin, 2022-02-28:
Thanks - that did the trick.

vzh, 2022-03-02:
Done, thanks.

closed 2022-03-02 by vzh

graahnul-grom, 2022-03-02:
@vzh Thank you.