lepton-eda/lepton-eda #930: Q: Is this Functional Electronic Computer-Aided Design (fECAD) solution? (Basically OpenSCAD for electronics?)

Kreyren, 2022-07-08:
For context.. found this project on GNU Guile website which alleges that it's using GNU Guile, but trying the software and looking at example i have no idea what it is and how to use it while i am actively looking for a solution that i would describe as "OpenSCAD for electronical design" meaning that i expect to write a code for the solution to return a schematics and gerber files for fabrication (fECAD).

Can lepton-eda be used this way?

vzh, 2022-07-09:
Lepton is a fork of gEDA/gaf, the part delivering mostly schematic capture and netlisting to various formats, including those for PCB creation and simulation (see [gEDA in wikipedia](https://en.wikipedia.org/wiki/GEDA)).  For creation of PCB and gerber files, other tools are used, such as **pcb** or **pcb-rnd** in combination with **gerbv**.  Lepton is an only part heavily based on Guile.  While you can use various forms of automation, I'm not sure it is comparable with OpenSCAD, though :-) You have to try it to understand what it is like. As for PCB and gerber tools, you can use scripts in various languages in some of them, too, e.g. in **pcb-rnd**, though you would better see their docs, I don't know much of this.

antonysigma, 2023-07-15:
The gEDA follows the [UNIX SW design philosophy](https://en.wikipedia.org/wiki/Unix_philosophy), that is:

1. Build one app to do one and only one job;
2. Optimize the app to do it well;
3. Write the output to plaintext format, so it can be ingested by another app.

So, unlike KiCAD or other monolithic EDA designer, we start the project by drawing custom schematic symbols with [a conversion script](http://www.gedasymbols.org/user/dj_delorie/tools/djboxsym.html). The output symbol `*.sym` is then imported into the schematic editor `gschem.exe` to draw a circuit diagram (`*.sch`). Another app `gnetlist.exe` converts your drawing (`*.sch`) into the netlist file `*.net`, followed by PCB manual/auto routing in the `pcb.exe` app. The PCB design is often exported as the Gerber file.

As you can see, the gEDA is pretty much a set of text conversion apps, daisy chained together (`sym -(geshem)-> sch -(gnetlist)-> net -(pcb)-> pcb -(pcb)-> gerber`). I don't see any other modern EDA suite following the similar UNIX approach. Feel free to correct me.

@Kreyren , does it match your definition of "functional programming EDA"?