lepton-eda/lepton-eda #985: lepton-netlist will not run

majenkotech, 2022-12-19:
```
Backtrace:
In ice-9/boot-9.scm:
  1752:10  7 (with-exception-handler _ _ #:unwind? _ # _)
In unknown file:
           6 (apply-smob/0 #<thunk 7fac0b7d42e0>)
In ice-9/boot-9.scm:
    724:2  5 (call-with-prompt _ _ #<procedure default-prompt-handle…>)
In ice-9/eval.scm:
    619:8  4 (_ #(#(#<directory (guile-user) 7fac0b7d8c80>)))
In ice-9/boot-9.scm:
   2836:4  3 (save-module-excursion _)
  4388:12  2 (_)
In lepton/toplevel.scm:
     69:2  1 (with-toplevel _ _)
     57:9  0 (main)

lepton/toplevel.scm:57:9: In procedure main:
In procedure fluid-ref: Wrong type argument in position 1 (expecting fluid): #f
```

* Guile version 3.0.8
* Arch Linux

vzh, 2022-12-19:
Have you tried to remove guile cache?  Something like `rm -rf ~/.cache/guile/ccache/`

majenkotech, 2022-12-19:
Yes, that's the first thing I tried. BTW, it works fine on Guile 2.2.7. It's a "3" thing....

vzh, 2022-12-20:
OK, I managed to trigger the issue on Lepton `make distcheck` tests.  AFAICR, I saw the issue before on my local PC.  Even though my own [main](https://github.com/vzh/lepton-eda/tree/main) branch doesn't trigger it, despite the fact it uses Guile 3.0 for tests, I could reproduce it on my own PCs.  Even there, only `lepton-netlist` test suite fails, installed `lepton-netlist` works as expected AFAICS.  OK, I've prepared a patch that I tested on both Guile 2.2 and Guile 3.0.  Could you please test it in your configuration if it solves the issue?  I attach it below. (you could also just sudo edit the affected file in your Lepton installation and run the program as I sometimes do for testing  :-) ).

[0001-lepton-Initialize-toplevel-fluid-in-definition-of-le.patch.txt](https://github.com/lepton-eda/lepton-eda/files/10266707/0001-lepton-Initialize-toplevel-fluid-in-definition-of-le.patch.txt)
 
The commit description may shed some light on my *IMHO* on what's going on here.

graahnul-grom, 2022-12-28:
I can reproduce it on FreeBSD 13.1-STABLE amd64 with guile 3.0.8.
No errors on normal `lepton-netlist` runs, but during `make distcheck`
all netlist tests failed with similar output.
PR #991 fixes this.

PR #991 (2022-12-28): Fix `make distcheck` when compiled with Guile 3.0
As it is mentioned in [Guile 3.0.8 NEWS](https://git.savannah.gnu.org/cgit/guile.git/tree/NEWS?h=v3.0.8#n8),
in the 3.0 series Guile developers introduced new kinds of
optimizations treating Scheme modules as *declarative* and
inlining some of their toplevel exported definitions into other
modules.  This facility is enabled at the default `-O2`
optimization level when a module is compiled, that is, it is used
during auto-compilation as well.

I didn't notice any problems with this approach until #985
revealed one issue.  Afterwards, I could rep [... cut]
closed 2022-12-28 by vzh