lepton-eda/lepton-eda #1005: Links to gitter.im are broken

vzh, 2023-03-07:
Both Lepton EDA github interface (README, wikis, maybe something else) and links in the repository itself including interfaces of our programs and several manuals include links to gitter.  Now they're broken due to moving the functionality of gitter to other servers by its owners.  We have to somehow fix this but I'm not sure what new link should we use: https://app.gitter.im/#/room/#Lepton-EDA_Lobby:gitter.im or maybe https://matrix.to/#/#Lepton-EDA_Lobby:gitter.im?

graahnul-grom, 2023-04-01:
Yes, the link works, we could use it.
Though today the original URL is also OK - it redirects
to the new address (app.gitter.im/...). A couple of days
ago it didn't work at all.
Other observations: in web browsers on Android gitter.im is
totally broken. You have to install some matrix client app.
Also I managed to login via the matrix.org server (clicking
"login via github.com" there); in this case a different user
name is displayed in the chat room. But if I login via the
gitter.im server, my old account is recognized.

graahnul-grom, 2023-07-06:
The first URL works, we could replace all gitter links with it.
Now it also works on Android, though it's barely usable in web
browsers, a dedicated Android app (a matrix messenger) should
be used instead.
I've tested "FluffyChat" (https://f-droid.org/en/packages/chat.fluffy.fluffychat).

vzh, 2023-07-06:
> @vzh Could you please check your `gitter` account? Does it work?

Yes, it works.

> I've sent you a message. 

:-)

> Did you get a notification via e-mail?

No, there are no notifications at all.

vzh, 2023-07-06:
> The first URL works, we could replace all gitter links with it. Now it also works on Android, though it's barely usable in web browsers, a dedicated Android app (a matrix messenger) should be used instead. I've tested "FluffyChat" (https://f-droid.org/en/packages/chat.fluffy.fluffychat).

We have to document it somewhere.  (Please go for it, if you can.)

PR #1031 (2023-07-14): Update the gitter URL
The new URL is
https://app.gitter.im/#/room/#Lepton-EDA_Lobby:gitter.im

Closes #1005
closed 2023-07-14 by graahnul-grom