lepton-eda/lepton-eda #1010: Cannot load lepton-schematic configuration

leguizar, 2023-03-23:
I tried to run lepton-schematic and close with the following message: "Failed to load RC file [/usr/local/share/lepton-eda/system-gschemrc]: Wrong type to apply:#<syntax-transformer check-string>"

**What appears in the terminal window is the following message:** 
leguizar@Dell-Inspiron-7386:~$ lepton-schematic
;;; note: auto-compilation is enabled, set GUILE_AUTO_COMPILE=0
;;;       or pass the --no-auto-compile argument to disable.
;;; compiling /usr/local/share/lepton-eda/scheme/netlist.scm
;;; compiling /usr/local/share/lepton-eda/scheme/netlist/net.scm
;;; WARNING: compilation of /usr/local/share/lepton-eda/scheme/netlist/net.scm failed:
;;; Wrong type to apply: #<syntax-transformer check-string>
;;; WARNING: compilation of /usr/local/share/lepton-eda/scheme/netlist.scm failed:
;;; Unbound variable: netlist-config-ref
Backtrace:
In ice-9/boot-9.scm:
  1752:10 10 (with-exception-handler _ _ #:unwind? _ # _)
In unknown file:
           9 (apply-smob/0 #<thunk 7fabf834e540>)
In ice-9/boot-9.scm:
    724:2  8 (call-with-prompt _ _ #<procedure default-prompt-handle…>)
In ice-9/eval.scm:
    619:8  7 (_ #(#(#<directory (guile-user) 7fabf8352c80>)))
In ice-9/boot-9.scm:
   2835:4  6 (save-module-excursion _)
  4380:12  5 (_)
In /usr/local/bin/lepton-schematic:
   328:15  4 (_)
    269:2  3 (main ())
   226:13  2 (open-log-window #<pointer 0x56138fdcf000>)
In lepton/config.scm:
    147:2  1 (path-config-context "/home/leguizar")
    147:2  0 (path-config-context "/home/leguizar")

lepton/config.scm:147:2: In procedure path-config-context:
Wrong type to apply: #<syntax-transformer check-string>

*** What appears in the log messages is:** 
Lepton EDA/lepton-schematic 1.9.18.20220529 (git: d24967d)

Loaded RC file [/usr/local/share/lepton-eda/system-gafrc]
Library at "/home/leguizar/gaf/gschem-sym" has been already added.
Library at "/home/leguizar/gaf/symbols" has been already added.
Loaded RC file [/home/leguizar/.config/lepton-eda/gafrc]

Backtrace:
          19 (apply-smob/0 #<thunk 7f899df4e540>)
In ice-9/boot-9.scm:
    724:2 18 (call-with-prompt _ _ #<procedure default-prompt-handle…>)
In ice-9/eval.scm:
    619:8 17 (_ #(#(#<directory (guile-user) 7f899df52c80>)))
In ice-9/boot-9.scm:
   2835:4 16 (save-module-excursion _)
  4380:12 15 (_)
In /usr/local/bin/lepton-schematic:
   328:15 14 (_)
   258:17 13 (main ())
   250:37 12 (make-schematic-window)
In unknown file:
Exception thrown while printing backtrace:
In procedure primitive-call-ip: Wrong type argument in position 1 (expecting PRIMITIVE_P): #<procedure 7f899dec8e20 ()>

ERROR: In procedure apply-smob/1:
Wrong type to apply: #<syntax-transformer check-string>

ERROR: Failed to load RC file [/usr/local/share/lepton-eda/system-gschemrc]: Wrong type to apply: #<syntax-transformer check-string>

Loaded RC file [/home/leguizar/.config/lepton-eda/gschemrc]
**The version already installed is:**
lepton-cli --version
Lepton EDA/lepton-cli 1.9.18.20220529 (git: d24967d)
Copyright (C) 1998-2016 gEDA developers
Copyright (C) 2017-2022 Lepton EDA developers
This is free software, and you are welcome to redistribute it
under certain conditions. For details, see the file `COPYING',
which is included in the Lepton EDA distribution.
There is NO WARRANTY, to the extent permitted by law.

vzh, 2023-03-23:
Please look inside of your `/usr/local/share/lepton-eda/system-gschemrc`, maybe it just has the symbol `#` somewhere in it.  If in doubt, just attach the file here.  If GH doesn't allow to attach the file without extension, just rename it to `system-gschemrc.txt`.

vzh, 2023-03-23:
Hmm, I've probably misread your message.  Your first message contains
> ;;; WARNING: compilation of /usr/local/share/lepton-eda/scheme/netlist/net.scm failed:
> ;;; Wrong type to apply: #

Maybe your compiled guile files `*.go` and their corresponding sources `*.scm` differ?  Please try to `rm -r ~/.cache/guile/` first.

leguizar, 2023-03-23:
Hello,
to your first comment system-gshemrc is as follows:
(use-modules (gschem deprecated)
             (schematic action)
             (schematic attrib)
             (schematic builtins)
             (schematic gui keymap)
             (schematic gui stroke)
             (schematic menu)
             (schematic netlist)
             (lepton config))

( load-from-path "conf/schematic/deprecated.scm" )
( load-from-path "conf/schematic/attribs.scm"    )
( load-from-path "conf/schematic/stroke.scm"     )
( load-from-path "conf/schematic/keys.scm"       )
( load-from-path "conf/schematic/menu.scm"       )

leguizar, 2023-03-23:
To your second recommendation, it works perfect!! Thanks a lot!!
First I removed /guile/ from /.cache/
Second in the terminal I ran lepton-sehematic and after compiling it rebuild /guile/ in /.cache/ AND
IT WORKS!!, at least it starts, please allow me to try the soft and I'll be back to tell you.

leguizar, 2023-03-23:
Hello again, it appears another problem when I running the software, when I tried to generate a new symbol archive and adding elements to the drawing, suddendly the program closed showing this message in the terminal window:

lepton-schematic TDA1085C.sym

(guile-3.0:30230): GLib-GObject-WARNING **: 17:16:08.364: invalid unclassed pointer in cast to 'GtkToggleButton'

(guile-3.0:30230): Gtk-CRITICAL **: 17:16:08.364: IA__gtk_toggle_button_get_active: assertion 'GTK_IS_TOGGLE_BUTTON (toggle_button)' failed

(guile-3.0:30230): Gtk-CRITICAL **: 17:16:08.364: IA__gtk_tree_view_get_model: assertion 'GTK_IS_TREE_VIEW (tree_view)' failed

(guile-3.0:30230): Gtk-CRITICAL **: 17:16:08.364: IA__gtk_list_store_clear: assertion 'GTK_IS_LIST_STORE (list_store)' failed

(guile-3.0:30230): Gtk-CRITICAL **: 17:16:08.364: IA__gtk_list_store_append: assertion 'GTK_IS_LIST_STORE (list_store)' failed

(guile-3.0:30230): Gtk-CRITICAL **: 17:16:08.364: IA__gtk_list_store_set_valist: assertion 'GTK_IS_LIST_STORE (list_store)' failed

(guile-3.0:30230): Gtk-CRITICAL **: 17:16:08.364: IA__gtk_list_store_append: assertion 'GTK_IS_LIST_STORE (list_store)' failed

(guile-3.0:30230): Gtk-CRITICAL **: 17:16:08.364: IA__gtk_list_store_set_valist: assertion 'GTK_IS_LIST_STORE (list_store)' failed

(guile-3.0:30230): Gtk-CRITICAL **: 17:16:08.364: IA__gtk_list_store_append: assertion 'GTK_IS_LIST_STORE (list_store)' failed

(guile-3.0:30230): Gtk-CRITICAL **: 17:16:08.364: IA__gtk_list_store_set_valist: assertion 'GTK_IS_LIST_STORE (list_store)' failed

(guile-3.0:30230): Gtk-CRITICAL **: 17:16:08.364: IA__gtk_list_store_append: assertion 'GTK_IS_LIST_STORE (list_store)' failed

(guile-3.0:30230): Gtk-CRITICAL **: 17:16:08.365: IA__gtk_list_store_set_valist: assertion 'GTK_IS_LIST_STORE (list_store)' failed

(guile-3.0:30230): GLib-GObject-WARNING **: 17:16:08.365: invalid uninstantiatable type '(null)' in cast to 'GtkWidget'

(guile-3.0:30230): Gtk-CRITICAL **: 17:16:08.365: IA__gtk_widget_set_sensitive: assertion 'GTK_IS_WIDGET (widget)' failed

(guile-3.0:30230): GLib-GObject-WARNING **: 17:16:08.365: invalid unclassed pointer in cast to 'GtkWidget'

(guile-3.0:30230): Gtk-CRITICAL **: 17:16:08.365: IA__gtk_widget_set_sensitive: assertion 'GTK_IS_WIDGET (widget)' failed

(guile-3.0:30230): GLib-GObject-WARNING **: 17:16:08.365: invalid unclassed pointer in cast to 'GtkWidget'

(guile-3.0:30230): Gtk-CRITICAL **: 17:16:08.365: IA__gtk_widget_get_style: assertion 'GTK_IS_WIDGET (widget)' failed

(guile-3.0:30230): GLib-GObject-WARNING **: 17:16:08.365: invalid unclassed pointer in cast to 'GtkWidget'

(guile-3.0:30230): Gtk-CRITICAL **: 17:16:08.365: IA__gtk_widget_modify_text: assertion 'GTK_IS_WIDGET (widget)' failed

(guile-3.0:30230): GLib-GObject-WARNING **: 17:16:08.365: invalid unclassed pointer in cast to 'GtkToggleButton'

(guile-3.0:30230): Gtk-CRITICAL **: 17:16:08.366: IA__gtk_toggle_button_get_active: assertion 'GTK_IS_TOGGLE_BUTTON (toggle_button)' failed

(guile-3.0:30230): Gtk-CRITICAL **: 17:16:08.366: IA__gtk_tree_view_get_model: assertion 'GTK_IS_TREE_VIEW (tree_view)' failed

(guile-3.0:30230): Gtk-CRITICAL **: 17:16:08.366: IA__gtk_list_store_clear: assertion 'GTK_IS_LIST_STORE (list_store)' failed

(guile-3.0:30230): Gtk-CRITICAL **: 17:16:08.366: IA__gtk_list_store_append: assertion 'GTK_IS_LIST_STORE (list_store)' failed

(guile-3.0:30230): Gtk-CRITICAL **: 17:16:08.366: IA__gtk_list_store_set_valist: assertion 'GTK_IS_LIST_STORE (list_store)' failed

(guile-3.0:30230): Gtk-CRITICAL **: 17:16:08.366: IA__gtk_list_store_append: assertion 'GTK_IS_LIST_STORE (list_store)' failed

(guile-3.0:30230): Gtk-CRITICAL **: 17:16:08.366: IA__gtk_list_store_set_valist: assertion 'GTK_IS_LIST_STORE (list_store)' failed

(guile-3.0:30230): Gtk-CRITICAL **: 17:16:08.366: IA__gtk_list_store_append: assertion 'GTK_IS_LIST_STORE (list_store)' failed

(guile-3.0:30230): Gtk-CRITICAL **: 17:16:08.366: IA__gtk_list_store_set_valist: assertion 'GTK_IS_LIST_STORE (list_store)' failed

(guile-3.0:30230): GLib-GObject-WARNING **: 17:16:08.366: invalid uninstantiatable type '(null)' in cast to 'GtkWidget'

(guile-3.0:30230): Gtk-CRITICAL **: 17:16:08.366: IA__gtk_widget_set_sensitive: assertion 'GTK_IS_WIDGET (widget)' failed

(guile-3.0:30230): GLib-GObject-WARNING **: 17:16:08.366: invalid unclassed pointer in cast to 'GtkWidget'

[... 282 lines cut ...]
(guile-3.0:30230): GLib-GObject-WARNING **: 17:16:08.382: invalid unclassed pointer in cast to 'GtkWidget'

(guile-3.0:30230): Gtk-CRITICAL **: 17:16:08.382: IA__gtk_widget_set_sensitive: assertion 'GTK_IS_WIDGET (widget)' failed

(guile-3.0:30230): GLib-GObject-WARNING **: 17:16:08.382: invalid unclassed pointer in cast to 'GtkWidget'

(guile-3.0:30230): Gtk-CRITICAL **: 17:16:08.382: IA__gtk_widget_get_style: assertion 'GTK_IS_WIDGET (widget)' failed

(guile-3.0:30230): GLib-GObject-WARNING **: 17:16:08.382: invalid unclassed pointer in cast to 'GtkWidget'

(guile-3.0:30230): Gtk-CRITICAL **: 17:16:08.382: IA__gtk_widget_modify_text: assertion 'GTK_IS_WIDGET (widget)' failed
Segmentation fault (core dumped)

The same message appears using Lepton-eda 1.9.16, that's why I moved to 1.9.18 version, but the problem, remains.

The log message is as follows:
Lepton EDA/lepton-schematic 1.9.18.20220529 (git: d24967d)

Loaded RC file [/usr/local/share/lepton-eda/system-gafrc]
Library at "/home/leguizar/gaf/gschem-sym" has been already added.
Library at "/home/leguizar/gaf/symbols" has been already added.
Loaded RC file [/home/leguizar/.config/lepton-eda/gafrc]
Loaded RC file [/usr/local/share/lepton-eda/system-gschemrc]
Loaded RC file [/home/leguizar/.config/lepton-eda/gschemrc]
Loading schematic [/home/leguizar/gaf/symbols/linear/TDA1085C.sym]

graahnul-grom, 2023-03-24:
Do you remember what exactly were you doing when crash happened?
Is this reproducible?
Please provide some info about your system: OS, version, graphics,
compositor, desktop environment, GTK theme/engine, versions of Guile, GTK.

leguizar, 2023-03-24:
Hi, It doesn't matter if I make a circuit or a symbol, The program starts without problem, after add one element in a circuit and has been modified it's values or after make a connection and I try to add a new element the program freezes and closes. When I build an symbol and again after edit attributes in an element in the symbol using ee command I close the pop up window and when I try to edit another one, the program freezes and close.
Laptop: Dell  Inspiron 7386
Processor: Intel® Core™ i5-8265U CPU @ 1.60GHz × 8
Memory 8 Gbytes
Graphics: Mesa Intel® UHD Graphics 620 (WHL GT2)
OS Ubuntu 22.04.2 LTS
Gnome Vers.: 42.5, Windowin System: Wayland.
GTK theme: Yaru
dpkg -l libgtk* | grep -e '^i' | grep -e 'libgtk-*[0-9]'
ii  libgtk-3-0:amd64           3.24.33-1ubuntu2  amd64        GTK graphical user interface library
ii  libgtk-3-bin               3.24.33-1ubuntu2  amd64        programs for the GTK graphical user interface library
ii  libgtk-3-common            3.24.33-1ubuntu2  all          common files for the GTK graphical user interface library
ii  libgtk-3-dev:amd64         3.24.33-1ubuntu2  amd64        development files for the GTK library
ii  libgtk-4-1:amd64           4.6.6+ds-0ubuntu1 amd64        GTK graphical user interface library
ii  libgtk-4-bin               4.6.6+ds-0ubuntu1 amd64        programs for the GTK graphical user interface library
ii  libgtk-4-common            4.6.6+ds-0ubuntu1 all          common files for the GTK graphical user interface library
ii  libgtk-4-dev:amd64         4.6.6+ds-0ubuntu1 amd64        development files for the GTK library
ii  libgtk2.0-0:amd64          2.24.33-2ubuntu2  amd64        GTK graphical user interface library - old version
ii  libgtk2.0-bin              2.24.33-2ubuntu2  amd64        programs for the GTK graphical user interface library
ii  libgtk2.0-common           2.24.33-2ubuntu2  all          common files for the GTK graphical user interface library
ii  libgtk2.0-dev:amd64        2.24.33-2ubuntu2  amd64        development files for the GTK library
ii  libgtk2.0-doc              2.24.33-2ubuntu2  all          documentation for the GTK graphical user interface library
ii  libgtk3-perl               0.038-1           all          Perl bindings for the GTK+ graphical user interface library
ii  libgtk3.0-cil              2.99.3-4build1    amd64        CLI binding for the GTK+ 3.0 toolkit
ii  libgtk3.0-cil-dev          2.99.3-4build1    amd64        CLI binding for the GTK+ toolkit 3.0
Guile 3.0 vers.

leguizar, 2023-03-24:
Sorry, Many thanks in advance.

vzh, 2023-03-24:
@leguizar, could you please follow this debug instruction:
https://lepton-eda.github.io/lepton-manual.html/Debug-Glib-and-GTK.html

After you started gdb, run the program by hitting `r` `Enter`, and then, if it segfaults, use the command `bt` and hit `Enter` to get the backtrace.  I'd like to see the backtrace, maybe it can provide us with some clues.

Another thought, if you installed lepton from sources, you could try to remove it, and look if `/usr/local` has any paths/files rest after removing it.  Sometimes they can influence the program action.

vzh, 2023-03-24:
BTW, I've not yet heard any report mentioned Wayland.  JFTR.

leguizar, 2023-03-25:
Hello,
Running the program it appears the following:

> guile -s /usr/local/bin/lepton-schematic
GNU gdb (Ubuntu 12.1-0ubuntu1~22.04) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from guile...
(No debugging symbols found in guile)
(gdb) r
Starting program: /usr/bin/guile -s /usr/local/bin/lepton-schematic
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff7bab8f2 in GC_find_limit_with_bound () from /lib/x86_64-linux-gnu/libgc.so.1
(gdb) bt
#0  0x00007ffff7bab8f2 in GC_find_limit_with_bound ()
   from /lib/x86_64-linux-gnu/libgc.so.1
#1  0x00007ffff7bab9da in GC_init_linux_data_start ()
   from /lib/x86_64-linux-gnu/libgc.so.1
#2  0x00007ffff7bacf67 in GC_init () from /lib/x86_64-linux-gnu/libgc.so.1
#3  0x00007ffff7ebac8e in ?? () from /lib/x86_64-linux-gnu/libguile-3.0.so.1
#4  0x00007ffff7f1e90f in ?? () from /lib/x86_64-linux-gnu/libguile-3.0.so.1
#5  0x00007ffff7f1ecde in ?? () from /lib/x86_64-linux-gnu/libguile-3.0.so.1
#6  0x00007ffff7bab80b in GC_call_with_stack_base ()
   from /lib/x86_64-linux-gnu/libgc.so.1
#7  0x00007ffff7f19dbc in scm_with_guile ()
   from /lib/x86_64-linux-gnu/libguile-3.0.so.1
#8  0x00007ffff7ebabd9 in scm_boot_guile ()
   from /lib/x86_64-linux-gnu/libguile-3.0.so.1
#9  0x0000555555555193 in ?? ()
#10 0x00007ffff7c29d90 in __libc_start_call_main (
    main=main@entry=0x555555555130, argc=argc@entry=3, 
    argv=argv@entry=0x7fffffffdfd8)
    at ../sysdeps/nptl/libc_start_call_main.h:58
#11 0x00007ffff7c29e40 in __libc_start_main_impl (main=0x555555555130, argc=3, 
    argv=0x7fffffffdfd8, init=<optimized out>, fini=<optimized out>, 
    rtld_fini=<optimized out>, stack_end=0x7fffffffdfc8)
    at ../csu/libc-start.c:392
--Type <RET> for more, q to quit, c to continue without paging--c
#12 0x0000555555555225 in ?? ()

I'm sorry for the delay, and Many Thanks in advance

leguizar, 2023-03-25:
Another thought, if you installed lepton from sources, you could try to remove it, and look if /usr/local has any paths/files rest after removing it. Sometimes they can influence the program action.
Ok, I will try to remove and see what happens
Best Regards

leguizar, 2023-03-27:
Hello, good morning.
I did make uninstall and remove from /usr/local/ all the existing components of lepton. Then I make the installation again and the problem still remain, the I ran the debuger gdb and get the following messages:

**(gdb) r**
Starting program: /usr/bin/guile -s /usr/local/bin/lepton-schematic
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff7bab8f2 in GC_find_limit_with_bound () from /lib/x86_64-linux-gnu/libgc.so.1

**(gdb) bt**
#0  0x00007ffff7bab8f2 in GC_find_limit_with_bound () from /lib/x86_64-linux-gnu/libgc.so.1
#1  0x00007ffff7bab9da in GC_init_linux_data_start () from /lib/x86_64-linux-gnu/libgc.so.1
#2  0x00007ffff7bacf67 in GC_init () from /lib/x86_64-linux-gnu/libgc.so.1
#3  0x00007ffff7ebbc8e in ?? () from /lib/x86_64-linux-gnu/libguile-3.0.so.1
#4  0x00007ffff7f1f90f in ?? () from /lib/x86_64-linux-gnu/libguile-3.0.so.1
#5  0x00007ffff7f1fcde in ?? () from /lib/x86_64-linux-gnu/libguile-3.0.so.1
#6  0x00007ffff7bab80b in GC_call_with_stack_base () from /lib/x86_64-linux-gnu/libgc.so.1
#7  0x00007ffff7f1adbc in scm_with_guile () from /lib/x86_64-linux-gnu/libguile-3.0.so.1
#8  0x00007ffff7ebbbd9 in scm_boot_guile () from /lib/x86_64-linux-gnu/libguile-3.0.so.1
#9  0x0000555555555193 in ?? ()
#10 0x00007ffff7c29d90 in __libc_start_call_main (main=main@entry=0x555555555130, 
    argc=argc@entry=3, argv=argv@entry=0x7fffffffdf38) at ../sysdeps/nptl/libc_start_call_main.h:58
#11 0x00007ffff7c29e40 in __libc_start_main_impl (main=0x555555555130, argc=3, argv=0x7fffffffdf38, 
    init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffdf28)
    at ../csu/libc-start.c:392
#12 0x0000555555555225 in ?? ()
Kind Regards and Thank you so much Большое спасибо

vzh, 2023-03-27:
From what I've seen so far, I can say two things:
- The program segfaults on garbage collection stage (all those `GC_*` calls).
- Some widget gets unreachable for the program while it still tries to work with it (`invalid uninstantiatable type '(null)' in cast to 'GtkWidget'`), and the widget uses `GtkTreeView` or its derivatives.

Well, AFAIK, guile won't garbage collect any objects that are alive as top level C pointer since garbage collector scans heap for such objects and marks the memory for those objects as still in use.

First it would be helpful to localize the issue by narrowing down it.  You've mentioned that you hit 'e e' and some time after that the crash happens.  Could you please to try to not close the MultiAttrib dialog during your editing session and look if the crash still happens?  (Other widgets that work with `GtkTreeView` are Close confirmation dialog, Find text state, Hotkey dialog, Page manager, Component selector, and Major refdes change dialog, as I can see from the code.)

vzh, 2023-03-27:
@leguizar, please report what the command `guile --version` outputs.
Another question: do you use `--with-gtk3` on the `./configure` stage when compiling the program?

leguizar, 2023-03-27:
First, running the program with the MultiAttrib dialog open, editing a complete symbol, there's no crash.

guile (GNU Guile) 3.0.7
Packaged by Debian (3.0.7-deb+3.0.7-1)
Copyright (C) 2021 Free Software Foundation, Inc.
To your another question, no I didn't use the command --with-gtk3 when I compiled the program, Should I?
Regards

vzh, 2023-03-27:
> Should I?
Oh no, I was just curious what gtk version crashes as you showed you have several versions.

OK, it seems the issue is related to the MultiAttrib dialog.

Let's try to localize it a bit more.  Could you try to install guile 3.0.8 (or even 3.0.9)? The version has several bugs fixed as compared with 3.0.7.  You could also try guile 2.2.  In the latter case you have to remove guile cache, as we discussed above, and remove guile 3.0 development package, replacing it with one for guile 2.2.

graahnul-grom, 2023-03-28:
https://hboehm.info/gc/debugging.html
https://gitlab.alpinelinux.org/alpine/aports/-/issues/11447

`boehm-gc`(garbage collector used by Guile, GC_XXX)
uses `SIGSEGV` internally, so the actual error could be elsewhere.

Please try to run "handle SIGSEGV SIGBUS nostop noprint"
in the `gdb` prompt before issuing the `run` command.

leguizar, 2023-04-09:
I'm sorry for the delay, but I had removed some packages that I should not, and now I'm dealing to install glb2.38.

graahnul-grom, 2023-04-11:
@leguizar How can we help you?

> now I'm dealing to install glb2.38

This is, probably, a typo.
What package you are trying to install?

leguizar, 2023-04-12:
Hi, I am trying to install glib-2.76.1, I have some issues as follows when I run meson so I trying to find if the "NO" in the compilation could cause the problem with the Lepton, there are some lines that I typed in bold which are worrying me,
The Meson build system
Version: 0.61.2
Source dir: /home/ramiro/Downloads/glib-2.76.1
Build dir: /home/ramiro/Downloads/glib-2.76.1/_build
Build type: native build
Project name: glib
Project version: 2.76.1
C compiler for the host machine: cc (gcc 11.3.0 "cc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0")
C linker for the host machine: cc ld.bfd 2.38
Host machine cpu family: x86_64
Host machine cpu: x86_64
Compiler for C supports arguments -std=c90: YES 
Compiler for C supports arguments -std=c99: YES 
Compiler for C supports arguments -std=c11: YES 
Compiler for C supports arguments -std=c17: YES 
C++ compiler for the host machine: c++ (gcc 11.3.0 "c++ (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0")
C++ linker for the host machine: c++ ld.bfd 2.38
Compiler for C++ supports arguments -std=c++98: YES 
Compiler for C++ supports arguments -std=c++03: YES 
Compiler for C++ supports arguments -std=c++11: YES 
Compiler for C++ supports arguments -std=c++14: YES 
Compiler for C++ supports arguments -std=c++17: YES 
Compiler for C++ supports arguments -std=c++20: YES 
Compiler for C++ supports arguments -std=c++2b: YES 
**_Compiler for C++ supports arguments -std=c++latest: NO_** 
Program valgrind found: YES (/usr/bin/valgrind)
Compiler for C supports arguments -fno-strict-aliasing: YES 
Message: Disabling cast checks
Has header "alloca.h" : YES 
Has header "afunix.h" : NO 
Has header "crt_externs.h" : NO 
Has header "dirent.h" : YES 
Has header "float.h" : YES 
Has header "fstab.h" : YES 
Has header "grp.h" : YES 
Has header "inttypes.h" : YES 
Has header "limits.h" : YES 
Has header "locale.h" : YES 
Has header "mach/mach_time.h" : NO 
Has header "memory.h" : YES 
Has header "mntent.h" : YES 
Has header "poll.h" : YES 
Has header "pwd.h" : YES 
Has header "sched.h" : YES 
Has header "spawn.h" : YES 
Has header "stdatomic.h" : YES 
Has header "stdint.h" : YES 
Has header "stdlib.h" : YES 
Has header "string.h" : YES 
Has header "strings.h" : YES 
Has header "sys/auxv.h" : YES 
Has header "sys/event.h" : NO 
Has header "sys/filio.h" : NO 
Has header "sys/inotify.h" : YES 
**_Has header "sys/mkdev.h" : NO 
**Has header "sys/mntctl.h" : NO** 
**Has header "sys/mnttab.h" : NO_** 
Has header "sys/mount.h" : YES 
Has header "sys/param.h" : YES 
Has header "sys/resource.h" : YES 
Has header "sys/select.h" : YES 
Has header "sys/statfs.h" : YES 
Has header "sys/stat.h" : YES 
Has header "sys/statvfs.h" : YES 
Has header "sys/sysctl.h" : NO 
Has header "sys/time.h" : YES 
Has header "sys/times.h" : YES 
Has header "sys/types.h" : YES 
Has header "sys/uio.h" : YES 
[... 352 lines cut ...]
  Options
    selinux            : False
    libmount           : False
    xattr              : False
    man                : False
    dtrace             : False
    systemtap          : False
    sysprof            : False
    gtk_doc            : False
    bsymbolic_functions: True
    force_posix_threads: False
    tests              : True
    installed_tests    : False
    nls                : auto
    oss_fuzz           : disabled
    glib_debug         : auto
    glib_assert        : True
    glib_checks        : True
    libelf             : auto
    multiarch          : False

  Subprojects
    gvdb               : YES
Well I am working on that just to be sure that glib are going to work fine. 
Regards

graahnul-grom, 2023-04-12:
@leguizar
The messages you emphasized are harmless.
The question is why are you building `glib` yourself,
in the first place? Your distribution (Ubuntu 22.04.2)
provides the `glib 2.72.4` package, which is OK for Lepton.
Do you have any problems installing it from Ubuntu repositories? Or with `lepton-eda` installation?
Did you try to run `lepton-schematic` in `gdb` using the commands I suggested in my previous message, i.e.:
```
(gdb) handle SIGSEGV SIGBUS nostop noprint
(gdb) run
```

closed 2023-10-04 by vzh