lepton-eda/lepton-eda #1016: LM317 has pinseq visible instead of pinnumber

cestrauss, 2023-04-20:
I'm using Lepton EDA 1.9.18.

Please try:

1) Place an LM317-1 from the default library on an schematic
2) Hierarchy -> Down symbol
3) Double-click the "2" value on the Vout pin

You see it edits the "pinseq" attribute. The actual pinnumber attribute is hidden.

I noticed it because I created an LM317LM, which has multiple Vout pins, by copying the Vout pin and changing the visible number on the pin in each instance. The netlist was wrong, because what I was actually editing was pinseq, instead of pinnumber.

The other pins seem OK.

vzh, 2023-04-21:
Hi Cesar,
Thank you for the report! I attached a new version in that `pinseq` and `pinnumber` were swapped and I added missing `symversion` (github requires the file has to have one of the extensions its owner allows, hence `.txt`).  Please report if the symbol suits you.  If you want some additional, maybe artistic changes, or if you have your own better looking version we could use in the library, I would be very glad to see it here.
[lm317-1.sym.txt](https://github.com/lepton-eda/lepton-eda/files/11298031/lm317-1.sym.txt)

cestrauss, 2023-05-07:
Looking good! The new version does fix the issue I found.
Sorry, I don't have any artistic suggestion at this time. I'm not really using that exact symbol (the package I use has four Vout pins) but I'm glad to see the standard library become more correct.

Regards,
Cesar

PR #1019 (2023-05-10): symbols: Fix lm317-1.sym.
The pin 2 of the symbol has visible attribute 'pinseq=' instead of 'pinnumber=', which has been changed.  The missing attribute 'symversion=' has been added.

Closes #1016.
closed 2023-05-10 by vzh