lepton-eda/lepton-eda #1020: Segfault after multiple opening/closing of Edit Attributes dialog box

djcardinale, 2023-05-06:
Sequence:
1) Double click a symbol to open Edit Attributes dialog box.
2) Close dialog box
3) Double click same (or other symbol) to open dialog box.
4) Close dialog box
5) Double click again: immediate segfault.

Lepton versions tried with exact same results:

1.9.16 (git: 13cbc49)  from Ubuntu Unity 22.04

$ /usr/bin/lepton-cli --version
Lepton EDA 1.9.16.20210731 (git: 13cbc49)
Copyright (C) 1998-2016 gEDA developers
Copyright (C) 2017-2021 Lepton EDA developers
This is free software, and you are welcome to redistribute it
under certain conditions. For details, see the file `COPYING',
which is included in the Lepton EDA distribution.
There is NO WARRANTY, to the extent permitted by law.

1.9.18 (git: d647623) built from git repository

$ lepton-cli --version
;;; note: auto-compilation is enabled, set GUILE_AUTO_COMPILE=0
;;;       or pass the --no-auto-compile argument to disable.
;;; compiling /usr/local/bin/lepton-cli
;;; /usr/local/bin/lepton-cli:47:2: warning: non-literal format string
;;; /usr/local/bin/lepton-cli:54:4: warning: non-literal format string
;;; /usr/local/bin/lepton-cli:63:2: warning: non-literal format string
;;; /usr/local/bin/lepton-cli:91:2: warning: non-literal format string
;;; /usr/local/bin/lepton-cli:116:9: warning: non-literal format string
;;; compiled /home/dennis/.cache/guile/ccache/2.2-LE-8-3.A/usr/local/bin/lepton-cli.go
Lepton EDA/lepton-cli 1.9.18.20220529 (git: d647623)
Copyright (C) 1998-2016 gEDA developers
Copyright (C) 2017-2022 Lepton EDA developers
This is free software, and you are welcome to redistribute it
under certain conditions. For details, see the file `COPYING',
which is included in the Lepton EDA distribution.
There is NO WARRANTY, to the extent permitted by law.

From syslog:

May  6 00:31:22 dlenovo kernel: [12629.343405] guile-2.2[57712]: segfault at 3c00 ip 00007f38b8c238c9 sp 00007ffee8726e90 error 4 in libgobject-2.0.so.0.7200.4[7f38b8bf9000+33000]
May  6 00:31:22 dlenovo kernel: [12629.343418] Code: 00 f3 0f 1e fa 41 55 41 54 49 89 fc 55 53 48 83 ec 08 48 85 ff 0f 84 d0 00 00 00 48 8b 07 48 89 f3 48 85 c0 0f 84 d7 00 00 00 <48> 8b 28 48 81 fd fc 03 00 00 0f 86 37 01 00 00 48 89 ef 48 83 e7'

From output of lepton-schematic -v:

(guile-2.2:57822): GLib-GObject-WARNING **: 00:32:54.228: invalid unclassed pointer in cast to 'GtkToggleButton'

(guile-2.2:57822): Gtk-CRITICAL **: 00:32:54.228: IA__gtk_toggle_button_get_active: assertion 'GTK_IS_TOGGLE_BUTTON (toggle_button)' failed

(guile-2.2:57822): Gtk-CRITICAL **: 00:32:54.228: IA__gtk_tree_view_get_model: assertion 'GTK_IS_TREE_VIEW (tree_view)' failed

(guile-2.2:57822): Gtk-CRITICAL **: 00:32:54.228: IA__gtk_list_store_clear: assertion 'GTK_IS_LIST_STORE (list_store)' failed

(guile-2.2:57822): Gtk-CRITICAL **: 00:32:54.228: IA__gtk_list_store_append: assertion 'GTK_IS_LIST_STORE (list_store)' failed

(guile-2.2:57822): Gtk-CRITICAL **: 00:32:54.228: IA__gtk_list_store_set_valist: assertion 'GTK_IS_LIST_STORE (list_store)' failed

(guile-2.2:57822): Gtk-CRITICAL **: 00:32:54.228: IA__gtk_list_store_append: assertion 'GTK_IS_LIST_STORE (list_store)' failed

(guile-2.2:57822): Gtk-CRITICAL **: 00:32:54.228: IA__gtk_list_store_set_valist: assertion 'GTK_IS_LIST_STORE (list_store)' failed

(guile-2.2:57822): Gtk-CRITICAL **: 00:32:54.228: IA__gtk_list_store_append: assertion 'GTK_IS_LIST_STORE (list_store)' failed

(guile-2.2:57822): Gtk-CRITICAL **: 00:32:54.228: IA__gtk_list_store_set_valist: assertion 'GTK_IS_LIST_STORE (list_store)' failed

(guile-2.2:57822): Gtk-CRITICAL **: 00:32:54.228: IA__gtk_list_store_append: assertion 'GTK_IS_LIST_STORE (list_store)' failed

(guile-2.2:57822): Gtk-CRITICAL **: 00:32:54.228: IA__gtk_list_store_set_valist: assertion 'GTK_IS_LIST_STORE (list_store)' failed
[... 454 lines cut ...]

(guile-2.2:57822): Gtk-CRITICAL **: 00:32:56.678: IA__gtk_list_store_set_valist: assertion 'GTK_IS_LIST_STORE (list_store)' failed

(guile-2.2:57822): Gtk-CRITICAL **: 00:32:56.678: IA__gtk_list_store_append: assertion 'GTK_IS_LIST_STORE (list_store)' failed

(guile-2.2:57822): Gtk-CRITICAL **: 00:32:56.678: IA__gtk_list_store_set_valist: assertion 'GTK_IS_LIST_STORE (list_store)' failed

(guile-2.2:57822): Gtk-CRITICAL **: 00:32:56.678: IA__gtk_list_store_append: assertion 'GTK_IS_LIST_STORE (list_store)' failed

(guile-2.2:57822): Gtk-CRITICAL **: 00:32:56.678: IA__gtk_list_store_set_valist: assertion 'GTK_IS_LIST_STORE (list_store)' failed

(guile-2.2:57822): Gtk-CRITICAL **: 00:32:56.678: IA__gtk_list_store_append: assertion 'GTK_IS_LIST_STORE (list_store)' failed

(guile-2.2:57822): Gtk-CRITICAL **: 00:32:56.678: IA__gtk_list_store_set_valist: assertion 'GTK_IS_LIST_STORE (list_store)' failed

(guile-2.2:57822): Gtk-CRITICAL **: 00:32:56.678: IA__gtk_list_store_append: assertion 'GTK_IS_LIST_STORE (list_store)' failed

(guile-2.2:57822): Gtk-CRITICAL **: 00:32:56.678: IA__gtk_list_store_set_valist: assertion 'GTK_IS_LIST_STORE (list_store)' failed

(guile-2.2:57822): GLib-GObject-WARNING **: 00:32:56.678: invalid uninstantiatable type '(null)' in cast to 'GtkWidget'

(guile-2.2:57822): Gtk-CRITICAL **: 00:32:56.678: IA__gtk_widget_set_sensitive: assertion 'GTK_IS_WIDGET (widget)' failed
Segmentation fault

...

vzh, 2023-05-06:
@djcardinale, thank you for the detailed report! It looks similar to the second issue discussed at the page of the issue #1010.  There are some hints on how to debug it there.  Looks like something related to garbage collection.  I couldn't reproduce it on my PC yet, though.

djcardinale, 2023-05-06:
Here is the output after rm -Rf ~/.cache/guile:

$ lepton-schematic -v
;;; note: auto-compilation is enabled, set GUILE_AUTO_COMPILE=0
;;;       or pass the --no-auto-compile argument to disable.
;;; compiling /usr/local/bin/lepton-schematic
;;; compiling /usr/local/share/lepton-eda/scheme/lepton/m4.scm
;;; compiled /home/dennis/.cache/guile/ccache/2.2-LE-8-3.A/usr/local/share/lepton-eda/scheme/lepton/m4.scm.go
;;; compiling /usr/local/share/lepton-eda/scheme/lepton/color-map.scm
;;; compiling /usr/local/share/lepton-eda/scheme/lepton/ffi.scm
;;; compiling /usr/local/share/lepton-eda/scheme/lepton/ffi/lib.scm
;;; compiled /home/dennis/.cache/guile/ccache/2.2-LE-8-3.A/usr/local/share/lepton-eda/scheme/lepton/ffi/lib.scm.go
;;; compiling /usr/local/share/lepton-eda/scheme/lepton/ffi/lff.scm
;;; compiled /home/dennis/.cache/guile/ccache/2.2-LE-8-3.A/usr/local/share/lepton-eda/scheme/lepton/ffi/lff.scm.go
;;; compiled /home/dennis/.cache/guile/ccache/2.2-LE-8-3.A/usr/local/share/lepton-eda/scheme/lepton/ffi.scm.go
;;; compiling /usr/local/share/lepton-eda/scheme/lepton/gettext.scm
;;; compiled /home/dennis/.cache/guile/ccache/2.2-LE-8-3.A/usr/local/share/lepton-eda/scheme/lepton/gettext.scm.go
;;; compiling /usr/local/share/lepton-eda/scheme/lepton/log.scm
;;; compiling /usr/local/share/lepton-eda/scheme/lepton/ffi/glib.scm
;;; compiled /home/dennis/.cache/guile/ccache/2.2-LE-8-3.A/usr/local/share/lepton-eda/scheme/lepton/ffi/glib.scm.go
;;; compiled /home/dennis/.cache/guile/ccache/2.2-LE-8-3.A/usr/local/share/lepton-eda/scheme/lepton/log.scm.go
;;; compiled /home/dennis/.cache/guile/ccache/2.2-LE-8-3.A/usr/local/share/lepton-eda/scheme/lepton/color-map.scm.go
;;; compiling /usr/local/share/lepton-eda/scheme/lepton/config.scm
;;; compiling /usr/local/share/lepton-eda/scheme/lepton/ffi/boolean.scm
;;; compiled /home/dennis/.cache/guile/ccache/2.2-LE-8-3.A/usr/local/share/lepton-eda/scheme/lepton/ffi/boolean.scm.go
;;; compiling /usr/local/share/lepton-eda/scheme/lepton/ffi/check-args.scm
;;; compiled /home/dennis/.cache/guile/ccache/2.2-LE-8-3.A/usr/local/share/lepton-eda/scheme/lepton/ffi/check-args.scm.go
;;; compiling /usr/local/share/lepton-eda/scheme/lepton/gerror.scm
;;; compiled /home/dennis/.cache/guile/ccache/2.2-LE-8-3.A/usr/local/share/lepton-eda/scheme/lepton/gerror.scm.go
;;; compiled /home/dennis/.cache/guile/ccache/2.2-LE-8-3.A/usr/local/share/lepton-eda/scheme/lepton/config.scm.go
;;; compiling /usr/local/share/lepton-eda/scheme/lepton/eval.scm
;;; lepton/eval.scm:46:7: warning: non-literal format string
;;; compiled /home/dennis/.cache/guile/ccache/2.2-LE-8-3.A/usr/local/share/lepton-eda/scheme/lepton/eval.scm.go
;;; compiling /usr/local/share/lepton-eda/scheme/lepton/file-system.scm
;;; compiled /home/dennis/.cache/guile/ccache/2.2-LE-8-3.A/usr/local/share/lepton-eda/scheme/lepton/file-system.scm.go
;;; compiling /usr/local/share/lepton-eda/scheme/lepton/os.scm
;;; compiled /home/dennis/.cache/guile/ccache/2.2-LE-8-3.A/usr/local/share/lepton-eda/scheme/lepton/os.scm.go
;;; compiling /usr/local/share/lepton-eda/scheme/lepton/object/foreign.scm
;;; compiled /home/dennis/.cache/guile/ccache/2.2-LE-8-3.A/usr/local/share/lepton-eda/scheme/lepton/object/foreign.scm.go
;;; compiling /usr/local/share/lepton-eda/scheme/lepton/page/foreign.scm
;;; compiled /home/dennis/.cache/guile/ccache/2.2-LE-8-3.A/usr/local/share/lepton-eda/scheme/lepton/page/foreign.scm.go
;;; compiling /usr/local/share/lepton-eda/scheme/lepton/srfi-37.scm
;;; lepton/srfi-37.scm:183:26: warning: non-literal format string
;;; lepton/srfi-37.scm:183:26: warning: non-literal format string
;;; lepton/srfi-37.scm:216:23: warning: non-literal format string
;;; lepton/srfi-37.scm:216:23: warning: non-literal format string
;;; lepton/srfi-37.scm:220:23: warning: non-literal format string
;;; lepton/srfi-37.scm:220:23: warning: non-literal format string
;;; compiled /home/dennis/.cache/guile/ccache/2.2-LE-8-3.A/usr/local/share/lepton-eda/scheme/lepton/srfi-37.scm.go
;;; compiling /usr/local/share/lepton-eda/scheme/lepton/version.scm
;;; compiled /home/dennis/.cache/guile/ccache/2.2-LE-8-3.A/usr/local/share/lepton-eda/scheme/lepton/version.scm.go
;;; compiling /usr/local/share/lepton-eda/scheme/schematic/gettext.scm
;;; compiled /home/dennis/.cache/guile/ccache/2.2-LE-8-3.A/usr/local/share/lepton-eda/scheme/schematic/gettext.scm.go
;;; compiling /usr/local/share/lepton-eda/scheme/schematic/ffi.scm
;;; compiled /home/dennis/.cache/guile/ccache/2.2-LE-8-3.A/usr/local/share/lepton-eda/scheme/schematic/ffi.scm.go
;;; compiling /usr/local/share/lepton-eda/scheme/schematic/ffi/gtk.scm
;;; compiled /home/dennis/.cache/guile/ccache/2.2-LE-8-3.A/usr/local/share/lepton-eda/scheme/schematic/ffi/gtk.scm.go
;;; compiling /usr/local/share/lepton-eda/scheme/schematic/gui/keymap.scm
;;; compiling /usr/local/share/lepton-eda/scheme/schematic/action.scm
;;; compiling /usr/local/share/lepton-eda/scheme/schematic/hook.scm
;;; compiled /home/dennis/.cache/guile/ccache/2.2-LE-8-3.A/usr/local/share/lepton-eda/scheme/schematic/hook.scm.go
;;; compiling /usr/local/share/lepton-eda/scheme/schematic/window.scm
;;; compiling /usr/local/share/lepton-eda/scheme/lepton/page.scm
;;; compiling /usr/local/share/lepton-eda/scheme/lepton/object/type.scm
;;; compiled /home/dennis/.cache/guile/ccache/2.2-LE-8-3.A/usr/local/share/lepton-eda/scheme/lepton/object/type.scm.go
;;; compiling /usr/local/share/lepton-eda/scheme/lepton/toplevel/foreign.scm
[... cut ...]
(guile-2.2:65396): Gtk-CRITICAL **: 09:27:22.294: IA__gtk_widget_modify_text: assertion 'GTK_IS_WIDGET (widget)' failed

(guile-2.2:65396): GLib-GObject-WARNING **: 09:27:26.324: invalid unclassed pointer in cast to 'GtkToggleButton'

(guile-2.2:65396): Gtk-CRITICAL **: 09:27:26.324: IA__gtk_toggle_button_get_active: assertion 'GTK_IS_TOGGLE_BUTTON (toggle_button)' failed

(guile-2.2:65396): GLib-GObject-WARNING **: 09:27:26.325: invalid unclassed pointer in cast to 'GtkWidget'

(guile-2.2:65396): Gtk-CRITICAL **: 09:27:26.325: IA__gtk_widget_set_sensitive: assertion 'GTK_IS_WIDGET (widget)' failed

graahnul-grom, 2023-05-08:
Cannot reproduce this on Ubuntu 22.04.2 in `VirtualBox`
(Wayland or X11). Cannot reproduce #1010 either.

djcardinale, 2023-05-09:
So, what's the next step?  How can I help.
 
Currently,  I cannot use lepton-schematic on my laptop (Ubuntu Unity).  The segfault happens _every_  time I use the EA dialog box on either version.

I even deleted my entire home directory thinking something somewhere was corrupted.   No Joy.

Otherwise, working fine on two different Ubuntu Budgie desktops.  Having it on my laptop would be nice.

graahnul-grom, 2023-05-10:
@djcardinale
The best you can do is to run `lepton-schematic` with `gdb`
on your laptop and get a backtrace. Install it with
`sudo apt-get install gdb` and run (assuming `Lepton EDA`
is installed to `/usr/local/`; correct if needed):

```
$ gdb --args guile -s /usr/local/bin/lepton-schematic
```

Then at the `gdb` prompt, "(gdb)", issue the following commands:

```
(gdb) set logging file /tmp/gdb.log
(gdb) set logging enabled on
(gdb) handle SIGSEGV SIGBUS nostop noprint
(gdb) run
```

When `lepton-schematic` crashes, you are back at the `gdb`
prompt; run these commands:

```
(gdb) backtrace
(gdb) quit
```

If everything's fine, the file `/tmp/gdb.log` would contain
the recorded `gdb` session.
Please attach it to your comment here.

tomdean1939, 2023-09-20:
> gdb --args guile -s /usr/local/bin/lepton-schematic      
GNU gdb (Ubuntu 12.0.90-0ubuntu1) 12.0.90
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from guile...
(gdb) r
Starting program: /usr/local/bin/guile -s /usr/local/bin/lepton-schematic
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff7bab8f2 in GC_find_limit_with_bound () from /lib/x86_64-linux-gnu/libgc.so.1
(gdb) bt
#0  0x00007ffff7bab8f2 in GC_find_limit_with_bound () from /lib/x86_64-linux-gnu/libgc.so.1
#1  0x00007ffff7bab9da in GC_init_linux_data_start () from /lib/x86_64-linux-gnu/libgc.so.1
#2  0x00007ffff7bacf67 in GC_init () from /lib/x86_64-linux-gnu/libgc.so.1
#3  0x00007ffff7e984fe in scm_storage_prehistory () at ../../libguile/gc.c:465
#4  0x00007ffff7ea1f6f in scm_i_init_guile (base=0x7fffffffdac0) at ../../libguile/init.c:361
#5  0x00007ffff7f0495f in scm_i_init_thread_for_guile (base=base@entry=0x7fffffffdac0, dynamic_state=dynamic_state@entry=0x0)
    at ../../libguile/threads.c:570
#6  0x00007ffff7f04a7e in scm_i_init_thread_for_guile (dynamic_state=0x0, base=0x7fffffffdac0) at ../../libguile/threads.c:653
#7  with_guile (base=0x7fffffffdac0, data=0x7fffffffdaf0) at ../../libguile/threads.c:638
#8  0x00007ffff7bab80b in GC_call_with_stack_base () from /lib/x86_64-linux-gnu/libgc.so.1
#9  0x00007ffff7f04e9c in scm_i_with_guile (dynamic_state=<optimized out>, data=0x7fffffffdaf0, func=0x7ffff7ea1d50 <invoke_main_func>)
    at ../../libguile/threads.c:688
#10 scm_with_guile (func=func@entry=0x7ffff7ea1d50 <invoke_main_func>, data=data@entry=0x7fffffffdb20) at ../../libguile/threads.c:694
#11 0x00007ffff7ea1f16 in scm_boot_guile (argc=argc@entry=3, argv=argv@entry=0x7fffffffdcb8, 
    main_func=main_func@entry=0x555555555320 <inner_main>, closure=closure@entry=0x0) at ../../libguile/init.c:296
#12 0x00005555555551aa in main (argc=3, argv=0x7fffffffdcb8) at ../../libguile/guile.c:94

tomdean1939, 2023-09-20:
I removed guile-2.2-libs and not only have 
> dpkg -l | grep guile
ii  guile-3.0-libs:amd64                         3.0.7-1                                 amd64        Core Guile libraries
configure, make, and install without error.

> gdb --args guile -s /usr/local/bin/lepton-schematic
GNU gdb (Ubuntu 12.0.90-0ubuntu1) 12.0.90
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from guile...
(gdb) r
Starting program: /usr/local/bin/guile -s /usr/local/bin/lepton-schematic
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff7bab8f2 in GC_find_limit_with_bound () from /lib/x86_64-linux-gnu/libgc.so.1

So, it appears the problem is in garbage colledtion.

vzh, 2023-09-21:
Hi Tom,
Could you please try commands proposed by @graahnul-grom  above before issuing `r` in the *gdb* prompt?

vzh, 2023-09-21:
I mean, we know that garbage collection leads to _SIGSEGV_, we have to know which data are affected under the hood to localize the issue.

vzh, 2023-09-21:
@tomdean1939, your issue may be related to https://github.com/ivmai/bdwgc/issues/409 (and see the links there for more).  I cannot just point you to related link on the guile-user mailing list yet 'cause it is unavailable for me today for some reason.  It looks like your issue is specific for some versions of `bdwgc` that went to several popular repositories (under the name `libgc` or something like) and may be unrelated to the issue @djcardinale wrote about here.

tomdean1939, 2023-09-21:
lepton-schematic started, the window opened, and everything seemed OK. 
Open a schematic seemed OK.
Double clicking on a symbol, close, repeat caused the segfault and exit.
[gdb.log](https://github.com/lepton-eda/lepton-eda/files/12690949/gdb.log)

tomdean

tomdean1939, 2023-09-21:
> to ivmai/bdwgc#409 <https://github.com/ivmai/bdwgc/issues/409> (and see

I managed to get a coredump. Been a long time since I looked at one.

The segfault is at 0x00007fadfd14d8c9. Looking at ProcMaps, this seems 
to be in
7fadfd123000-7fadfd156000 r-xp 0000f000 103:04 8656231 \
      /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.7200.4

Maybe I am reading this wrong???

Tom Dean

vzh, 2023-09-22:
Hi Tom.
Full backtrace would be more helpful, I believe.
IIUC, you've hit the same issue the topic starter wrote about.  Double click on a symbol opens Multiattrib dialog, right?

tomdean1939, 2023-09-22:
Yes, the double click did open a Multiattrib dialog.
The sequence of actions was:
   start lepton-schematic from the command line
   add a capacitor from basic components
   double click on the capacitor
   close dialog
   double click on the capacitor
   -- segfault (and core dump)

I could not figure out how to get a full backtrace from the core dump.
The core dump seems to point to a different library than the
'gdb lepton-schemaitc' backtrace.

if I start lepton-schematic with gdb,
   gdb --args guile -s /usr/local/bin/lepton-schematic
and do the same actions as above, I get a segfault and the backtrace 
points to libgc.so.1. I posted this earlier.

If I start lepton-schematic from the command line and do the same 
actions as above, I get a segfault and a core dump. Starting gdb on the 
core dump shows the segfault address as 0x00007fadfd14d8c9. ProcMaps 
shows this to be in the address range of libgobject-2.0.so.0.7200.4.

I do not understand the difference.

How do I get a full backtrace from the coredump with symbols? Since I am 
using Ubuntu 22.04, I have to use 'apport-unpack' to extract information 
from the raw core dump file.

Tom Dean

vzh, 2023-09-25:
Hi Tom, @djcardinale.
Could you please check if the attached patch makes any difference in the behavior of `lepton-schematic`?
[0001-gui-Don-t-destroy-Multiattrib-widget-on-close-and-de.patch.txt](https://github.com/lepton-eda/lepton-eda/files/12717777/0001-gui-Don-t-destroy-Multiattrib-widget-on-close-and-de.patch.txt)
(While GH suggests `*.patch` as a supported type for attachments, it doesn't really support the extension :-/ so I renamed the patch to `*.txt`).

tomdean1939, 2023-10-02:
Sorry, I was away...
Yes, this seems to fix the multi0open-close problem
Tom Dean

tomdean1939, 2023-10-02:
After 30 minutes of editing, no crashes. I think the patch can be applied and the issue closed.
Thanks,

Tom Dean

closed 2023-10-04 by vzh