lepton-eda/lepton-eda #1021: Custom Color Maps?

djcardinale, 2023-05-08:
I'm trying to interpret this page:

https://graahnul-grom.github.io/cs/index.html

Added to gschemrc in project directory:

( primitive-load "./mycolormap" )

and copied darkbg scheme tp mycolormap in project directory.

Log shows:

Lepton EDA/lepton-schematic version 1.9.10.20200319 git: 56bb6e5
Loaded RC file [/usr/share/lepton-eda/system-gafrc]
Loaded RC file [/home/dennis/.config/lepton-eda/gafrc]
Loaded RC file [/home/dennis/Lepton/gafrc]
Loaded RC file [/usr/share/lepton-eda/system-gschemrc]

Backtrace:
          13 (apply-smob/1 #<catch-closure 556454ded1c0>)
          12 (apply-smob/1 #<catch-closure 556455423d80>)
          11 (primitive-load "/home/dennis/Lepton/mycolormap")
In ice-9/eval.scm:
   721:20 10 (primitive-eval (use-modules (lepton color-map)))
In ice-9/psyntax.scm:
  1262:36  9 (expand-top-sequence ((use-modules (lepton #))) _ _ #f e …)
  1209:24  8 (parse _ (("placeholder" placeholder)) ((top) #(# # …)) …)
   285:10  7 (parse _ (("placeholder" placeholder)) (()) _ c&e (eval) …)
In ice-9/boot-9.scm:
  3377:20  6 (process-use-modules _)
   222:17  5 (map1 (((lepton color-map))))
  3378:31  4 (_ ((lepton color-map)))
   2803:6  3 (resolve-interface _ #:select _ #:hide _ #:prefix _ # _ …)
In unknown file:
           2 (scm-error misc-error #f "~A ~S" ("no code for modu…" …) …)
In ice-9/boot-9.scm:
   751:25  1 (dispatch-exception 0 misc-error (#f "~A ~S" ("no…" …) …))
In unknown file:
           0 (apply-smob/1 #<catch-closure 556455423d40> misc-error # …)

ERROR: In procedure apply-smob/1:
no code for module (lepton color-map)

ERROR: Failed to load RC file [/home/dennis/Lepton/gschemrc]: no code for module (lepton color-map)

----------------

mycolormap:

( use-modules ( lepton color-map ) )

;
; Dark background color map for gschem
;

;  ;'s are comments
;  Keywords are case sensitive (Guile feature)
;  Colors are not case sensitive
;

; Colors may be specified in "#RRGGBB" or "#RRGGBBAA" format. If the alpha
; value is not specified, full opacity is assumed.
;
; If a color is #f (not a string; Scheme 'false'), then objects of
; that color will never be drawn.

(display-color-map
 '((background         "#000000")
   (pin                "#ffffff")
   (net-endpoint       "#ff0000")
   (graphic            "#00ff00")
   (net                "#0000ff")
   (attribute          "#ffff00")
   (logic-bubble       "#00ffff")
   (dots-grid          "#bebebe")
   (mesh-grid-major    "#1e1e1e")
   (mesh-grid-minor    "#171717")
   (detached-attribute "#ff0000")
   (text               "#00ff00")
   (bus                "#00ff00")
   (select             "#ffa500")
   (bounding-box       "#ffa500")
   (zoom-box           "#00ffff")
   (stroke             "#e5e5e5")
   (lock               "#bebebe")
   (output-background  #f)
   (junction           "#ffff00")
   (freestyle1         "#308a8e")
   (freestyle2         "#d02f2f")
   (freestyle3         "#b93ac1")
   (freestyle4         "#cccccc")
   ))

(display-outline-color-map
 '((background         #f)
   (pin                "#ababab")
   (net-endpoint       "#00ab00")
   (graphic            "#00ab00")
   (net                "#5500ab")
   (attribute          "#abab00")
   (logic-bubble       "#00abab")
   (dots-grid          #f)
   (mesh-grid-major    #f)
   (mesh-grid-minor    #f)
   (detached-attribute "#ab0000")
   (text               "#00ab00")
   (bus                "#00ab00")
   (select             "#ffa500")
   (bounding-box       "#ffa500")
   (zoom-box           "#00ffff")
   (stroke             "#e5e5e5")
   (lock               "#a9a9a9")
   (output-background  #f)
   (junction           "#abab00")
   (freestyle1         "#308a8e")
   (freestyle2         "#d02f2f")
   (freestyle3         "#b93ac1")
   (freestyle4         "#cccccc")
   ))

graahnul-grom, 2023-05-10:
The `( lepton color-map )` module has been added
in `Lepton EDA 1.9.12 (20201204)`.
Please check if its file exists:
`/usr/share/lepton-eda/scheme/lepton/color-map.scm`.

vzh, 2024-03-02:
No reply and no other info for months, closing this.  Please reopen if you feel like to add some more details.

closed 2024-03-02 by vzh