lepton-eda/lepton-eda #1032: lepton-schematic not starting

122christo, 2023-07-09:
Hi
the new installed lepton-eda is not starting. This is my terminal:

christoph@chris:~$ lepton-schematic
;;; note: auto-compilation is enabled, set GUILE_AUTO_COMPILE=0
;;;       or pass the --no-auto-compile argument to disable.
;;; compiling /usr/local/bin/lepton-schematic
;;; compiling /usr/local/share/lepton-eda/scheme/lepton/m4.scm
;;; compiled /home/christoph/.cache/guile/ccache/3.0-LE-8-4.5/usr/local/share/lepton-eda/scheme/lepton/m4.scm.go
;;; compiling /usr/local/share/lepton-eda/scheme/lepton/color-map.scm
;;; compiling /usr/local/share/lepton-eda/scheme/lepton/ffi.scm
;;; compiling /usr/local/share/lepton-eda/scheme/lepton/ffi/lib.scm
;;; compiled /home/christoph/.cache/guile/ccache/3.0-LE-8-4.5/usr/local/share/lepton-eda/scheme/lepton/ffi/lib.scm.go
;;; compiled /home/christoph/.cache/guile/ccache/3.0-LE-8-4.5/usr/local/share/lepton-eda/scheme/lepton/ffi.scm.go
;;; WARNING: compilation of /usr/local/share/lepton-eda/scheme/lepton/color-map.scm failed:
;;; In procedure dlsym: Error resolving "lepton_toplevel_new": "/usr/lib/x86_64-linux-gnu/liblepton.so: undefined symbol: lepton_toplevel_new"
;;; compiling /usr/local/share/lepton-eda/scheme/lepton/gettext.scm
;;; compiled /home/christoph/.cache/guile/ccache/3.0-LE-8-4.5/usr/local/share/lepton-eda/scheme/lepton/gettext.scm.go
;;; compiling /usr/local/share/lepton-eda/scheme/lepton/log.scm
;;; compiled /home/christoph/.cache/guile/ccache/3.0-LE-8-4.5/usr/local/share/lepton-eda/scheme/lepton/log.scm.go
;;; compiling /usr/local/share/lepton-eda/scheme/lepton/config.scm
;;; compiling /usr/local/share/lepton-eda/scheme/lepton/gerror.scm
;;; compiled /home/christoph/.cache/guile/ccache/3.0-LE-8-4.5/usr/local/share/lepton-eda/scheme/lepton/gerror.scm.go
;;; compiled /home/christoph/.cache/guile/ccache/3.0-LE-8-4.5/usr/local/share/lepton-eda/scheme/lepton/config.scm.go
;;; compiling /usr/local/share/lepton-eda/scheme/lepton/eval.scm
;;; compiled /home/christoph/.cache/guile/ccache/3.0-LE-8-4.5/usr/local/share/lepton-eda/scheme/lepton/eval.scm.go
;;; compiling /usr/local/share/lepton-eda/scheme/lepton/file-system.scm
;;; compiled /home/christoph/.cache/guile/ccache/3.0-LE-8-4.5/usr/local/share/lepton-eda/scheme/lepton/file-system.scm.go
;;; compiling /usr/local/share/lepton-eda/scheme/lepton/os.scm
;;; compiled /home/christoph/.cache/guile/ccache/3.0-LE-8-4.5/usr/local/share/lepton-eda/scheme/lepton/os.scm.go
;;; compiling /usr/local/share/lepton-eda/scheme/lepton/srfi-37.scm
;;; compiled /home/christoph/.cache/guile/ccache/3.0-LE-8-4.5/usr/local/share/lepton-eda/scheme/lepton/srfi-37.scm.go
;;; compiling /usr/local/share/lepton-eda/scheme/lepton/version.scm
;;; compiled /home/christoph/.cache/guile/ccache/3.0-LE-8-4.5/usr/local/share/lepton-eda/scheme/lepton/version.scm.go
;;; compiling /usr/local/share/lepton-eda/scheme/schematic/core/gettext.scm
;;; compiled /home/christoph/.cache/guile/ccache/3.0-LE-8-4.5/usr/local/share/lepton-eda/scheme/schematic/core/gettext.scm.go
;;; compiling /usr/local/share/lepton-eda/scheme/schematic/ffi.scm
;;; compiled /home/christoph/.cache/guile/ccache/3.0-LE-8-4.5/usr/local/share/lepton-eda/scheme/schematic/ffi.scm.go
;;; WARNING: compilation of /usr/local/bin/lepton-schematic failed:
;;; In procedure dlsym: Error resolving "schematic_keys_get_event_keyval": "/usr/lib/x86_64-linux-gnu/libleptongui.so: undefined symbol: schematic_keys_get_event_keyval"
;;; compiling /usr/local/share/lepton-eda/scheme/schematic/ffi/gtk.scm
;;; compiled /home/christoph/.cache/guile/ccache/3.0-LE-8-4.5/usr/local/share/lepton-eda/scheme/schematic/ffi/gtk.scm.go
;;; compiling /usr/local/share/lepton-eda/scheme/schematic/gui/keymap.scm
;;; compiling /usr/local/share/lepton-eda/scheme/schematic/action.scm
;;; compiling /usr/local/share/lepton-eda/scheme/schematic/hook.scm
;;; compiled /home/christoph/.cache/guile/ccache/3.0-LE-8-4.5/usr/local/share/lepton-eda/scheme/schematic/hook.scm.go
;;; compiling /usr/local/share/lepton-eda/scheme/schematic/window.scm
;;; compiling /usr/local/share/lepton-eda/scheme/lepton/page/foreign.scm
;;; compiled /home/christoph/.cache/guile/ccache/3.0-LE-8-4.5/usr/local/share/lepton-eda/scheme/lepton/page/foreign.scm.go
;;; compiling /usr/local/share/lepton-eda/scheme/lepton/toplevel.scm
;;; compiled /home/christoph/.cache/guile/ccache/3.0-LE-8-4.5/usr/local/share/lepton-eda/scheme/lepton/toplevel.scm.go
;;; compiled /home/christoph/.cache/guile/ccache/3.0-LE-8-4.5/usr/local/share/lepton-eda/scheme/schematic/window.scm.go
;;; compiled /home/christoph/.cache/guile/ccache/3.0-LE-8-4.5/usr/local/share/lepton-eda/scheme/schematic/action.scm.go
;;; compiling /usr/local/share/lepton-eda/scheme/schematic/keymap.scm
;;; compiled /home/christoph/.cache/guile/ccache/3.0-LE-8-4.5/usr/local/share/lepton-eda/scheme/schematic/keymap.scm.go
;;; compiled /home/christoph/.cache/guile/ccache/3.0-LE-8-4.5/usr/local/share/lepton-eda/scheme/schematic/gui/keymap.scm.go
;;; compiling /usr/local/share/lepton-eda/scheme/schematic/menu.scm
;;; compiling /usr/local/share/lepton-eda/scheme/schematic/ffi/gobject.scm
;;; compiled /home/christoph/.cache/guile/ccache/3.0-LE-8-4.5/usr/local/share/lepton-eda/scheme/schematic/ffi/gobject.scm.go
;;; compiled /home/christoph/.cache/guile/ccache/3.0-LE-8-4.5/usr/local/share/lepton-eda/scheme/schematic/menu.scm.go
Backtrace:
           3 (primitive-load "/usr/local/bin/lepton-schematic")
In ice-9/eval.scm:
   187:27  2 (_ _)
   223:20  1 (proc #<directory (guile-user) 7f7e7c238c80>)
In unknown file:
           0 (%resolve-variable (7 . register-data-dirs) #<directory…>)

ERROR: In procedure %resolve-variable:
Unbound variable: register-data-dirs

What to do?
Thank you for helping
Chris

graahnul-grom, 2023-07-10:
Hi, Chris

Please try the following:
- make sure /usr/local/lib* is in your /etc/ld.so.conf config file
- rm -fr /home/christoph/.cache/guile/
- sudo ldconfig

122christo, 2023-07-10:
Hi,
thanks for your answer.

- rm -fr /home/christoph/.cache/guile/
- sudo ldconfig
makes no difference

- In /etc/ld.so.conf  I read:
- "include /etc/ld.so.conf.d/*.conf"

-A ls in  /etc/ld.so.conf.d/ gives me

christoph@chris:/etc/ld.so.conf.d$ ls
fakeroot-x86_64-linux-gnu.conf
libc.conf
zz_i386-biarch-compat.conf
i386-linux-gnu.conf
x86_64-linux-gnu.conf  
zz_x32-biarch-compat.conf

My system is a Linux Mint 21

What to do?
Chris

122christo, 2023-07-10:
addendum: The files content

- fakeroot-x86_64-linux-gnu.conf
/usr/lib/x86_64-linux-gnu/libfakeroot

- libc.conf
# libc default configuration
/usr/local/lib

- zz_i386-biarch-compat.conf
# Legacy biarch compatibility support
/lib32
/usr/lib32

- i386-linux-gnu.conf
# Multiarch support
/usr/local/lib/i386-linux-gnu
/lib/i386-linux-gnu
/usr/lib/i386-linux-gnu
/usr/local/lib/i686-linux-gnu
/lib/i686-linux-gnu
/usr/lib/i686-linux-gnu

- x86_64-linux-gnu.conf
# Multiarch support
/usr/local/lib/x86_64-linux-gnu
/lib/x86_64-linux-gnu
/usr/lib/x86_64-linux-gnu

- zz_x32-biarch-compat.conf
# Legacy biarch compatibility support
/libx32
/usr/libx32

vzh, 2023-07-10:
Hi @122christo,
In your log I see two `WARNING`s that may be a sign of inconsistencies in your installation.  Please try to uninstall Lepton, and carefully check there are no lepton files present in `/usr/local`.  You could use something like `find /usr/local -name '*lepton*`.  Besides, please make sure you have not installed lepton by other means, I mean the package manager of your distribution.  If there are no suspicious files, do `rm -rf ~/.cache/guile` once again, and try to re-install lepton.  HTH.

vzh, 2023-07-10:
BTW, on my system I have `ldconfig` output as follows
```
$ /sbin/ldconfig -p|grep lepton
        libleptongui.so.5 (libc6,x86-64) => /usr/local/lib/libleptongui.so.5
        libleptongui.so (libc6,x86-64) => /usr/local/lib/libleptongui.so
        libleptonattrib.so.5 (libc6,x86-64) => /usr/local/lib/libleptonattrib.so.5
        libleptonattrib.so (libc6,x86-64) => /usr/local/lib/libleptonattrib.so
        liblepton.so.11 (libc6,x86-64) => /usr/local/lib/liblepton.so.11
        liblepton.so (libc6,x86-64) => /usr/local/lib/liblepton.so
```
Could you please show what your `ldconfig` outputs?

122christo, 2023-07-10:
Hi
ok, I uninstalled lepton and rm all "*lepton*" files in usr/local.
Than I made a fresh install with
./configure --prefix=$HOME/lepton
make
sudo make install

This changed something, but not all. This is the new output:

christoph@chris:~/lepton/bin$ ./lepton-schematic
;;; note: auto-compilation is enabled, set GUILE_AUTO_COMPILE=0
;;;       or pass the --no-auto-compile argument to disable.
;;; compiling /home/christoph/lepton/bin/./lepton-schematic
;;; compiling /home/christoph/lepton/share/lepton-eda/scheme/lepton/m4.scm
;;; compiled /home/christoph/.cache/guile/ccache/3.0-LE-8-4.5/home/christoph/lepton/share/lepton-eda/scheme/lepton/m4.scm.go
;;; compiling /home/christoph/lepton/share/lepton-eda/scheme/lepton/color-map.scm
;;; compiling /home/christoph/lepton/share/lepton-eda/scheme/lepton/ffi.scm
;;; compiling /home/christoph/lepton/share/lepton-eda/scheme/lepton/ffi/lib.scm
;;; compiled /home/christoph/.cache/guile/ccache/3.0-LE-8-4.5/home/christoph/lepton/share/lepton-eda/scheme/lepton/ffi/lib.scm.go
;;; WARNING: compilation of /home/christoph/lepton/share/lepton-eda/scheme/lepton/ffi.scm failed:
;;; In procedure dlopen: file "liblepton.so", message "liblepton.so: Kann die Shared-Object-Datei nicht öffnen: Datei oder Verzeichnis nicht gefunden"
;;; WARNING: compilation of /home/christoph/lepton/share/lepton-eda/scheme/lepton/color-map.scm failed:
;;; Unbound variable: liblepton
;;; compiling /home/christoph/lepton/share/lepton-eda/scheme/lepton/gettext.scm
;;; compiled /home/christoph/.cache/guile/ccache/3.0-LE-8-4.5/home/christoph/lepton/share/lepton-eda/scheme/lepton/gettext.scm.go
;;; compiling /home/christoph/lepton/share/lepton-eda/scheme/lepton/log.scm
;;; compiled /home/christoph/.cache/guile/ccache/3.0-LE-8-4.5/home/christoph/lepton/share/lepton-eda/scheme/lepton/log.scm.go
;;; WARNING: compilation of /home/christoph/lepton/bin/./lepton-schematic failed:
;;; Unbound variable: liblepton
;;; compiling /home/christoph/lepton/share/lepton-eda/scheme/lepton/config.scm
;;; compiling /home/christoph/lepton/share/lepton-eda/scheme/lepton/gerror.scm
;;; compiled /home/christoph/.cache/guile/ccache/3.0-LE-8-4.5/home/christoph/lepton/share/lepton-eda/scheme/lepton/gerror.scm.go
;;; compiled /home/christoph/.cache/guile/ccache/3.0-LE-8-4.5/home/christoph/lepton/share/lepton-eda/scheme/lepton/config.scm.go
;;; compiling /home/christoph/lepton/share/lepton-eda/scheme/lepton/eval.scm
;;; compiled /home/christoph/.cache/guile/ccache/3.0-LE-8-4.5/home/christoph/lepton/share/lepton-eda/scheme/lepton/eval.scm.go
;;; compiling /home/christoph/lepton/share/lepton-eda/scheme/lepton/file-system.scm
;;; compiled /home/christoph/.cache/guile/ccache/3.0-LE-8-4.5/home/christoph/lepton/share/lepton-eda/scheme/lepton/file-system.scm.go
;;; compiling /home/christoph/lepton/share/lepton-eda/scheme/lepton/os.scm
;;; compiled /home/christoph/.cache/guile/ccache/3.0-LE-8-4.5/home/christoph/lepton/share/lepton-eda/scheme/lepton/os.scm.go
;;; compiling /home/christoph/lepton/share/lepton-eda/scheme/lepton/srfi-37.scm
;;; compiled /home/christoph/.cache/guile/ccache/3.0-LE-8-4.5/home/christoph/lepton/share/lepton-eda/scheme/lepton/srfi-37.scm.go
;;; compiling /home/christoph/lepton/share/lepton-eda/scheme/lepton/version.scm
;;; compiled /home/christoph/.cache/guile/ccache/3.0-LE-8-4.5/home/christoph/lepton/share/lepton-eda/scheme/lepton/version.scm.go
Backtrace:
In ice-9/boot-9.scm:
   222:29 19 (map1 _)
   222:29 18 (map1 _)
   222:29 17 (map1 _)
   222:29 16 (map1 _)
   222:29 15 (map1 _)
   222:29 14 (map1 _)
   222:29 13 (map1 _)
   222:29 12 (map1 _)
   222:29 11 (map1 _)
   222:17 10 (map1 (((lepton version)) ((schematic core gettext)) # …))
  3928:31  9 (_ ((lepton version)))
  3326:17  8 (resolve-interface (lepton version) #:select _ #:hide _ …)
In ice-9/threads.scm:
    390:8  7 (_ _)
In ice-9/boot-9.scm:
  3252:13  6 (_)
In ice-9/threads.scm:
    390:8  5 (_ _)
In ice-9/boot-9.scm:
  3536:20  4 (_)
   2835:4  3 (save-module-excursion _)
  3556:26  2 (_)
In unknown file:
           1 (primitive-load-path "lepton/version" #<procedure 7f36e…>)
In lepton/version.scm:
    36:32  0 (_)

lepton/version.scm:36:32: Unbound variable: liblepton

$ /sbin/ldconfig -p|grep lepton gives:

christoph@chris:~$ /sbin/ldconfig -p|grep lepton
christoph@chris:~$ 

As I understand: "lepton" is now only local, in home installed. And there is a lib missing ""liblepton.so"

What next to do?

vzh, 2023-07-10:
After a fresh install you need to do `sudo ldconfig` first to make sure the gnu linker *ld* knows the paths to necessary libraries.  However, there may be other issues with libraries on some systems, please look [here](https://lepton-eda.github.io/lepton-manual.html/Hardcode-library-names-for-Guile.html).  It may be in your case as well since AFAIK Linux Mint is a Debian derivative.  ITW, use `./configure USE_LDCONFIG=/sbin/ldconfig` instead of plain `./configure`.

122christo, 2023-07-10:
Now it works! Why? I don't know. After:

christoph@chris:~/Elektronik/lepton/lepton-eda-1.9.18$ ./configure USE_LDCONFIG=/sbin/ldconfig >> compile.txt
christoph@chris:~/Elektronik/lepton/lepton-eda-1.9.18$ make >> make.txt
christoph@chris:~/Elektronik/lepton/lepton-eda-1.9.18$ sudo make install >> make_install.txt
[sudo] Passwort für christoph:       
gtk-update-icon-cache: Cache file created successfully.
libtool: warning: relinking 'libleptonattrib.la'
gtk-update-icon-cache: Cache file created successfully.
gtk-update-icon-cache: Cache file created successfully.
gtk-update-icon-cache: Cache file created successfully.
libtool: warning: relinking 'libleptongui.la'
christoph@chris:~/Elektronik/lepton/lepton-eda-1.9.18$ sudo ldconfig
christoph@chris:~/Elektronik/lepton/lepton-eda-1.9.18$ lepton-schematic 

lepton starting successfull

Thanks for helping,
yours Chris

vzh, 2023-07-11:
Glad to know that it helped :-)
About crossed out lines.  Github uses Markdown markup for representing messages you input as plain text.  In this markup, tildas (`~`) mark text as ~crossed out~.  To avoid this, you could use quoting or code syntax as offered by markup icons on the top of the message you edit on github.

closed 2023-07-11 by vzh