lepton-eda/lepton-eda #1047: Behavioral source in Lepton-eda

niuniuaba, 2023-09-18:
Is behavioral source (for spice simulation) implemented in Lepton-eda? I cannot find a symbol for it.

vzh, 2023-09-18:
Not for SPICE, AFAIK.  It should work for Icarus Verilog but this is another story.  First, please make sure the free SPICE engines such as `ngspice` or `gnucap` support such sources.  Then creating your symbol for this purpose and sharing it would be the way to make Lepton better :-)

niuniuaba, 2023-09-27:
Thanks for the advice. I found I can use voltage source symbol by changing refdes to match B source type and then the schematic capture parses it to the netlister correctly. I think we can close this issue.

vzh, 2023-10-02:
Hi @niuniuaba,
Thank you for your feedback!  I would only ask you for sharing your symbol and adding a couple of lines of description for it in order to make our FOSS tool better.  Thank you in advance.
(BTW, didn't even imagine Omron engineers use our tools :-)  Incidentally, I have several Omron controllers at my current work.)

niuniuaba, 2023-10-14:
Hi @vzh,

So I mad symbols for behavioral voltage source (bv.sym)
[bv.sym.txt](https://github.com/lepton-eda/lepton-eda/files/12905813/bv.sym.txt)
and behavioral current source (bi.sym)
[bi.sym.txt](https://github.com/lepton-eda/lepton-eda/files/12905814/bi.sym.txt)
I just made them by a little bit of modification in voltage-3.sym.
for behavioral voltage source:
`device=BEHAVIORAL_VOLTAGE_SOURCE` 
`refdes=B?`
`value=V=expression`
for behavioral current source:
`device=BEHAVIORAL_CURRENT_SOURCE` (for behavioral current source)
`refdes=B?`
`value=I=expression`
When I generate the spice netlist for a simple example schematic I got error message reads
"Found unknown component.  Refdes = B****" 
but anyway the netlist is generated as expected and works for the spice engines I use.
For reference I also attach the code of example schematic and netlist here:
[bsource_example.sch.txt](https://github.com/lepton-eda/lepton-eda/files/12905815/bsource_example.sch.txt)
[bsource_example.netlist.txt](https://github.com/lepton-eda/lepton-eda/files/12905816/bsource_example.netlist.txt)

BTW, we have all those commercial EDA suits but I don't use them. My role is about device modeling and circuit simulation and I prefer something light, fast, and robust and geda (now lepton-eda) is one of my favorite. Actually I have been use this great tool for about 10 years.

And, I really hope you enjoy you time with our controllers.

Best wishes,
Y.J. Liu