lepton-eda/lepton-eda #1055: lepton-tragesym fails

tomdean1939, 2023-10-21:
This is the first time I have tried lepton-tragesym. It seems the GEDA link for docs is no longer there.
The tragesym input file:
[mcp4922.txt](https://github.com/lepton-eda/lepton-eda/files/13061945/mcp4922.txt)

Ubuntu 22.04
lepton-eda from git is up-to-date

> lepton-tragesym mcp4922.txt MCP4922-1.sym
Backtrace:
In ice-9/boot-9.scm:
  1752:10 10 (with-exception-handler _ _ #:unwind? _ # _)
In unknown file:
           9 (apply-smob/0 #<thunk 7fecdccb40c0>)
In ice-9/boot-9.scm:
    724:2  8 (call-with-prompt _ _ #<procedure default-prompt-handle?>)
In ice-9/eval.scm:
    619:8  7 (_ #(#(#<directory (guile-user) 7fecdccbac80>)))
In ice-9/boot-9.scm:
   2835:4  6 (save-module-excursion _)
  4380:12  5 (_)
In /usr/local/bin/lepton-tragesym:
  1021:27  4 (_)
    227:2  3 (file-contents->alist _)
    227:2  2 (match-dotted _)
    227:2  1 (match-dotted _)
    227:2  0 (match-dotted _)

/usr/local/bin/lepton-tragesym:227:2: In procedure match-dotted:
sxml-match "no matching clause found"

What am I doing wrong?

Tom Dean

graahnul-grom, 2023-11-04:
Hi, Tom!
Sorry for the long time it takes to respond
(being quite busy at $JOB).

- Pin position is defined by single letter: `l`, `r`, `t`, `b`
  for left, right, top, bottom, respectively.
- Source file should contain both `[options]` and `[geda_attr]`
  sections, each having at least one attribute.
  I think it could be considered as a bug - if not present,
  default values should be used.
- `lepton-tragesym` is documented in its man page:
  https://lepton-eda.github.io/man/lepton-tragesym.html.
  Also, you may take a look at examples in [tools/tragesym/examples](https://github.com/lepton-eda/lepton-eda/tree/master/tools/tragesym/examples).
  PR #1056 fixes the test script in that directory.

```
[geda_attr]
name=MCP4922
refdes=U?

[options]
sort_labels=off

[pins]
1	1	pwr	line	b		3V3
12	12	pwr	line	b		GND
14	14	out	line	r		VOUTA
10	10	out	line	r		VOUTB
3	3	in	dot	l		*CS
4	4	in	line	l		SCK
5	5	in	line	l		SDI
8	8	in	dot	l		*LDAC
9	9	in	dot	l		*SHDN
13	13	in	line	l		VREFA
11	11	in	line	l		VREFB
```

[mcp4922_fixed.txt](https://github.com/lepton-eda/lepton-eda/files/13256189/mcp4922_fixed.txt)

tomdean1939, 2023-11-05:
On 11/4/23 02:00, dmn wrote:
> Hi, Tom!
> Sorry for the long time it takes to respond
> (being quite busy at $JOB).
>
>   * Pin position is defined by single letter: |l|, |r|, |t|, |b|
>     for left, right, top, bottom, respectively.
>   * [...]

Thanks, I read the man page, but, not carefully.

Please close this issue.

Sorry for the noise.

Tom Dean

closed 2023-11-05 by graahnul-grom