lepton-eda/lepton-eda #1075: Windows Binary

SpaceDreams, 2024-12-02:
Hi! Excellent Program! I currently use it in fedora and ubuntu, but I need it in windows too. Would it be possible to share build steps for a windows build?

Thanks!

vzh, 2024-12-04:
Hi @SpaceDreams,
There was an [outstanding report by Dr. Peter Brett](https://debbugs.gnu.org/db/22/22406.html) about an issue with types collision between Windows and Linux, at least in Guile.  The issue raised back in 2016 or before, probably in Guile 2.0.11, and has not been solved yet.  Since then, there are many things changed in lepton, and now we support Guile 2.2 and 3.0 which still have the same issue. The versions 2.0.x are no longer supported.  While there are some WIP code snippets by the guile devs, e.g. by Mike Gran who participated in the discussion on that issue, they have not been merged into upstream stock guile branch yet.

I've tried to cross-compile lepton for windows several years ago.  However, the problem was still present. Lepton won't work without working guile.  If you have a desire, you can try to search for patches for guile on GH, I know, there were some back in 2022 or so, and try to compile guile for windows yourself first.  Then we could probably move forward.

Thanks,
  Vladimir

Kreijstal, 2025-02-24:
https://github.com/lyrra/guile/tree/wip-mingw

Kreijstal, 2025-03-17:
hi, we managed to build guile on windows

https://github.com/lyrra/guile/issues/2

vzh, 2025-03-17:
Great, thanks!