lepton-eda/lepton-eda #1076: --enable-atribs --with-gtk3

Kreijstal, 2025-02-24:
unfortunately your dependencies fail to build with gtk3...

vzh, 2025-03-03:
Debian disabled building the gtk2 port of Lepton more than a year ago.  They now always build with gtk3.
What's your OS?

BTW, **lepton-attrib** is built by default unless you yield `--disable-attrib` on the configure stage.

Kreijstal, 2025-03-03:
> Debian disabled building the gtk2 port of Lepton more than a year ago. They now always build with gtk3. What's your OS?
> 
> BTW, **lepton-attrib** is built by default unless you yield `--disable-attrib` on the configure stage.

archlinux, I had to disable attrib so it builds. otherwise it errors

vzh, 2025-03-03:
Ah, OK, I see.  Just found the info in the archlinux package repository I'd already seen before:
- https://aur.archlinux.org/packages/lepton-eda-gtk3-git

Thank you for the report!

vzh, 2025-03-08:
Could you please provide us with the error output you get?

Kreijstal, 2025-03-08:
<details><summary>Details</summary>
<p>

```
kreijstal@kreijstalnuc:~/git/lepton-eda$ ./autogen.sh
autoreconf: export WARNINGS=
autoreconf: Entering directory '.'
autoreconf: running: autopoint --force
autoreconf: running: aclocal --force -I m4
autoreconf: configure.ac: tracing
autoreconf: running: libtoolize --copy --force
libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, 'build-tools'.
libtoolize: copying file 'build-tools/ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: copying file 'm4/libtool.m4'
libtoolize: copying file 'm4/ltoptions.m4'
libtoolize: copying file 'm4/ltsugar.m4'
libtoolize: copying file 'm4/ltversion.m4'
libtoolize: copying file 'm4/lt~obsolete.m4'
autoreconf: configure.ac: not using Intltool
autoreconf: configure.ac: not using Gtkdoc
autoreconf: running: aclocal --force -I m4
autoreconf: running: /usr/bin/autoconf --force
configure.ac:22: warning: The macro 'AC_CONFIG_HEADER' is obsolete.
configure.ac:22: You should run autoupdate.
./lib/autoconf/status.m4:719: AC_CONFIG_HEADER is expanded from...
configure.ac:22: the top level
configure.ac:28: warning: The macro 'AC_GNU_SOURCE' is obsolete.
configure.ac:28: You should run autoupdate.
./lib/autoconf/specific.m4:489: AC_GNU_SOURCE is expanded from...
configure.ac:28: the top level
configure.ac:40: warning: The macro 'AC_TRY_CPP' is obsolete.
configure.ac:40: You should run autoupdate.
./lib/autoconf/general.m4:2762: AC_TRY_CPP is expanded from...
m4/lepton-host.m4:22: AX_HOST is expanded from...
m4/lepton-windows.m4:37: AX_WINDOWS_FLAGS is expanded from...
configure.ac:40: the top level
configure.ac:51: warning: The macro 'AC_LIBTOOL_WIN32_DLL' is obsolete.
configure.ac:51: You should run autoupdate.
m4/ltoptions.m4:148: AC_LIBTOOL_WIN32_DLL is expanded from...
configure.ac:51: the top level
configure.ac:51: warning: AC_LIBTOOL_WIN32_DLL: Remove this warning and the call to _LT_SET_OPTION when you
configure.ac:51: put the 'win32-dll' option into LT_INIT's first parameter.
./lib/autoconf/general.m4:2434: AC_DIAGNOSE is expanded from...
m4/ltoptions.m4:148: AC_LIBTOOL_WIN32_DLL is expanded from...
configure.ac:51: the top level
configure.ac:53: warning: The macro 'AC_PROG_LIBTOOL' is obsolete.
configure.ac:53: You should run autoupdate.
m4/libtool.m4:100: AC_PROG_LIBTOOL is expanded from...
configure.ac:53: the top level
configure.ac:59: warning: AC_PROG_LEX without either yywrap or noyywrap is obsolete
./lib/autoconf/programs.m4:743: _AC_PROG_LEX is expanded from...
./lib/autoconf/programs.m4:736: AC_PROG_LEX is expanded from...
aclocal.m4:1045: AM_PROG_LEX is expanded from...
configure.ac:59: the top level
configure.ac:69: warning: AM_PROG_MKDIR_P: this macro is deprecated, and will soon be removed.
configure.ac:69: You should use the Autoconf-provided 'AC_PROG_MKDIR_P' macro instead,
configure.ac:69: and use '$(MKDIR_P)' instead of '$(mkdir_p)'in your Makefile.am files.
./lib/autoconf/general.m4:2434: AC_DIAGNOSE is expanded from...
aclocal.m4:1140: AM_PROG_MKDIR_P is expanded from...
m4/po.m4:23: AM_PO_SUBDIRS is expanded from...
m4/gettext.m4:57: AM_GNU_GETTEXT is expanded from...
configure.ac:69: the top level
configure.ac:69: warning: The macro 'AC_TRY_LINK' is obsolete.
configure.ac:69: You should run autoupdate.
./lib/autoconf/general.m4:2918: AC_TRY_LINK is expanded from...
lib/m4sugar/m4sh.m4:690: _AS_IF_ELSE is expanded from...
lib/m4sugar/m4sh.m4:697: AS_IF is expanded from...
./lib/autoconf/general.m4:2249: AC_CACHE_VAL is expanded from...
./lib/autoconf/general.m4:2270: AC_CACHE_CHECK is expanded from...
[... 1395 lines cut ...]
  -> Erstelle Arbeitskopie des gtksheet git Repos...
Klone nach 'gtksheet'...
Fertig.
Zu neuem Branch 'makepkg' gewechselt
==> Beginne prepare()...
patching file glade/glade-gtksheet-editor.c
Reversed (or previously applied) patch detected!  Assume -R? [n] n
Apply anyway? [n] n
Skipping patch.
3 out of 3 hunks ignored -- saving rejects to file glade/glade-gtksheet-editor.c.rej
patching file gtksheet/meson.build
==> FEHLER: Ein Fehler geschah in prepare().
    Breche ab...
 -> Fehler beim Erstellen: gtksheet-git-exit status 4
 -> Die folgenden Pakete konnten nicht installiert werden. Ein manueller Eingriff ist erforderlich:
gtksheet-git - exit status 4
kreijstal@kreijstalnuc:~/git/lepton-eda$
```

<p>

</p>
</details> 

I guess gtksheet dependency is too out of date.

Kreijstal, 2025-03-08:
> Thank you.
> 
> Just a few notices for the today state of things:
> 
>     * Debian people have managed to build _gtksheet-4.3.13_ (probably
>       especially for **lepton-attrib** :-)): https://packages.debian.org/source/trixie/gtksheet
> 
>     * The same version of the package is mentioned for Arch here: https://aur.archlinux.org/packages/gtksheet
> 
> 
> I know nothing about the status of the latter, but maybe it is now working and you could just get it from your distribution?
> 
> If not, taking into account that the package for Debian works, there is a chance to get some patches from their sources.
> 
> BTW, AFAICS, there are only two errors in the gtksheet build output you've provided, and both about missing casts, IIUC.

yeah aur is just a source distribution and yay gets the package from the aur, so the errors you see come from the aur package themselves, you can argue it's an aur bug and I'd agree but it is a dependency if your package as well.. I don't know how mantained is gtksheet currently but it seems abandoned