lepton-eda/lepton-eda #1077: Debian package re-build failure in test suite

bdale, 2025-03-04:
As reported in Debian bug #1096231, the lepton-eda package in Debian has begun failing in the test suite, perhaps due to a change in some dependency.  I can re-create this on my laptop running current Debian unstable, the error looks like this (excerpting only the relevant portion).  I would appreciate any hints on how to address this issue.

PASS: geda:set-attrib-value! (1)
PASS: geda:set-attrib-value! (2)

** (process:483975): CRITICAL **: 22:02:04.601: eda_config_set_property: assertion '!eda_config_is_descendent (parent, config)' failed

(process:483975): GLib-ERROR **: 22:02:04.661: ../../../glib/gmem.c:177: failed to allocate 1152921504606846976 bytes
make[6]: *** [Makefile:1179: unit-tests/geda-config.log] Trace/breakpoint trap
make[6]: *** Deleting file 'unit-tests/geda-config.log'
make[6]: *** Waiting for unfinished jobs....

Loading: "unit-tests/geda-object-bounds.scm"
PASS: geda:bounds (1)

vzh, 2025-03-07:
Thank you for the report, I knew it'll happen some day, and let the user know :-)

Seems, it has nothing to do with lepton.

It is a known issue with some changes in Linux kernels which emerge every now and then:
- https://forums.opensuse.org/t/qemu-error-after-20460629-update-rollback-fixes-it/176497/20

It hit me as well when I updated the package `linux-image-amd64` on one of my machines from `6.12.11-1` to `6.12.12-1`.  Two days of figuring out what happened wrong without success until I found this.

bdale, 2025-03-09:
For what it's worth, this is still happening for me with the Debian kernel package linux-image-6.12.17-amd64.

vzh, 2025-03-10:
@bdale, my fault, sorry.  The kernel version doesn't matter, AFAICS.
Please try to apply the commit [1af992c](https://github.com/lepton-eda/lepton-eda/commit/1af992c7689c1545766656b14d79d34f9d855203) to  check if it fixes the issue.

bdale, 2025-03-12:
Yep, that fixes it!  Thanks.  I'll upload 1.9.18-3 to Debian with this commit cherry-picked on top of the existing package, resolving the bug for now.