lepton-eda/lepton-eda #1079: parameters

felix-salfelder, 2025-04-02:
Hi there.

I am looking for an established or existing approach to represent parameters in a geda/lepton schematic.

For example, consider a circuit that would look something like this

    .subckt my_resistor p n
      .parameter r0 = 1;
      .parameter w = 1;
      .parameter l = 1;
      R1 p n {r0*w/l}
    .ends

in a Spice dump or

    module my_resistor(p, n);
      [...]
      parameter real r0 = 1 from (0:inf);
      parameter real w = 1 from (0:inf);
      parameter real l = 1 from (0:inf);
      resistor #(.r(r0*w/l)) R1(p, n);
    endmodule;

in Verilog.

I found the netlist backend "spice-sdb" that produces .subckt declarations close to the former, but without the parameters. There seems to be no mention of parameters in the documentation. As everything works as expected with only constants, I must be missing something. Please help.

noqsi, 2025-04-02:
https://github.com/noqsi/gnet-spice-noqsi￼
noqsi/gnet-spice-noqsi: Improved SPICE netlister for Lepton EDA and gEDA
github.com
 i

You might try gnet-spice-noqsi. It can do things like this. But I’m no longer maintaining it: I’ve moved to KiCAD...

> 
> 
> felix-salfelder
>  created an issue 
> (lepton-eda/lepton-eda#1079)
> Hi there.
> 
> I am looking for an established or existing approach to represent parameters in a geda/lepton schematic.
> 
> For example, consider a circuit that would look something like this
> 
> .subckt my_resistor p n
>   .parameter r0 = 1;
>   .parameter w = 1;
>   .parameter l = 1;
>   R1 p n {r0*w/l}
> .ends
> in a Spice dump or
> 
> module my_resistor(p, n);
>   [...]
>   parameter real r0 = 1 from (0:inf);
>   parameter real w = 1 from (0:inf);
>   parameter real l = 1 from (0:inf);
>   resistor #(.r(r0*w/l)) R1(p, n);
> endmodule;
> in Verilog.
> 
> I found the netlist backend "spice-sdb" that produces .subckt declarations close to the former, but without the parameters. There seems to be no mention of parameters in the documentation. As everything works as expected with only constants, I must be missing something. Please help.
> 
> 
> 
> felix-salfelder
>  created an issue 
> (lepton-eda/lepton-eda#1079)
> Hi there.
> 
> I am looking for an established or existing approach to represent parameters in a geda/lepton schematic.
> 
> For example, consider a circuit that would look something like this
> 
> .subckt my_resistor p n
>   .parameter r0 = 1;
>   .parameter w = 1;
>   .parameter l = 1;
>   R1 p n {r0*w/l}
> .ends
> in a Spice dump or
> 
> module my_resistor(p, n);
>   [...]
>   parameter real r0 = 1 from (0:inf);
>   parameter real w = 1 from (0:inf);
>   parameter real l = 1 from (0:inf);
>   resistor #(.r(r0*w/l)) R1(p, n);
> endmodule;
> in Verilog.
> 
> I found the netlist backend "spice-sdb" that produces .subckt declarations close to the former, but without the parameters. There seems to be no mention of parameters in the documentation. As everything works as expected with only constants, I must be missing something. Please help.

John Doty
***@***.***

felix-salfelder, 2025-04-02:
On Wed, Apr 02, 2025 at 01:05:58PM -0700, John P. Doty wrote:
> You might try gnet-spice-noqsi. It can do things like this.

Thanks. It might indeed. I had tried the backends listed by
"lepton-netlist -b", including spice-noqsi, but without much success.

Now there is no relevant mention of "parameter" or "param" in the
default branch at https://github.com/noqsi/gnet-spice-noqsi. Could you
give a more direct hint please?

regards
felix

noqsi, 2025-04-03:
gnet-spice-noqsi gives you a flexible capability to compose SPICE code, but you still have to tell it what you want.

One way to proceed:

Draw a schematic of your subcircuit. Give it a global spice-prolog attribute starting with .subckt. Include your parameter definitions. Something like:

spice-prolog=.subckt my_resistor %up
  .parameter r0 = 1;
  .parameter w = 1;
  .parameter l = 1;
Give the resistor in the subcircuit the attribute:
value={r0*w/l}
Use gnet-spice-noqsi to make that into a circuit file.
Then, in the main schematic, add a resistor with the following attribute:
spice-prototype=X? %down my_resistor r0= w= l=

You can then attach additional attributes (r0,w, l) to the resistor symbol to adjust the parameters.

Use gnet-spice-noqsi to make that into a circuit file.

Combine the main and subcircuit files however you like (.include, command line, …).

> 
> 
> felix-salfelder
>  left a comment 
> (lepton-eda/lepton-eda#1079)
> On Wed, Apr 02, 2025 at 01:05:58PM -0700, John P. Doty wrote:<br> &gt; You might try gnet-spice-noqsi. It can do things like this.<br> <br> Thanks. It might indeed. I had tried the backends listed by<br> &quot;lepton-netlist -b&quot;, including spice-noqsi, but without much success.<br> <br> Now there is no relevant mention of &quot;parameter&quot; or &quot;param&quot; in the<br> default branch at https://github.com/noqsi/gnet-spice-noqsi. Could you<br> give a more direct hint please?<br> <br> regards<br> felix<br>
> 
> 
> felix-salfelder
>  left a comment 
> (lepton-eda/lepton-eda#1079)
>  <https://github.com/lepton-eda/lepton-eda/issues/1079#issuecomment-2773669238>On Wed, Apr 02, 2025 at 01:05:58PM -0700, John P. Doty wrote:<br> &gt; You might try gnet-spice-noqsi. It can do things like this.<br> <br> Thanks. It might indeed. I had tried the backends listed by<br> &quot;lepton-netlist -b&quot;, including spice-noqsi, but without much success.<br> <br> Now there is no relevant mention of &quot;parameter&quot; or &quot;param&quot; in the<br> default branch at https://github.com/noqsi/gnet-spice-noqsi. Could you<br> give a more direct hint please?<br> <br> regards<br> felix<br>

John Doty
***@***.***

felix-salfelder, 2025-04-03:
Thanks John. I am not looking for an inline SPICE hack. My SPICE example was for illustration only. My question is, how to express a parameter in a schematic, for tools to recognise as a parameter.

But I did some research. I found Attribute-compatibility.html in the docs (is there an online version of the manual to link to?) and it mentions a potentially useful "symbol" attribute that might come close. There already is

    symbol=schematic-port

to be *used [..] in order to simplify the netlister logic*. Are there any plans to complete the section? I propose to add

    symbol=schematic-parameter

to that section. Clearly both, ports and parameters belong to the interface of a subcircuit or module (i.e. schematic), hence this addition would be completely natural.

That said, I am unsure about how and where to use this "symbol" attribute. What I tried ties them up to the letter "C" somehow, which indicates components according to the manual. I found uses of "C" for non-components too. But how does it all add up?

noqsi, 2025-04-03:
> 
> 
> felix-salfelder
>  left a comment 
> (lepton-eda/lepton-eda#1079)
> Thanks John. I am not looking for an inline SPICE hack. My SPICE example was for illustration only. My question is, how to express a parameter in a schematic, for tools to recognise as a parameter.
> 
> 
> felix-salfelder
>  left a comment 
> (lepton-eda/lepton-eda#1079)
> Thanks John. I am not looking for an inline SPICE hack. My SPICE example was for illustration only. My question is, how to express a parameter in a schematic, for tools to recognise as a parameter.
Well, of course that’s opposed to the toolkit approach, where tools are agnostic to features that don’t concern them.
> But I did some research. I found Attribute-compatibility.html in the docs (is there an online version of the manual to link to?) and it mentions a potentially useful "symbol" attribute that might [...]
Ports have to do with connectivity, a netlister concern. Parameters are not a netlister concern, and have different semantics for different downstream tools. SPICE parameters are the responsibility of SPICE.
> That said, I am unsure about how and where to use this "symbol" attribute. What I tried ties them up to the letter "C" somehow, which indicates components according to the manual. I found uses of "C" for non-components too. But how does it all add up?
What's a component to the schematic editor is just a graphic, not necessarily an electrical object. Again, separation of concerns of different tools. Components have attached attributes, both inherited from the symbol file and addeded in the schematic.

John Doty
***@***.***

felix-salfelder, 2025-04-04:
> SPICE parameters are the responsibility of SPICE.

Dear John.

The symbol library has various port symbols on offer. Following your logic, port symbols are not needed because they "are the responsibility of SPICE". I'm not saying it makes no sense, but it defies the purpose of a schematic editor.

Schematic editors are normally used to enter models into a simulation system. Such models can be hierarchical.

The use of parameters is common practice, in SPICE and any other modelling/simulation framework. Parameters are user data and not specific to a tool.

It seems a standard symbol for "parameter" is needed. I will send a patch once I have figured out the technicalities.

Thanks
Felix

noqsi, 2025-04-04:
> 
> 
> felix-salfelder
>  left a comment 
> (lepton-eda/lepton-eda#1079)
> SPICE parameters are the responsibility of SPICE.
> 
> Dear John.
> 
> The symbol library has various port symbols on offer. Following your logic, port symbols are not needed because they "are the responsibility of SPICE". I'm not saying it makes no sense, but it defies the purpose of a schematic editor.
> 
> 
> felix-salfelder
>  left a comment 
> (lepton-eda/lepton-eda#1079)
> SPICE parameters are the responsibility of SPICE.
> 
> Dear John.
> 
> The symbol library has various port symbols on offer. Following your logic, port symbols are not needed because they "are the responsibility of SPICE". I'm not saying it makes no sense, but it defies the purpose of a schematic editor.
Symbols syntactically communicate designers’ intentions to downstream processing. In a toolkit, the semantics associated with the symbols are a downstream responsibility. That’s the architecture of Lepton.
> Schematic editors are normally used to enter models into a simulation system. Such models can be hierarchical.
Simulation is only a minority of flows. Type ‘lepton-netlist -b’ to see all the flows supported.
> The use of parameters is common practice, in SPICE and any other modelling/simulation framework. Parameters are user data and not specific to a tool.
But their *semantics* *are* specific to the downstream tool. This is one of the annoying things about KiCAD as an integrated tool: it wraps SPICE in a KiCAD environment, cutting you off from most of the interactive capabilities of an implementation like ngspice. Fortunately, it also lets you operate in a toolkit mode.
> It seems a standard symbol for "parameter" is needed. I will send a patch once I have figured out the technicalities.
You’re fighting the toolkit paradigm: you desperately want an integrated tool for *your* specific way of operating. Lepton isn’t the architecture you want. The “technicalities” include not breaking the dozens of downstream flows that you don’t understand.
> Thanks
> Felix

John Doty
***@***.***

felix-salfelder, 2025-04-04:
> You’re fighting the toolkit paradigm: you desperately want an integrated tool for *your* specific way of operating. Lepton isn’t the architecture you want. The “technicalities” include not breaking the dozens of downstream flows that you don’t understand.

No John. You are basically saying that parameters are not integral part
of a circuit model, which is complete nonsense. Your assertion contradicts
common sense and common practice. Make of it what you will.

To summarize, what could have become clear from my previous responses.

- You have been trying to tell me that I want to create SPICE netlists,
  which is not at all what I want to do. Neither SPICE nor netlists. 
- You are implying that I want to simulate, just to tell me that
  simulation is only a niche application. I never said I wanted to
  simulate anything.
- You are saying that KiCAD is acting to your annoyance. Go complain
  to them.
- You are saying I am breaking downstream flows. Maybe yours. Well
  deserved. All others can ignore parameter declarations they don't need.

Sorry, this conversation is fruitless, because you are dragging in the
wrong direction. I'm stopping here. Please don't try to evangelize me
any further and leave this discussion to developers with a sense for
reality. Lepton deserves better. 

Best wishes
felix

noqsi, 2025-04-04:
> On Apr 4, 2025, at 11:22 AM, felix ***@***.***> wrote:
> 
> You are saying I am breaking downstream flows. Maybe yours. Well
> deserved. All others can ignore parameter declarations they don't need.
The unique strength of gEDA/Lepton is its support of all those flows. It is otherwise undistinguished. Why are you even using it? Be that as it may, I’ve given up on it. My collaborators are all using KiCAD now, and it’s not terribly inflexible, so that’s what I’m using, too.

John Doty
***@***.***

vzh, 2025-04-05:
Hi Felix,
The documentation is [here](https://lepton-eda.github.io/).  I update it every now and then.

felix-salfelder, 2025-05-07:
> Hi Felix, The documentation is [here](https://lepton-eda.github.io/). I update it every now and then.

Thanks Vladimir.

This is now #1081. Closing.

closed 2025-05-07 by felix-salfelder