lepton-eda/lepton-eda #1084: unable to run tools from git repo

felix-salfelder, 2025-05-16:
With git commit 06816166a

    $ ./autogen.sh
    $ mkdir build; cd build
    $ ../configure; make

(seems to work fine)

then

      $ tools/schematic/lepton-schematic
      ;;; WARNING: compilation of /home/.../tools/schematic/lepton-schematic failed:
      ;;; no code for module (lepton m4)
      [backtrace stuff]

Seems to be looking for other files?

     $ make install # maybe here?
     $ /usr/local/bin/lepton-schematic
     [compiling stuff] .. error
     $ rm ~/.cache/guile/ -rf
     $ /usr/local/bin/lepton-schematic
     [compiling stuff] .. error, still.
     $ /usr/local/bin/lepton-schematic
     ... undefined symbol in /usr/lib/x86_64-linux-gnu/liblepton.so

Ok ok, but:

    $ export LD_LIBRARY_PATH=/usr/local/lib
    $ /usr/local/bin/lepton-schematic
     ... undefined symbol in /usr/lib/x86_64-linux-gnu/liblepton.so, still

So I also tried

    root# ldconfig
    $ hash -r
    $ which lepton-schematic
    /usr/local/bin/lepton-schematic <= PATH looks ok.
    $ /usr/local/bin/lepton-schematic
    ... undefined symbol in /usr/lib/x86_64-linux-gnu/liblepton.so ...
    $ ldconfig -p |grep liblepton.so
        liblepton.so.11 (libc6,x86-64) => /usr/local/lib/liblepton.so.11
        liblepton.so.11 (libc6,x86-64) => /lib/x86_64-linux-gnu/liblepton.so.11
        liblepton.so (libc6,x86-64) => /usr/local/lib/liblepton.so
        liblepton.so (libc6,x86-64) => /lib/x86_64-linux-gnu/liblepton.so

I use an ordinary Debian system, and normally things involving autotools and ld.so just work. There is no INSTALL file in Lepton, and README.md just says "make install". This shouldn't be required, but sounds failsafe. It does not work for me. What am I missing?

TIA

vzh, 2025-05-18:
From the output of `ldconfig` and the previous output, it looks like you have lepton from Debian already installed?  If so, it won't work this way.

felix-salfelder, 2025-05-18:
On Sat, May 17, 2025 at 11:27:26PM -0700, Vladimir Zhbanov wrote:
> >From the output of `ldconfig` and the previous output, it looks like
> >you have lepton from Debian already installed?  If so, it won't work
> >this way.

Sure, I have installed Lepton from Debian. But I would like to run the
freshly built one, e.g. in order to see the difference in behaviour.

I have no ambition to install two different Leptons, I merely tried it,
because running directly from $builddir seems broken. This second
installation was in /usr/local, and was not supposed to have anything to
do with the first one from Debian.

> If so, it won't work this way.

This looks like a bug to me. Other autotools packages do not have this
problem. The errors I get seem pretty convoluted, but clearly
LD_LIBRARY_PATH is not honoured (as it should). What is in the way to
run the tools from $builddir?

Something like a second OS installation is prohibitive for me because
unfortunately, I won't find the time to make one.

vzh, 2025-05-18:
The issue here is that if you install another lepton copy from sources, the versions of C libraries are the same but interface changes.  As I can see from your previous message, when you launch a lepton program from `/usr/local/bin`, it tries to use the library from `/usr/lib/x86_64-linux-gnu/` which has different API.

Yes, it's a bug, and ideally, I should have changed the version of a lib when the library changes.  However, I usually do it before a new release, so it happened.

I see a couple of workarounds in your case you could try:
- You could try to change the versions of modified libraries (`liblepton`, `libleptongui`, `libleptonattrib`) in `configure.ac` yourself before installation.
- You could try to set the `LIBLEPTON` and `LIBLEPTONGUI` environment variables to point to your installed libraries in `/usr/local` and see if the programs work.

HTH

vzh, 2025-05-18:
Hmm, the `GUILE_LOAD_PATH` variable has to be amended as well... The following command worked for me:
> GUILE_LOAD_PATH=./liblepton/scheme:./libleptongui/scheme:$GUILE_LOAD_PATH LIBLEPTON=liblepton LIBLEPTONGUI=libleptongui ./tools/schematic/lepton-schematic

felix-salfelder, 2025-05-19:
Thanks. With the following environment settings I got past the initial undefined symbol error.

    # sourcedir=${PWD}/..
    GUILE_LOAD_PATH=$PWD/liblepton/scheme:$sourcedir/libleptongui/scheme:$sourcedir/liblepton/scheme
    LIBLEPTON=$PWD/liblepton/src/.libs/liblepton.so
    LIBLEPTONGUI=$PWD/libleptongui/src/.libs/libleptongui.so

I had to set the full paths to those libraries. Now I'm stuck at

    $ ./tools/schematic/lepton-schematic
    Gtk-Message: 09:26:05.074: Failed to load module "gail"
    Gtk-Message: 09:26:05.074: Failed to load module "atk-bridge"
    Backtrace:
    In ice-9/boot-9.scm:
      1755:12 11 (with-exception-handler _ _ #:unwind? _ # _)
    In unknown file:
              10 (apply-smob/0 #<thunk 7fc501f10300>)
    In ice-9/boot-9.scm:
        724:2  9 (call-with-prompt _ _ #<procedure default-prompt-handle…>)
    In ice-9/eval.scm:
        619:8  8 (_ #(#(#<directory (guile-user) 7fc501f13c80>)))
    In ice-9/boot-9.scm:
       2858:4  7 (save-module-excursion _)
      4408:12  6 (_)
    In /[...]/lepton-eda/build/./tools/schematic/lepton-schematic:
        331:6  5 (_)
       321:17  4 (activate _ _)
       256:17  3 (main #<pointer 0x0> ())
    In schematic/window.scm:
      1125:13  2 (make-schematic-window _ _)
    In lepton/config.scm:
        391:6  1 (config-boolean _ _ _)
       127:10  0 (gerror-error _ config-boolean)
    
    lepton/config.scm:127:10: In procedure gerror-error:
    Throw to key `config-error' with args `(config-boolean "Configuration does not have group 'schematic.gui'" () (group-not-found))'.

I do not know what gail or atk-bridge is. I would expect them to be in the ordinary places and not require any further environment tweaking. I also do not know how to interpret the error in config.scm.

vzh, 2025-05-19:
`gail` and `atk-bridge` are GTK modules which GTK tries to initialize somehow and probably without success.  Lepton has nothing to do with it as it is some internal GTK stuff, and the messages are purely informational.

The message from Guile is more interesting for me as it shows:
a) The function `make-schematic-window()` in `schematic/window.scm` at the line 1125 calls the function `config-boolean()`
b) In the sources, at least in the `master` branch, the call for `config-boolean()` is at the line 1099 of `windows.scm`.
c) There were no group `schematic.gui` in lepton 1.9.18, I suspect.

The three points suggest, that your program tries to use wrong code when loading `(schematic window)` module defined in the file `schematic/window.scm`, which hints it uses the previously compiled code of the program from your Debian distribution.

The first thing I would do is to remove Guile cache once again and try to launch the program and see if anything changes.

HTH

vzh, 2025-05-19:
And the Guile User Manual suggest to use something like:
`GUILE_AUTO_COMPILE=fresh ./your/path/to/lepton-schematic your-schematic.sch`
to force compilation of any newer Scheme files. (The compilation can be disabled as well, though I'm not sure it is of any use in this case.)

felix-salfelder, 2025-05-19:
OK, I managed to just run the editor from builddir, but only after "make install" as follows

    GUILE_LOAD_PATH=
    GUILE_LOAD_PATH+=$PWD/liblepton/scheme
    GUILE_LOAD_PATH+=:$PWD/libleptongui/scheme
    GUILE_LOAD_PATH+=:$PWD/libleptongui/scheme/schematic
    GUILE_LOAD_PATH+=:$PWD/libleptongui/scheme/schematic/window
    GUILE_LOAD_PATH+=:$PWD/libleptongui/scheme/geda
    GUILE_LOAD_PATH+=:$PWD/libleptongui/scheme/lepton
    GUILE_LOAD_PATH+=:$PWD/libleptongui/scheme/lepton/library
    GUILE_LOAD_PATH+=:$PWD/libleptongui/scheme/netlist
    GUILE_LOAD_PATH+=:$PWD/../liblepton/scheme
    GUILE_LOAD_PATH+=:$PWD/../libleptongui/scheme
    GUILE_LOAD_PATH+=:$PWD/../libleptongui/scheme/schematic
    GUILE_LOAD_PATH+=:$PWD/../libleptongui/scheme/schematic/window
    GUILE_LOAD_PATH+=:$PWD/../libleptongui/scheme/geda
    GUILE_LOAD_PATH+=:$PWD/../libleptongui/scheme/conf
    GUILE_LOAD_PATH+=:$PWD/../liblepton/scheme/lepton
    GUILE_LOAD_PATH+=:$PWD/../liblepton/scheme/lepton/library
    GUILE_LOAD_PATH+=:$PWD/../liblepton/scheme/netlist
    export GUILE_LOAD_PATH
    export LIBLEPTON=$PWD/liblepton/src/.libs/liblepton.so
    export LIBLEPTONGUI=$PWD/libleptongui/src/.libs/libleptongui.so
    ./tools/schematic/lepton-schematic

It seems at least some configuration files are expected in /usr/local (==$prefix).
The excessive need for subdirectories in GUILE_LOAD_PATH hints at some broken relative include, which I did not trace down any further.

It would be nice to see this fixed in one way or another. I'll stick to those workarounds, or to 1.9.18 for now.

Thanks for your help!